universam1/iSpindel #134: WiFi config not saving?

jfarcher, 2018-01-30:
Started building a couple of devices tonight, flashed the image to 2 Wemos D1 with gyros attached,
I'm getting the initial config mode portal, but when I add the SSID and passphrase it doesn't seem to save.
Upon attaching a serial monitor, both devices are slightly different.
device 1 after reboot:

> FW 5.6.2
> 2.1.0(deb1901)
> Worker run!
> Boot-Mode: 2
> mounting FS... mounted!

> ERROR config corrupted
> ...........
> ERROR no Wifi credentials

> going to Config Mode
> reboot RFCAL
> sd␀$��|␀�l�<␃␌␄␄�␄l�␄c|��␃�␓�r�#l�c��gn�dog���␄c␜8ćlsd;l␛�;�g�c�'c8␀�

and it just freezes

Second device:
> FW 5.6.2
> 2.1.0(deb1901)
> Worker run!
> Boot-Mode: 2
> mounting FS... mounted!
> reading config file
> opened config file

> parsed json
> applying offsets
> parsed config:
> {"Name":"iSpindel000","Token":"","Sleep":900,"Server":"","API":0,"Port":80,"URL":"","Vfact":191.8,"SSID":"",
> "PSK":"","POLY":"-0.00031*tilt^2+0.557*tilt-14.054","aX":640,"aY":3280,"aZ":1070}...........
> ERROR no Wifi credentials

> going to Config Mode
> reboot RFCAL
> {l␀l��|␀�d�|␂␄␄␌�␌d�␌b<��␃�␛�{�c$�#��og�l'o���␌#␜p��${ls$␓�s��n�c�ocp␀�

Both devices hang there until the reset button is pressed, which then drops them back into config mode.

Any ideas?

jfarcher, 2018-01-30:
I just flashed the original (Arduino based) WiFiManager example to the device and it configures wifi and saves the configuration ok.

universam1, 2018-01-30:
You are missing the ~300Ohm resistor btw. D0 and RST!
Try to shortcut these pins for testing.

jfarcher, 2018-01-30:
I haven't finished soldering up the boards yet, didn't realise there was a dependency for everything to be in place. Will try shorting out d0 and rst this evening.

Thanks

universam1, 2018-01-30:
These kind of issue reports usually cost a beer! 😋

jfarcher, 2018-01-30:
Shorted D0 and RST with a resistor, the boot process doesn't seem to hang any more but its still failing to save the settings.
FW 5.6.2
2.1.0(deb1901)
Worker run!
Boot-Mode: 5
mounting FS... mounted!
reading config file
opened config file

parsed json
applying offsets
parsed config:
{"Name":"iSpindel000","Token":"","Sleep":900,"Server":"","API":0,"Port":80,"URL":"","Vfact":191.8,"SSID":"",
"PSK":"","POLY":"-0.00031*tilt^2+0.557*tilt-14.054","aX":640,"aY":3280,"aZ":1070}...........
ERROR no Wifi credentials

jfarcher, 2018-01-30:
I guess I'll get all fully soldered up and retry once all components are connected.

universam1, 2018-01-30:
Can you post the log immediately when pressing the save button?
What browser are you using? Chrome is optimized for.

jfarcher, 2018-01-30:
Log:
*WM: Sent wifi save page
ERROR: OW DISCONNECTED

Exception (9):
epc1=0x40104a55 epc2=0x00000000 epc3=0x00000000 excvaddr=0xffffffff depc=0x00000000

ctx: sys
sp: 3ffffdb0 end: 3fffffb0 offset: 01a0

>>>stack>>>
3fffff50:  00000056 00000001 40104186 3ffee890
3fffff60:  4022a599 3ffef670 00000001 00000000

3fffff70:  4023eb57 00000004 00000000 00000000
3fffff80:  3fff22a4 4023e506 3fffdab0 00000000
3fffff90:  3fffdcc0 3ffea820 00000000 40249637
3fffffa0:  3ffea820 40000f49 3fffdab0 40000f49
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v00000000
~ld

And I'm using Chrome.

jfarcher, 2018-01-30:
Further up the log:

*WM: WiFi save
*WM: Parameter
*WM: name
*WM: iSpindel000
*WM: Parameter
*WM: sleep
*WM: 900
*WM: Parameter
*WM: vfact
*WM: 191.80

*WM: Parameter
*WM:
*WM:
*WM: Parameter
*WM:
*WM:
*WM: Parameter
*WM: selAPI
*WM: 0
*WM: Parameter
*WM: token
*WM:
*WM: Parameter
*WM: server
*WM:
*WM: Parameter
*WM: port
*WM: 80
*WM: Parameter
*WM: url
*WM:
*WM: Parameter
*WM:
*WM:
*WM: Parameter
*WM: POLYN

*WM: -0.00031*tilt^2+0.557*tilt-14.054
*WM: Sent wifi save page

Not sure if its by design, but there doesn't seem to be any SSID or WiFi details listed

universam1, 2018-01-31:
Uh line #2 is already the answer, you have no DS18b20 connected and as you see that’s where it is crashing!
Looks like it’s going to be a sixpack 😁

closed 2018-01-31 by universam1

jfarcher, 2018-01-31:
Yeah, as I previously mentioned I didn't realise there was a reliance on all components to be present to test.

Cheers.