universam1/iSpindel #135: Tilt = 0.00

kjenset, 2018-02-01:
Hi

My ISpindel not showing the degrees. I have tried the calibration under maintenance, but even after 15min the blue led dosen't change from solid to blink.

The temperature is correct and the connection to ubidots works!

Any clue?

[image]

Regards
Kjenset

JensW2000, 2018-02-01:
Either the connections to the GY-251 have cold solder joints or a short circuit or the GY-521 is broken.

[screenshot_2018-02-01-20-17-22]<https://user-images.githubusercontent.com/36051241/35698792-6ed55960-078e-11e8-83e1-112e4240e688.png>

closed 2018-02-02 by kjenset

Naesstrom, 2019-10-22:
What was the problem, mine is showing 0.0 to

studawg66, 2020-01-22:
Did anyone find the answer to this?  I have re-soldered joints and replaced the gyro board and I still get 0.0. The LED on the gyro module is lit when it gets power and all other functions work just fine. Any ideas?  Can I just add jumper wires from the board to the D1 board to see if my PCB is bad?

Naesstrom, 2020-01-28:
I read somewhere else that the adress of the gyro might differ from one source to another and that makes it show 0 as tilt... Wish there would be a fix in the software for it since I have 5 ispindels soldered and ready that doesn't work right now!

universam1, 2020-01-28:
High guessing. Please attach your console logs at least

studawg66, 2020-01-28:
In my case, 2 out of the 5 gyros I ordered (same batch) did not work.  I tried some troubleshooting of the GY-521 board, but couldn't identify the issue.  Replaced the GY-521 with a new one and the iSpindel reads the angle as it should.  I have read that these GY-521 boards are really flaky.  If you are going to build 5 iSpindels, order 10 of the GY-521 to be safe.  They are cheap.

Naesstrom, 2020-01-29:
I wonder if it's related to this: https://github.com/universam1/iSpindel/issues/302
That some gyros have the correct adress and some don't depending on production

studawg66, 2020-01-29:
Could be related, but mine were from the same batch and the two that reported 0.0 reported 0.0 to Ubidots as well, so that seems to be a different issue.