universam1/iSpindel #136: Software Issue

Alchomist, 2018-02-03:
My Ispindel seems to have gone haywire.
After 1000 unsuccessful attemps to link with CraftbeerPi, I tried to revert back to Ubidots.
But both the Configuration page & the results you can see below, the Token & Server seem to be getting mixed up. It seems to be trying to connect to TControl. 
I've tried updating the firmware, but that also failed.

_PSK has been removed for clarity_

Any idea whats happening?

wait
wait
wait
wait
wait

Final-sleep: 5s; RT:27198
{d d⸮⸮| ⸮d⸮|⸮l⸮c|⸮⸮⸮⸮{⸮c⸮c⸮⸮oo⸮lgn⸮⸮⸮cx⸮⸮dslsdx⸮'⸮⸮d⸮⸮co⸮|l⸮⸮c⸮⸮gg⸮ d⸮⸮l`⸮ool o{⸮⸮⸮gc⸮ds⸮⸮oc⸮l⸮b⸮ld⸮d⸮⸮d⸮⸮n⸮
FW 5.1.2
2.0.0(656edbf)
Worker run!
Boot-Mode: 5
mounting FS... mounted!
reading config file
opened config file

parsed json
applying offsets
parsed config:
{"Name":"iSpindel001","Token":"10.0.0.14","Sleep":5,"Server":"i52NINXZ2D6xkPVpx1wbgCx9knE2Mb","API":4,"Port":5000,"URL":"","Vfact":191.800003,"SSID":"NETGEAR56_EXT","PSK":"","POLY":"-0.00031*tilt^2+0.557*tilt-14.054","aX":-1535,"aY":-794,"aZ":1789}
woken from deepsleep, normal mode
Spl 0: 70.90
Spl 1: 70.90
Spl 2: 70.88
Spl 3: 70.87
Spl 4: 70.93
Spl 5: 70.87
Spl 6: 70.85

a: -6372	15282	930	absTilt: 70.88	T: 31.01	V: 3.48	owT: 27.25	Gravity: 23.87
After waiting 1.50 s, result 3
10.0.0.18

universam1, 2018-02-03:
You are running a very old firmware plus your battery is empty. Charge full before trying to update the firmware, this is a known condition

Alchomist, 2018-02-04:
So i followed the instructions for updating the firmware & this is as far as i got:

pi@raspberrypi:~/esptool-0.4.12-linux64 $ ls -l /dev/ttyUSB*
crw-rw---- 1 root dialout 188, 0 Feb  4 13:19 /dev/ttyUSB0
pi@raspberrypi:~/esptool-0.4.12-linux64 $ su
Password: 
root@raspberrypi:/home/pi/esptool-0.4.12-linux64# ./esptool -vv -cd nodemcu -cb 921600 -cp /dev/ttyUSB0 -ca 0x00000 -cf ./firmware.bin
bash: ./esptool: cannot execute binary file: Exec format error

added sudo:
root@raspberrypi:/home/pi/esptool-0.4.12-linux64# sudo ./esptool -vv -cd nodemcu -cb 921600 -cp /dev/ttyUSB0 -ca 0x00000 -cf ./firmware.bin
./esptool: 1: ./esptool: Syntax error: ")" unexpected

Even downloaded the file to my iphone & it refused to update.

Are there any instructions anywhere to completely remove the iSpindel software from the RPi so i can start from scratch?

universam1, 2018-02-21:
looks like more a generic linux question than a ispindel issue. Closing for now, you can reopen a specific question.

closed 2018-02-21 by universam1