universam1/iSpindel #137: Suggestion: Let iSpindel transmit signal strength

PjotrekSE, 2018-02-06:
For us with a stainless steel fermentor, it would be nice if the iSpindel could also transmit signal strength. At least as an option. That would make it easier to test out optimal or sufficient positioning of router or repeater.

universam1, 2018-02-06:
sounds like a good idea!

universam1, 2018-02-09:
@PjotrekSE Try the new firmware version 5.8.0 that transmits the RSSI in dBm. Let me know how it goes.

closed 2018-02-09 by universam1

PjotrekSE, 2018-02-09:
Nice. I will try it out as soon as the current battery test is finished.

PjotrekSE, 2018-02-19:
Tested. Works like a charm. Was able to find the best position for the repeater, astonishingly not directly above the fermenter, but about 10 cm above and 15cm to the side. Here are the results:
Free air, 15cm distance, -35 dB
Free air, 60cm distance, -45 dB
65 liter steel fermenter with 55 liter water, fitted with 15mm silicon plug and fermentation valve:
10 cm above, 00 cm to the side: -57 - -50 dB
30 cm above, 00 cm to the side: -56 - -55 dB
60 cm above, 00 cm to the side: -59 - -57 dB
60 cm above, 15 cm to the side: -73 - -68 dB
30 cm above, 15 cm to the side: -67 - -57 dB
30 cm above, 15 cm to the side: -63 - -56 dB
10 cm above, 15 cm to the side: -56 - -51 dB

universam1, 2018-02-19:
Awesome 👏

commit 904417a3 (2018-02-09): transmitting RSSI Wifi strength
fix #137
commit 1d027d66 (2018-02-11): transmitting RSSI Wifi strength
fix #137