universam1/iSpindel #138: Timer Interval: Int32?

DottoreTozzi, 2018-02-06:
Hi Sam,

in an attempt to put the iSpindel to its limits, I've put one in my Flanders Red Ale, which is supposed to ferment over the next 12 months.
So I set the interval to 6 hours, which apparently causes an overflow.

Would it cause a lot of trouble to use an Int64 for this value instead?

Thanks,
Stephan

universam1, 2018-02-07:
Hey Stephan,

so the variable is actually https://github.com/universam1/iSpindel/blob/master/pio/src/iSpindel.cpp#L71 a `uint32` in seconds, that are 136 years so I wonder if you are able to drink it in your life 😄 

Now, as the hardware only supports ~72min max sleep time  I am doing a trick here to allow longer sleep time.
Check this code from here https://github.com/universam1/iSpindel/blob/master/pio/src/iSpindel.cpp#L550 there you see that I am reincarnating if there a further sleep 'steps' left to do. 
Actually I tested that piece of code quite thoroughly, so I wonder if you are using an older Firmware version maybe?

universam1, 2018-02-07:
@DottoreTozzi I found that https://github.com/universam1/iSpindel/blob/master/pio/src/iSpindel.cpp#L332 was actually limiting to only 4 digits so only 9999s. Changed that and it works for me, please double check if thats the case for you too.
Released under 5.7.1

DottoreTozzi, 2018-02-07:
Nice, will test, thank you!

DottoreTozzi, 2018-02-08:
Confirmed, works like a charm now, thank you!

universam1, 2018-02-08:
thanks for the test!