universam1/iSpindel #139: Help to choose drawing

gh112, 2018-02-07:
I need some help to choose a drawing. (first time 3d printing)

The PET I am going to order is 40.5x149 mm (inner diameter)
I am guessing that i need drawer-38x150-Geocaching.stl and maybe use some sandpaper, or possibly drawer-38x145-Geocaching.stl

Please advise if another one is more suitable

drewharbin, 2018-03-06:
I went with this design: https://www.thingiverse.com/thing:2225597
with this container: https://www.ebay.com/itm/XXLBIG-PET-Micro-Geocaching-container-geocache-Petling-preform-Soda-bottle-FTF/132266274519?epid=938807721&hash=item1ecbaf2ed7:g:AJ4AAOxy9dVTkNlp

gh112, 2018-03-06:
Was it a good fit? 
(According to the spec its "Width: 36.5mm-37.5mm + flexible mounts")
Maybe it does not matter if it shorter than the tube?

drewharbin, 2018-03-08:
It was the best fit for me....especially because of the "flexible mounts".  the first drawer I tried, I had to sand with a dremel and it looked like shat.  The area at the bottom of the tube allowed me to add just enough smelted lead.

gh112, 2018-03-08:
Thanks I will order that one

closed 2018-03-08 by gh112