universam1/iSpindel #140: Possibility of save locally the data and upload when connection is available

Perimec, 2018-02-08:
Dear,
thanks a lot for the current project. it is really amazing the work that you are doing here.

However I have a simple question/suggestion. Most people, like me, do the fermentation in a remote place of the house (most probably in a temperature controlled fridge or similar). In my case, I have no wifi in this place. 

I was then thinking, would be possible to save the data locally on the memory and then use a phone (acting as access point) to upload the data on the cloud? 

Thanks

Philusha1983, 2018-02-08:
Have you thought about WiFi repeater? They not pricy and might solve your
problems

Perimec, 2018-02-08:
I have the fermentation in the attic and there are other apartment in the middle... a repeater would not help...

ollinator2000, 2018-02-08:
Ich had similar problems. I solved this by using a Mini Wifi AP and a Raspberry Pi . The Spindel and the Rpi are connected to this closed wifi network . everytime i m near the fermenter ( in my case this is the cellar ) i connecting to this network with my phone an check the data.

Perimec, 2018-02-08:
uhh that could be a good idea.. but I cannot understand how it works. The Spindel is sending the data to the RPI and do you have a code running there that saves the data?

universam1, 2018-02-08:
That is a better approach from @ollinator2000 to solve this problem. There is a principle problem with that the iSpindel has no internal clock available other than sourced from internet which is a catch 22 situation. Also it is questionable that you would be in range right in time when it wakes up. Also the battery lifetime makes it impossible to host an Accesspoint itself, so basically this is not possible.

ollinator2000, 2018-02-08:
Yes , I’m running the tcp Server programmed by Tozzi on my RPI ! 😉 
https://github.com/DottoreTozzi/iSpindel-TCP-Server

Perimec, 2018-02-14:
Thanks for all you comments :)

closed 2018-02-20 by Perimec