universam1/iSpindel #141: Alternative hardware

FernandoGarcia, 2018-02-09:
Hi!

I found this project yesterday and seems very cool.

I have to say thanks to all contributors.

Looking to buy parts I found [this board](https://pt.aliexpress.com/item/WeMos-D1-Esp-Wroom-02-Motherboard-ESP8266-Mini-WiFi-Nodemcu-Module-18650-Battery/32828007433.html?spm=a2g03.search0104.3.2.24067ce6Z6pE8s&ws_ab_test=searchweb0_0,searchweb201602_2_10152_10151_10065_10344_10068_10547_10342_10343_10340_10548_10341_10084_10083_10618_10307_5722313_5711215_10313_10059_10534_100031_10103_10624_10623_10622_10621_10620_10142,searchweb201603_28,ppcSwitch_5&algo_expid=a254017a-f86e-4c8f-a426-b56a76ebab79-0&algo_pvid=a254017a-f86e-4c8f-a426-b56a76ebab79&priceBeautifyAB=0) that seems very good option to replace Wemos + TP4056 + Micro switch and 330 ohms resistor (or BAT43 diode).

[image]

It has an Esp-Wroom-02 on board, the description say it's as a nodeMCU.

For me the big advantage is that this board don't need be turned off to charging. So using a wireless charger won't be need to open the container at all.

Someone know this board? It's a good option?

Best regards.

universam1, 2018-02-09:
Nice board indeed. 

Now first, there is no obligation to turn the iSpindel off during charging, not sure what you mean with this. 

Not to open the container any more is not a technical problem by all means but a hygienical!  It turned out to infect the worth sooner or later by yeast you can’t remove in the thread. 

Now the biggest problem is the weight displacement. The battery has to be in the very bottom while the antenna has to be on the very top, above the water level. That unfortunately seems to be a problem with this board. 

Good luck and report back.

FernandoGarcia, 2018-02-09:
Hi!

I found [this posts](https://www.homebrewtalk.com/forum/threads/ispindle-diy-electronic-hydrometer.598187/page-13#post-8108612) talking about the problem with switch and calibration.

Best regards.

FernandoGarcia, 2018-02-21:
I bought the parts let's see what happens. :D

universam1, 2018-03-26:
no progress on this one, closing for now

closed 2018-03-26 by universam1

FernandoGarcia, 2018-08-30:
Hi @universam1!

I built the controller using this board.

I'll post a tutorial coming soon.

The pinout used is:
A0 for battery monitor;
D1 - D2 for I2C BUS;
D7 for DS18B20.

Here the link for 3D printed part.

https://www.thingiverse.com/thing:3075683

Best regards.

universam1, 2018-08-30:
Be informed that I do not tolerate that my project is being sold or commercialized without my permission. Check the license associated.

FernandoGarcia, 2018-08-30:
Of course!

I'm not selling any hardware or software just tube and support.

FernandoGarcia, 2018-08-30:
Maybe you should call yours advocates for this guy. :D

https://produto.mercadolivre.com.br/MLB-1062940698-ispindel-densimetro-digital-montado-completo-_JM

aagilang, 2025-01-19:
how to flash this board.. i cant connect

FernandoGarcia, 2025-01-19:
Hi! 
This board is a waste of time and money. 
The battery won't last 2 days. 
It has serious engineering problems.
Anyway [here](https://github.com/FernandoGarcia/iSpindel-for-MQTT-only/blob/master/platformio.ini) you can see the configuration to flash this board.
Best regards.