universam1/iSpindel #142: Support for Fahrenheit

drewharbin, 2018-02-15:
Not really an issue, but will we see an option to change the temp reading to Fahrenheit in future releases?  I have my ispindel connected to my Craftbeerpi which is set to F and the temp is sent as C but denoted as F on my dashboard.

Thanks,
Drew

FernandoGarcia, 2018-03-02:
Hi!
Are you using compiled version?
Best regards.

drewharbin, 2018-03-02:
Just using latest firmware release.

drewharbin, 2018-03-04:
Awesome! Got it going!  The above overloaded the function but once I figured out how to compile I was able to make the edits to convert to Fahrenheit.  Thank you so much for the guidance,

universam1, 2018-03-04:
A proper implementation is always welcome as a PR!

FernandoGarcia, 2018-03-04:
I can make it selectable via code but not via web page for now.

universam1, 2018-03-04:
Actually the request of this feature is somewhat questionable since the iSpindel is not the system that is presenting a user interface thus coping with the users formatting. 
It provides a common interface of communication that the backend is expecting. Now if we introduce optional formats we break a standard!

In this case craftbeerpi is to support several temperatur formats to display, which it does! So we could even state that it is a big or missing implementation in CBP and a issue has to be open for that. 
I am not against Fahrenheit to support but I am worried about breaking a standard

FernandoGarcia, 2018-03-04:
Well, you are the admin of this branch so if do you want add this option let me know and will try a PR.

closed 2018-03-06 by drewharbin