universam1/iSpindel #146: calibration procedure

wizbrewery, 2018-02-24:
Hi,  firstly thanks for your fab and well documented project.  I'm calibrating my iSpindle and I'm a little puzzled by the firmware calibration results.  After selecting maintenance,  calibrate,  I get the following readings.

when laying ispindle at 0 deg  x,y it reads 68 deg,  and if I stand it upright it reads 62 deg.
the polynomial is -0.00031*tilt^2+0.557*tilt-14.054

The calibration tilt is wrong or am I missing a step please?

I'm using your latest firmware 5.8.2

universam1, 2018-02-27:
You can reset the config with latest firmware and start from scratch. Report back if it stays.

riisefrutti, 2018-03-17:
I just found the same issue with 5.8.2 - Laying flat on table 89 deg. In pure water 55deg, when held horizontally 40 deg. This is with 15gr of weight on the sled.

riisefrutti, 2018-03-17:
The issue is repeating when doing calibration offset through firmware. Just did a factory reset, and got "sain" values. After doing an update the 5.8.5 I did a new offset calibration, and god the same result. 48 deg at 0 vertical. 88-90 deg laying flat on surface.

Without the offset calibration I get 8,8 in pure water and 77 at horizontal. Any takes?

closed 2018-07-12 by universam1