universam1/iSpindel #149: Ispindel config - no com port

SimonHanson, 2018-02-27:
Windows 10
When i plug in my ispendel i can not getting a com port showing in device manager
Therefore cant flash it

JensW2000, 2018-02-27:
Do you probably forgot to install the CH350G Device driver?
https://wiki.wemos.cc/downloads

SimonHanson, 2018-02-27:
Installed still not working

https://wiki.wemos.cc/downloads

JensW2000, 2018-02-27:
Please connect a USB cable to the charging module (TP4056).
Wich lights does lit after a single reset? What lights does lit if you turn off the power switch?

SimonHanson, 2018-03-04:
Switch position one
PLug in USB Blue light on, tiny orange flash on the other led
hit reset  nothing

Switch position 2
Plug in uds blue and orange lights on 
hit reset blue light on the other board

SimonHanson, 2018-03-05:
Please can anyone help me with this problem

universam1, 2018-03-26:
no progress on this one, closing

closed 2018-03-26 by universam1

SimonHanson, 2018-03-26:
I still need to get an answer