universam1/iSpindel #150: POLYN size

badcrc, 2018-02-27:
Hi,

I just notice that the polynomial variable is limited to 70 chars

https://github.com/universam1/iSpindel/blob/e2c6818d3d064be46373b8ffb82ec9a521b96afb/pio/src/iSpindel.cpp#L66

The online tool gives me a 2nd degree and a 3rd degree polynomials I tried to use the 3rd degree one and I was getting very odd values.

Checking the configuration in 5.6.1 the text was truncated and in 5.8.2 it was garbled.

Given that the online tool can give a high in chars polynomial, can it be increased?  

For example mine is:

```
-0.000035707*(tilt)*(tilt)*(tilt)+0.009843649*(tilt)*(tilt)+-0.869739117*(tilt)+25.926019535
```

Or should I just round the numbers?

Thanks!

PS. I was going to do it myself and send the PR but I don't know the implications of being 70.

badcrc, 2018-02-27:
Also, I just noticed that in the config page it says `(tilt)*(tilt)` and in the .cpp it uses `tilt^2`

Are both notations accepted?

universam1, 2018-02-27:
You can shorten the addition, it won’t make a difference. Plus the power notation is also working fine

badcrc, 2018-02-28:
OK, thanks!

closed 2018-02-28 by badcrc