universam1/iSpindel #151: WiFi passwords containing a $ sign doesn't since FW 5.8.1

JensW2000, 2018-02-27:
Very strange behavior. 
When the WiFi PAS contains a double $ sign and you save the config, then the iSpindle stops the iSpinled hotspot but doesn't reboot and endless continues the "config-mode flashing" (without reboot). 

*WM: Sent wifi save page
*WM: Connecting to new AP
*WM: Connecting wifi with new parameters...
*WM: previous settings invalidated
*WM: After waiting...

After a manual reset the iSpindle can't connect to the WLAN. 
When you reenter the config mode each $ sign in the PSK was replaced with "%24".

The strange thing is: If I connect to another SSID (another VLAN), where the PSK doesn't contain a $ sign (WiFI works) and I reenter the config mode, go back to the 1st SSID, then the 1st SSID and the correct PSK is show in the config, but the 2nd SSID and the matching PSK is used in background for the still working WiFi connection. Even there's no trace entry "using rescue credentials" th this case.

universam1, 2018-02-27:
Can you please post the log, especially the part that covers the save and read of the config file. 

Also I am a bit confused what you mean with “VLAN”, guess that should be WLAN right?

JensW2000, 2018-02-27:
Nope VLAN. I've multiple strictly isolated WLAN SSIDs. Each SSID has it's own (untagged) VLAN and a own IP Subnet.

I used the SSID "JBLM" with the IP Subnet 10.10.10.0/24 and the PSK with a $ sign since the iSpindle firmware 1.x.
After I flashed 5.8.1 all my iSpindles wasn't able to connect to the SSID JBLM.
When I enter the Config-Mode, I see, that the $ signs in the PSK was replaced with %24.
When I enter the correct PSK ($ signs instead of %24) the ispindle doesn't reboot after saving the config (but still shows the config-flashing). 
After a manual reset thje iSpindle can't connect to WiFi. (after waiting: 6259 ms ... rescue WiFi Credentials ..  going to deepsleep)

Afterwards I entered the config-mode and coosed another of my SSIDs (JBLM-IOT-PLAYGROUND), with a PSK without $ signs and a separate IP Subnet. 
The WiFi Connection is working and posting the Data to Ubidots too.

Back in Config Mode I tried to go back to the JBLM SSID:

-----------
parsed config:
{"Name":"J2","Token":"myToken","Sleep":900,"Server":"","API":0,"Port":80,"URL":"","Vfact":191.8,"SSID":"JBLM-IoT-Playground","PSK":"PSK without $ sign","POLY":"-0.002193647*tilt^2+0.832191623*tilt-30.46536106","aX":0,"aY":0,"aZ":0}
going to Config Mode
*WM: Scan done
*WM: DUP AP: JBLM-Besucher
*WM: DUP AP: JBLM-Voucher-WLAN
*WM: DUP AP: JBLM
*WM: DUP AP: JBLM-IoT-Playground
*WM: Adding parameter
*WM: name
*WM: Adding parameter
*WM: sleep
*WM: Adding parameter
*WM: vfact
*WM: Adding parameter
*WM:
*WM: Adding parameter
*WM:
*WM: Adding parameter
*WM: selAPI
*WM: Adding parameter
*WM: token
*WM: Adding parameter
*WM: server
*WM: Adding parameter
*WM: port
*WM: Adding parameter
*WM: url
*WM: Adding parameter
*WM:
*WM: Adding parameter
*WM: POLYN
started Portal
*WM: SET AP
*WM:
*WM: Configuring access point...
*WM: iSpindel
*WM: AP IP address:
*WM: 192.168.4.1
*WM: HTTP server started
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Request redirected to captive portal
*WM: JBLM-Besucher
*WM: -25
*WM: JBLM-Voucher-WLAN
*WM: -25
*WM: JBLM
*WM: -27
*WM: JBLM-IoT-Playground
*WM: -27
*WM: DIRECT-52-HP JBLM
*WM: -52
*WM: TKMS-R1
*WM: -75
*WM: CCJBLM
*WM: -83
*WM: o2-WLAN89
*WM: -84
*WM: Stockton
*WM: -87
*WM: WLAN-437667
*WM: -88
*WM: EasyBox-164247
*WM: -89
*WM: Sent config page
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Handle root
*WM: WiFi save
*WM: Parameter
*WM: name
*WM: J2
*WM: Parameter
*WM: sleep
*WM: 900
*WM: Parameter
*WM: vfact
*WM: 191.80
*WM: Parameter
*WM:
*WM:
*WM: Parameter
*WM:
*WM:
*WM: Parameter
*WM: selAPI
*WM: 0
*WM: Parameter
*WM: token
*WM: myToken
*WM: Parameter
*WM: server
*WM:
*WM: Parameter
*WM: port
*WM: 80
*WM: Parameter
*WM: url
*WM:
*WM: Parameter
*WM:
*WM:
*WM: Parameter
*WM: POLYN
*WM: -0.002193647*tilt^2+0.832191623*tilt-30.46536106
*WM: Sent wifi save page
*WM: Connecting to new AP
*WM: Connecting wifi with new parameters...
*WM: previous settings invalidated
*WM: After waiting...
*WM: 13.27
*WM: seconds
*WM: Connection result:
*WM: WL_CONNECT_FAILED
*WM: Failed to connect.
saving config...{"Name":"J2","Token":"myToken","Sleep":900,"Server":"","API":0,"Port":80,"URL":"","Vfact":191.8,"SSID":"JBLM","PSK":"$$PSK","POLY":"-0.002193647*tilt^2+0.832191623*tilt-30.46536106","aX":0,"aY":0,"aZ":0}saved successfully
Spl 0: 0.00
Spl 1: 0.00
Spl 2: 0.00
Spl 3: 0.00
Spl 4: 0.00
Spl 5: 0.00
Spl 6: 0.00
a: 0     0       0       absTilt: 0.00   T: 36.53V: 4.24 owT: 21.50      Gravity: -30.47
After waiting 0ms, result 0
Rescue Wifi credentials
failed to connect
Final-sleep: 1s; RT:81724
rll▒▒|▒l▒|
▒d▒
c|▒▒▒r▒c▒c▒▒gg▒doo▒▒▒
bpćl{d;lp▒o▒
▒l
▒▒c
g▒|d▒
▒c▒▒oo▒l▒▒d`▒nnl`ns▒▒▒g
c▒lr▒▒n
#
▒l▒
l`▒g▒
FW 5.7.0
2.1.0(deb1901)
Worker run!
Boot-Mode: 5
mounting FS... mounted!
reading config file
opened config file
parsed json
offsets not available
parsed config:
{"Name":"J2","Token":"myToken","Sleep":900,"Server":"","API":0,"Port":80,"URL":"","Vfact":191.8,"SSID":"JBLM","PSK":"$$myPSK","POLY":"-0.002193647*tilt^2+0.832191623*tilt-30.46536106","aX":0,"aY":0,"aZ":0}
woken from deepsleep, normal mode
Spl 0: 0.00
Spl 1: 0.00
Spl 2: 0.00
Spl 3: 0.00
Spl 4: 0.00
Spl 5: 0.00
Spl 6: 0.00
a: 0  0       0       absTilt: 0.00   T: 36.53       V: 4.24  owT: 21.50      Gravity: -30.47
After waiting 2551ms, result 3
IP: 10.76.158.86

calling Ubidots
{"tilt":0,"temperature":21.5,"battery":4.244004,"gravity":-30.46536}Sender: Ubidots posting
POST /api/v1.6/devices/J2?token=myToken HTTP/1.1
Host: things.ubidots.com
User-Agent: ESP8266
Connection: close
Content-Type: application/json
Content-Length: 68

POST /api/v1.6/devices/J2?token=myToken HTTP/1.1
Host: things.ubidots.com
User-Agent: ESP8266
Connection: close
Content-Type: application/json
Content-Length: 68

HTTP/1.1 200 OK
Server: nginx
Date: Tue, 27 Feb 2018 22:16:49 GMT
Content-Type: application/json
Transfer-Encoding: chunked
Connection: close
Vary: Accept-Encoding
Vary: Cookie
Allow: GET, POST, HEAD, OPTIONS

8d
{"battery": [{"status_code": 201}], "tilt": [{"status_code": 201}], "temperature": [{"status_code": 201}], "gravity": [{"status_code": 201}]}
0

Final-sleep: 900s; RT:5349

----------------

As you can see:
If I was conencted to the SSID "JBLM-IOT-PLAYGROUND" once, I'm able to save the config, the iSpindle does a reboot after saving the config, the loaded config is showing the SSID "JBLM" and the correct PSK.
But ... the iSpindle connects hiddenly to the SSID "JBLM-IOT-PLAYGROUND", gets an IP of the related "Playground IP Network" (10.76.158.86), and it doesn's appear in the configured WLAN SSID.

universam1, 2018-02-27:
So IEEE 802.11 has no concept of VLANs by itself, that is logically analogous to WiFi SSIDs what you are referring to. 

Now, what you are to know is that the Esp8266 has a special register to hold the last WiFi credentials that are automagically loaded on boot up hence connecting if possible.

This is actually used to again a extremely fast connection after sleep interval. And in the log there is a message posted if these register are resetted or invalidated for new credentials. Now if this very refresh does not succeed therefore the old credentials are still existing in the register and will cause a reconnect to the old SSID. 

Now it would be interesting to see what and why the credentials are not updated. Unfortunately with your scrambled logs I can’t determine really what’s going on there...

JensW2000, 2018-02-27:
I scrambled just the passwords and tokens :-)
This are the log messages after saving a password starting with 2 $ signs (lets say: $$MyPassword123).

*WM: Sent wifi save page
*WM: Connecting to new AP
*WM: Connecting wifi with new parameters...
*WM: previous settings invalidated
*WM: After waiting...
*WM: 13.27
*WM: seconds
*WM: Connection result:
WM: WL_CONNECT_FAILED
WM: Failed to connect.
saving config...{"Name":"J2","Token":"myToken","Sleep":900,"Server":"","API":0,"Port":80,"URL":"","Vfact":191.8,"SSID":"JBLM","PSK":"$$MyPassword123","POLY":"-0.002193647tilt^2+0.832191623tilt-30.46536106","aX":0,"aY":0,"aZ":0}saved successfully
Spl 0: 0.00
Spl 1: 0.00
Spl 2: 0.00
Spl 3: 0.00
Spl 4: 0.00
Spl 5: 0.00
Spl 6: 0.00
a: 0 0 0 absTilt: 0.00 T: 36.53V: 4.24 owT: 21.50 Gravity: -30.47
After waiting 0ms, result 0
Rescue Wifi credentials
failed to connect
Final-sleep: 1s; RT:81724

universam1, 2018-02-28:
The question is if the password phrase was correctly received from the html form into the config file and can be read again correctly.

Thats why I was questioning your scrambled secret if it really reflects the conditions.

Please double check if this is the case. If so, then the error must be in the never SDK.

JensW2000, 2018-02-28:
I think it's an HTML encoding issue. Unfortunately the behavior isn't consistent. 
The config behaves different once the spindle was connected to a working SSID and has the credentials in it's mind. 

A spindle, that has just seen the SSID "JBLM" (with the $ sign PSK) is showing %24 instead of the $ in the PSK field, after loading the config. There's no reboot after saving but the iSpindel AP desapears and the config-flasing keeps alive.

A spindle, that was once connected to the SSID "JBLM-IOT-PLAYGROUND" (without the $ sign PSK) works fine. If I turn again into the config-mode and go back to the SSID "JBLM" and enter the "$ sign PSK" the iSpindle does a reboot after saving, sending data and so on. Everything looks fine, but the spindle uses secretly the priorly configured SSID "JBLM-IOT-PLAYGROUND". I doubleckecked this. 
If I enter the config-mode once more, I see, that the SSIS "JBLM" ist coosen and the PSK with the $ signs ist displayed currectly. 

Unfortunately I'm unable to investigate the 1st case deeper, because all of my ispindles were already connected to the 2nd SSID in the meantime.

PjotrekSE, 2018-02-28:
Maybe this worx?
wifiManager.resetSettings();

universam1, 2018-02-28:
I double checked and I am sure this is not a Firmware issue. The special character `$$` is correctly saved and restored. 
Thats why I ask you for a screenshot and logs of config save and restore otherwise I am unable to debug your problem.

universam1, 2018-02-28:
Did you compile yourself?

JensW2000, 2018-02-28:
No, it's the official firmware from the repo.
I can't provide any new logs at this time. See above. All my ispindles have seen the 2nd SSID and has fall into the behavior where the hidden credentals are used.
I've to order some Wemos D1 Mini to create new logs of the 1st behavior. Or you believe me, that the logs always presented the "$$ PSK" (with  $ signs) while saving and loading the config. The HTML presented the PSK as "%24%24PSK".

universam1, 2018-02-28:
Just tested, and it does not appear in my test setup. 
You must be clearly running an older version than 5.8.1, see --> https://github.com/universam1/iSpindel/commit/e2c6818d3d064be46373b8ffb82ec9a521b96afb

JensW2000, 2018-02-28:
The prior log was a try to go back to an older firmware that worked with that spindle and my WiFi.

Here's a fully unscrambled log for a try to change the SSID back to "JBLM".

1st save -> Reboot because of backup cedentials
2nd and 3rd save -> hanging

If you cant see anything close this issue.  I planned to change my WiFi PSK next time. 

------------- 5.8.2 ----------------

d`gr▒▒▒n
#
▒l{▒▒g
c▒d▒
l`▒o▒
     FW 5.8.2
             2.2.0(f28eaf2)
                           Worker run!
                                      Boot-Mode: 5
                                                  mounting FS... mounted!
                                                                         reading                                                                            config file
            opened config file

                              parsed json
                                         offsets not available
                                                              parsed config:
                                                                            {"Na                                                                           me":"J2","Token":"A1E-ToKeN123","Sleep":900,"Server":"","A                                                                           PI":0,"Port":80,"URL":"","Vfact":191.8,"SSID":"JBLM","PSK":"$$password123","PO                                                                           LY":"","aX":0,"aY":0,"aZ":0}
                            woken from deepsleep, normal mode
                                                             Spl 0: 70.70
                                                                         Spl 1:                                                                            70.36
     Spl 2: 70.24
                 Spl 3: 69.88
                             Spl 4: 69.89
                                         Spl 5: 69.75
                                                     Spl 6: 68.85

                                                                 a: -8184      1                                                                           6906    -1386   absTilt: 70.00  T: 27.29        V: 4.23 owT: 24.44Parse error at                                                                            1
        Gravity: 0.00
                     After waiting 1ms, result 3
                                                IP: 10.76.158.86

                                                                calling Ubidots
                                                                               {                                                                           "tilt":70.00491,"temperature":24.4375,"battery":4.233577,"gravity":0,"interval":                                                                           900,"RSSI":-45}Sender: Ubidots posting
POST /api/v1.6/devices/J2?token=A1E-ToKeN123 HTTP/1.1
Host: things.ubidots.com
User-Agent: ESP8266
Connection: close
Content-Type: application/json
Content-Length: 96

POST /api/v1.6/devices/J2?token=A1E-ToKeN123 HTTP/1.1
Host: things.ubidots.com
User-Agent: ESP8266
Connection: close
Content-Type: application/json
Content-Length: 96

Final-sleep: 900s; RT:5168
                          sll▒▒|▒l▒|
▒
l▒
#|▒▒▒▒s▒c▒
c▒▒no▒lgg▒▒▒cp▒▒l;lslp▒g▒▒
l
▒▒
c
g▒|$▒
▒b▒▒og▒l▒▒d`▒oo
l`o{▒▒▒g
[... 971 lines cut ...]
Parse error at 1
Parse error at 1
Parse error at 1
Parse error at 1
Parse error at 1
Parse error at 1
Parse error at 1
Parse error at 1
Parse error at 1
Parse error at 1
Parse error at 1
Parse error at 1
Parse error at 1
Parse error at 1
Parse error at 1
Parse error at 1
Parse error at 1
Parse error at 1
Parse error at 1
Parse error at 1
Parse error at 1
Parse error at 1
Parse error at 1
Parse error at 1
Parse error at 1

closed 2018-03-08 by universam1