universam1/iSpindel #152: Configuration problem (wrong SSID & firmware problem)

juhanius, 2018-02-28:
Hi,

After soldering the iSpindel kit, I did not manage to proceed with the instructions. When switched on and after pressing reset, the appearing wifi is not named "iSpindel", but "ESP_4EED91". 
The wifi allows me to connect, but doesn't appear to do anything. The address 192.168.4.1 is not found. Also, the blue LED keeps blinking in about 2 s interval also beyond the 5 minute period.

I continued with these instructions in order to access the log:
https://www.youtube.com/watch?v=6xDW5GbIbUw

I managed to access the device with Baud rate 74880 (not 115200, which were instructed). What happens is when I press the reset button, the following appears in the log:

> 
> ets Jan  8 2013,rst cause:2, boot mode:(3,6)
> 
> load 0x4010f000, len 1264, room 16
> tail 0
> chksum 0x42
> csum 0x42
> ~ld

Nothing else seems to happen. When pressing reset again, exactly the same log text reappears.

I tried to flash the firmware, (with these instructions, Windows & USB: https://github.com/universam1/iSpindel/blob/master/docs/Firmware_en.md) but NodeMCU only gives me the following error: "Error:Serial port connect failed, please check it." 
NodeMCU seems to automatically find the device in COM3 port (from where I found the log with PuTTY), but I have no idea how to proceed here. I tried both baudrates (115200, which was in the documentation and 74880, which worked with PuTTY and the log), but with no success.

I understood that configuring the device even without first updating the firmware should be possible. So, is there something wrong with this device? Or is the problem only in the firmware update (where I apparently don't know how to use the NodeMCU)?

All help is appreciated!

universam1, 2018-02-28:
Without uploading the firmware the esp8266 is brain dead! How did you come to the conclusion this is not required?

You might need to install the drivers first, go to Wemos to download them for your operating system. Then you can select the com port an upload.

juhanius, 2018-02-28:
Ah, like I somehow suspected, the main problem indeed was between chair and keyboard! As the firmware upload was not listed as a step in the configuration instructions, I assumed that there would be an existing firmware. Thank you for clarifying this!
The driver download solved this, resulting in succesfully uploading the firmware. However, now the log looks like this after pressing reset:

>   
>                                                                   ;l$▒▒<▒l▒|
> ▒
> d▒c|▒▒▒s▒b▒
> #▒▒og▒lng▒▒▒cp▒▒ds$s$p▒'▒
> ▒
> $
> ▒▒
> co▒<l▒
> ▒#▒▒og▒lćd`▒n
> d`'{▒▒▒g
> c
> ▒d;▒▒g
> c
> ▒d▒
> d`▒o▒
>      FW 5.7.0
>              2.1.0(deb1901)
>                            Worker run!
>                                       Boot-Mode: 6
>                                                   power-cycle or reset detected,                                                                      config mode
>             mounting FS... mounted!
> 
>                                    ERROR config corrupted
>                                                          ...........
>                                                                     ERROR no Wif                                                                     i credentials
> 
>              going to Config Mode
>                                  reboot RFCAL
> ;ld▒▒|▒l▒<
> ▒
> d▒c|ǃ▒▒r▒#▒
> #▒▒ng▒l'o▒▒▒c8▒▒dsdsdp▒'▒
> ▒
> d
> ▒▒
> c
> n▒<l▒
> ▒c▒▒ng▒l▒▒d`▒'
> d`gs▒▒▒o
> c▒l{▒▒g
> c▒l▒
> l`▒▒'▒
>       FW 5.7.0
>               2.1.0(deb1901)
>                             Worker run!
>                                        Boot-Mode: 5
>                                                    mounting FS... mounted!
> 
>                                                                           ERROR                                                                      config corrupted
>                 ...........
>                            ERROR no Wifi credentials
> 
>                                                     going to Config Mode
>                                                                         ERROR: O                                                                     W DISCONNECTED
>               ERROR: OW DISCONNECTED
>                                     *WM: Scan done
> *WM: Adding parameter
> *WM: name
> *WM: Adding parameter
> *WM: sleep
> *WM: Adding parameter
> *WM: vfact
> *WM: Adding parameter
> *WM:
> *WM: Adding parameter
> *WM:
> *WM: Adding parameter
> *WM: selAPI
> *WM: Adding parameter
> *WM: token
> *WM: Adding parameter
> *WM: server
> *WM: Adding parameter                                                                ;l$▒▒<▒l▒|
> ▒
> d▒c|▒▒▒s▒b▒
> #▒▒og▒lng▒▒▒cp▒▒ds$s$p▒'▒
> ▒
> $
> ▒▒
> co▒<l▒
> ▒#▒▒og▒lćd`▒n
> d`'{▒▒▒g
> c
> ▒d;▒▒g
> c
> ▒d▒
> d`▒o▒
>      FW 5.7.0
>              2.1.0(deb1901)
>                            Worker run!
>                                       Boot-Mode: 6
>                                                   power-cycle or reset detected,                                                                      config mode
>             mounting FS... mounted!
> 
>                                    ERROR config corrupted
>                                                          ...........
>                                                                     ERROR no Wif                                                                     i credentials
> 
>              going to Config Mode
>                                  reboot RFCAL
> ;ld▒▒|▒l▒<
> ▒
> d▒c|ǃ▒▒r▒#▒
> #▒▒ng▒l'o▒▒▒c8▒▒dsdsdp▒'▒
> ▒
> d
> ▒▒
> c
> n▒<l▒
> ▒c▒▒ng▒l▒▒d`▒'
> d`gs▒▒▒o
> c▒l{▒▒g
> c▒l▒
> l`▒▒'▒
>       FW 5.7.0
>               2.1.0(deb1901)
>                             Worker run!
>                                        Boot-Mode: 5
>                                                    mounting FS... mounted!
> 
>                                                                           ERROR                                                                      config corrupted
>                 ...........
>                            ERROR no Wifi credentials
> 
>                                                     going to Config Mode
>                                                                         ERROR: O                                                                     W DISCONNECTED
>               ERROR: OW DISCONNECTED
>                                     *WM: Scan done
> *WM: Adding parameter
> *WM: name
> *WM: Adding parameter
> *WM: sleep
> *WM: Adding parameter
> *WM: vfact
> *WM: Adding parameter
> *WM:
> *WM: Adding parameter
> *WM:
> *WM: Adding parameter
> *WM: selAPI
> *WM: Adding parameter
> *WM: token
> *WM: Adding parameter
> *WM: server
> *WM: Adding parameter
> *WM: port
> *WM: Adding parameter
> *WM: url
> *WM: Adding parameter
> *WM:
> *WM: Adding parameter
> *WM: POLYN
> started Portal
>               *WM: SET AP
> *WM:
> *WM: Configuring access point...
> *WM: iSpindel
> *WM: AP IP address:
> *WM: 192.168.4.1
> *WM: HTTP server started
> ERROR: OW DISCONNECTED
>                       ERROR: OW DISCONNECTED
>                                             ERROR: OW DISCONNECTED
>                                                                   ERROR: OW DISC                                                                     ONNECTED
> 

The "ERROR: OW DISCONNECTED" keeps repeating. Is this a symptom of a faulty temperature sensor or some other component? (or a faulty soldering).

closed 2018-03-02 by juhanius