universam1/iSpindel #153: Calibration, should i use y2 or y3

onoEsp1, 2018-03-01:
Hi!

Im having some problems with the formel. I can't get the excel sheet to work, even with macros on.
On my PC i get error 400. And with OpenOffice on my mac the sheet dont work at all.
So i have tried to use the online calibration. 
But on that one I get degree 2y and degree 3y. With of them should I use?
And I also noticed that I had to change the x with * and type in tilt. Is this correct?
Here is my formel
0.004854135x^2 + -0.736610748x + 28.917126017
Changed it to
0.004854135*(tilt)*(tilt) + -0.736610748*(tilt) + 28.917126017

Thanks
from 
Espen

universam1, 2018-03-01:
You can use 2 or 3 degree, all will work. And your interpretation is correct

onoEsp1, 2018-03-02:
Ok, thanks!
It seems to work in portal. But when I try to put that format into ubidots I get "undefined"
I try to put it in the field "Derived expression"

My spindel lays quite flat in water and the tilt is just between 78,8-80,6, so maby that makes to smal parameters in the formel?
Many its best to put some weight in my spindel to get the angle steeper.

universam1, 2018-03-02:
There is no need to set the calculation up in Ubidots any more as the iSpindel does itself. 

And it is mandatory to balance it to 25 degree, but this is mentioned in the docs!

onoEsp1, 2018-03-02:
Aha! Thanks a lot!!
I have been reading a lot around this. But have not found the 25 degree rule.
Il weight my spindel then!
Thanks again for great and fast help!

FernandoGarcia, 2018-03-02:
Hi @universam1 !

"There is no need to set the calculation up in Ubidots any more as the iSpindel does itself."

Can you remove this section from here please?

https://github.com/universam1/iSpindel/blob/master/docs/Kalibrierung.md#alternativ-in-ubidots-nach-berechnen

Best regards.

universam1, 2018-03-02:
Well, it says optionally 😃

FernandoGarcia, 2018-03-02:
Well, but if the value is sent converted using this "option" the user will get a wrong value as OP told above, not?

universam1, 2018-03-02:
Nope, the tilt value is not converted, the gravity is computed

onoEsp1, 2018-03-02:
What kind of gravity is showing on ubidots. SG? Mine now says 1.03
No 4th digit. Have my spindle in sugar water now, measured to 1.025 SG

universam1, 2018-03-02:
Check the data of te values, it’s correctly transmitted only Ubidots rounds later to two digits precision. 

Closing as the topic drifts, open a dedicated issue.

closed 2018-03-02 by universam1