universam1/iSpindel #154: Petling/Tube dimensions

einaruk, 2018-03-02:
Hello,
   Great project. Firstly thanks for that.

My issue where I need support is that, I found a petling which has 33 mm inner diameter and around 120 mm length.
When I see the stl files of 3D prints for drawer I see they are bigger which won't fit in the tube. Do you think it is possible to use the tube I mentioned with specs above? 
Does anybody have an experince fitting the project in a smaller tube? 

Below there are images of the tube with dimensions;

Diameter;
[image]

Length;
[image]

einaruk, 2018-03-02:
Is edited one of the drawer designs and trimmed it. Lets see if it can fit. Will let you know...

einaruk, 2018-03-05:
Update: I was able to fit the drawer inside the tube, but had issues to fit the drawer after installing the battery.

jamesgol, 2018-03-20:
I ended up getting some tubes with a similar ID.  This design fit nicely https://www.thingiverse.com/thing:2169567  I was hoping I wouldn't have to design my own.  It also fit the wemos d1 mini pretty good and I used a wemos prototype board for the extra bits.

Had to add about 23g of weight to the bottom to get it to read 25 degrees in plain water.  Still need to calibrate at gravity.

fodi666, 2018-03-28:
Hi,

can someone help me with resizing the drawer to fit a [50ml Falcon tube](https://images-na.ssl-images-amazon.com/images/I/31Zpw7D9RQL._SX342_.jpg)?
I made some measurements:
- total external length: 114mm
- total length before the conical end, which has full width: 99mm
- internal diameter: 27mm

Is it possible to squeeze everything in?

Thanks

cherryphilip74, 2018-06-05:
I designed the ispindel pcb for this petling 
http://www.geoshop.lt/index.php?route=product/product&path=20_26&product_id=285#.WxafoPluZhF

[image]

Filippo

arudmin, 2018-06-05:
@cherryphilip74 super! Could you please share PCB or EasyEDA?

cherryphilip74, 2018-06-05:
I made two pbs:
the first (link below) already used with good behavoiur and it's very stable
https://www.pcbway.com/project/shareproject/W10971ASK36_iSpindel_v2_6.html

the newest version with battery holder, but not tested yet. (i receved the pcb today)
https://www.pcbway.com/project/shareproject/New_iSpindel_PCB_2_0.html

As my personal suggestion, choose the first because already tested

Filippo

cherryphilip74, 2018-06-08:
Hello everyone!
here my ispindel pcb assembled.
Good results without ballast
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Best Regards
Filippo

cherryphilip74, 2018-06-18:
Hi,
P//N 1043 Keystone, link below:
https://eu.mouser.com/ProductDetail/Keystone-Electronics/1043?qs=sGAEpiMZZMt13fpse6CWDUGg%2f8fzvjZvv8iRCb3qg5g%3d

Let's me know pls it's important receive any feedback /suggestions.

Best  Regards 
Filippo

mr-bridger, 2018-06-18:
Thanks Filippo, I've ordered all the parts and v2 PCB from pcbway. I will let you know once I have everything :-) One quick question, there are three through holes and 2 pads above the resistors, is this purely for separate power / direct power without charge board or do these need soldering to something, I can't quite see from the built unit photos. Thanks again.

bertybert, 2018-06-19:
@mr-bridger - I believe that is for a switch (part number SK12D07VG3)

mr-bridger, 2018-06-19:
@bertybert Thanks, it could be, i'm not so sure tho, as that switch would need 5 through holes (2 outer anchor points) maybe @cherryphilip74 can confirm?

bertybert, 2018-06-19:
@mr-bridger
I was going off the instructions on his previous board here https://www.pcbway.com/project/shareproject/W10971ASK36_iSpindel_v2_6.html

I'm in the same boat as you though, I've ordered bits (including that switch)... Wonder if there are any more instructions floating about.

mr-bridger, 2018-06-19:
@bertybert ah yes i see that has a BOM list, but not on the one in this thread. that one also has 6 pin TP4056, but the one above has a 4 pin version.

cherryphilip74, 2018-06-19:
Hi @bertybert I'm sorry for my answer late but for me was been hot days.
Here below you find BOM list:

- 1 Plastic cylinder
( http://www.geoshop.lt/index.php?route=product/product&path=20_26&product_id=285#.WoBTg-jOXIU )
- 1 resistor 4k7ohm
- 1 resistor 220kohm
- 1 resistor 330ohm
- 1 Wemos D1 mini v2 (or compatible)
- 1 Gy-521 Gyro & Acceleration Sensor (MPU-6050)
- 1 Temperature Sensor (DS18B20)
- 1 TP4056 Lipo Charger 4 pin
- 1 microswitch SK12D07VG3
- 1 holder battery 1043 keystone
- 1 Samsung Battery Q30

Last pcb i used a TP4056 6 pins, but to keep the same dimension but with holder battery I had change the lipo module charger
 
Let me know when you complete the assembling.
Cheers

Filippo

mr-bridger, 2018-06-21:
Hi @cherryphilip74 Thanks for the info! :) 
one more quick question for now. do i still need to remove diode from D1 Mini? 
[image]

cherryphilip74, 2018-06-27:
Hi @mr-bridger ,
it is not necessary to remove it.

mr-bridger, 2018-07-05:
hi @cherryphilip74  Built my first unit last night, Used a BAT43 diode instead of the 330ohm resistor, all seems to be working fine, thanks! only thing i did notice is that the TP4056 board doesnt really line up with the through holes, tho i guess this is due to china manufacturing differences of the same board.  here some pics anyway

[image]
[image]
[image]
[image]
[image]

cherryphilip74, 2018-07-05:
Hello @mr-bridger !
I am very happy to have found the pcb easy assembling. I confirm that the LiPo TP4056 module has an incorrect footprint due to wrong eagle file (download from the internet) and when I received it, I had not possibility to change it.

I did a new revision with bug fix, I want to ask to depository  owner if is possible to share it on github/ispindel server for community.

Other question: Could you share the tilt found when it is in water? Just to confirm that the default value is around 25°C  without ballast.

Thanks again for your comments

Best Regards
Filippo

cherryphilip74, 2018-07-11:
Hi guys,
Here you find the new pcb with tp4056 footprint correction. 
https://www.pcbway.com/project/shareproject/iSpindel_PCB_2_1__new_TP4056_module_.html

Cheers!!
Filippo

mr-bridger, 2018-07-12:
@cherryphilip74 Great thanks! i'm sorry i haven't given you more info on the angle yet, been a bit busy and haven't had chance to calibrate. but from my initial test in water, i did get it close to 25° but i had to make sure the board wasn't pushed right to the bottom of the tube. (its a tight fit though) ill probably need to find a position and then put a spot of hot melt to make sure it never moves. pushed right to the bottom it was showing 16°, but bear in mind that was a fresh uncalibrated GY521. i will update when i can.

bertybert, 2018-07-18:
@cherryphilip74 

I've built 2 of these now but each time, the temperature sensor fails to read. it reads -127.00
This is on the PCB with the battery holder as shown above.
The temp sensors are good as i've tested them in breadboard on an arduino uno. I'm using a 4K7 resistor. I've checked continuity between 5v/gnd/D6 on wemos etc. The circuit looks fine to me.

What could I be doing wrong?
I'm also confused about the resistor values. as your BOM above doesn't match the values printed on the board. On the board, I see values of 230K, 470R, 4K7. @mr-bridger has used a diode...

Any help would be appreciated.

Thanks

mr-bridger, 2018-07-18:
The diode is to replace the 470ohm resistor. There was a change at some point to a 330ohm on the main ispindel BOM. Then it was realised that a BAT43 diode was a better option. This is purely for the sleep function of the D1 mini. 

I have seen a value of -127 on my initial build but then for some reason it started working again. It's been said this error is a hardware issue. Either bad connection or faulty sensor. Also check this issue page. https://github.com/universam1/iSpindel/issues/182

https://github.com/universam1/iSpindel/issues/180

https://github.com/universam1/iSpindel/issues/122

bertybert, 2018-07-18:
thanks @mr-bridger 
I put a temporary jumper wire between center pin of sensor and d6 and it started working, removed jumper wire, still working. Very odd.
I'll look to get a BAT43 if that's what's needed.

mr-bridger, 2018-07-18:
@cherryphilip74 Hi Just to let you know i calibrated my units today. built 4 and the angles were 25.65, 25.25, 22.31, 25.50. so all pretty great! not sure why one was 3 degs different but still close enough :)

cherryphilip74, 2018-07-18:
@MrBridge
Also mine they are near to base angle at 25...i think is that a great results without using ballast

cherryphilip74, 2018-07-21:
iSpindel PCB assembly instruction, link below:
https://www.dropbox.com/s/5lwl436oonqsmnx/iSpindel%20PCB_assembly.pdf?dl=0

Cheers!

bertybert, 2018-07-21:
I've built 2 so far and got tilts around 21. Managed to use the same calibration formula in both after getting the angles to match in water.

cherryphilip74, 2018-07-21:
Just to be sure, did you verify that the pcb was inserted until the end of the vial(bottom side) ?
My iSpindel are all aligned with a tilt between 24 and 26

[image]
[image]
[image]
[image]
[image]

bertybert, 2018-07-21:
I pushed it down as far as I could. It's possible I could get another mm or so but 21 is close enough and after calibration, both are reading accurately.

Great design. Had a small issue with one board where the gyro board pins piercing the battery sleeve even when cut very short. I decided to instead solder on the top side of the board and leave the battery side flush.

mr-bridger, 2018-07-21:
From my messing about, battery weight can effect slightly, and values lower than 25 mean it's too low in the petling (Assuming you calibrated 100% flat and level. I did this outside of the petling, supporting the board on the 2 edges) 

Great instructions tho @cherryphilip74 :)

cherryphilip74, 2018-07-22:
the weight of the battery can it influence the tilt result (default 25°)
I tried with some batterie of different weight:

- UltroFire battery chinese, 40gr >> TILT 41° (just one sample)
- Samsung battery 26J,Q30, 44gr >>TILT 24° - 26° (6 samples)
- Sanyo battery VTC5,VTC6, 46gr >> TILT 21°- 22°(one for type)

That chinese battery is unbalenced, my suggestion if possible to use of Samsung battery that.

Cheers!
Filippo

fowlerk, 2018-08-01:
@cherryphilip74 I ordered the parts for a build using your BOM and the latest v2.1 board.  I haven’t received all parts yet, but have the Petling, battery, holder, and the PCB’s.  In doing a quick test fit with the Petling, it seems pretty loose without something to hold it in place inside the tube.  Did you use a sled of any design with your build, or is it intended to just lay freely inside the tube?  Nice design on the board BTW!

bertybert, 2018-08-01:
@fowlerk 
With the right perking, the board (with battery and holder) fit very snug. Once pushed to the bottom, they are hard to get out without pliers.

fowlerk, 2018-08-01:
Okay, thanks @bertybert for the quick response.  After doing some more checking, it looks like I ended up with the XXL Petling somehow.  Don’t suppose you could let me know what the inside diameter and length of the correct one is?

sarico88, 2018-08-16:
@cherryphilip74 I’ve just bought al the part and the version 3.1 of your pcb and unfortunately I bought some battery holder that are different then yours and have the usual red and black cable. I notice that at the pcb have 2 hole on the top and bottom of the battery holder are just for holding the battery or are actually the b+ and b- pin? Should I just connect the battery on the b+ and b- on the charger module?  Thanks

cherryphilip74, 2018-08-16:
Hi, see picture in attachment. Positive is near 2 holes (big and small), but the problem is the different height. Before to cut wires, check as the pcb with holder battery fits inside vial.
[image]
[image]

sarico88, 2018-08-16:
Thanks a lot. I will see if is a problem the holder I got. I let you know if I can get it right and manage to build one.

sarico88, 2018-08-16:
Just I cut a bit the corner of the plastic and it fit perfectly maybe drill some hole in the same place of the hole on the pcb and poke the cable in there and solder them. I’ll keep posting when i make progress. 
[image]

tonarii, 2018-08-20:
Hi cherryphilip74, first thanks for your work on this project !
I was looking for a more tiny iSpindel version and i arriverd here.
I would like to know how i can order or print your New iSpindel PCB 3.1 from PCBway ?
Thanks.

bertybert, 2018-08-20:
@tonarii
Just sign up and order the boards.

tonarii, 2018-08-20:
Ok thanks, but there's many option after, i leave everything like it is or i need to do csome hanges ?

dendrophobe, 2018-08-24:
OK I ordered the PCB 3.1 boards from PCBway, but they emailed me with a question... And I'm not sure of the answer. Any help here? This is my first time ordering PCBs, so I'm not so sure of the build; all the other stuff I've ever done (not a lot) has been premade.

"Thanks for your order W221284ASZ1_W10971ASK46_new_TiltClone_3.1.zip, we are making the working file, but we have a question, pls check the image 1 in the attachment, are these two holes plated?

Can we make it in the image 2?"
1:
[image]
2:
[image]

mr-bridger, 2018-08-24:
They do not need to be plated as only anchor points for switch. Would not matter if they are plated tho as long as not connected to anything else.

dendrophobe, 2018-08-24:
Great, thanks! Can't wait to get these things started.

fowlerk, 2018-09-07:
@cherryphilip74 Just got my first iSpindel in operation today using your v3.1 board.  Thanks for your great work and instructions on this!  I was amazed at how easy it was to put together and how well it is working.  As @bertybert mentioned, I did have a similar issue with the pin headers and solder points for the gyro not allowing the battery holder to sit entirely flush, but I did not notice it until after it was fully assembled and it does not seem to affect the operation.  One other minor detail if you happen to make any updates on the board in the future, the support pins for the small slide switch would not fit through the holes in the board, so I ended up having to trim these off.  Maybe increase the dimensions for these if you further revise?  Like I said, just a minor issue.

My initial tilt in clean water was 18.35.  May be a bit too low in the petling, but I tried to match your instructions and picture as closely as possible.  Didn’t seem to matter in the end, as it still calibrated and works fine.  FWIW, I am using the Samsung battery you referenced above.

cherryphilip74, 2018-09-14:
Hi, all resistances are 1/4watt. You can use BAT43 or 330R resistor instead of 470R.
Regarding to battery my suggestion is to use Samsung 30Q.

Cheers!!

CherryPhilip

tonarii, 2018-09-14:
Ok thanks a lot ;)

tonarii, 2018-09-14:
Sorry, last querstion from a noob.
My local shop only have 220K resistor, is it ok or i really need a 230K ?

cherryphilip74, 2018-09-14:
220k it's fine.in fact is a  commercial value.

tonarii, 2018-09-14:
Ok perfect, again thanks :)

tonarii, 2018-09-14:
Ok, so that was not the last question sorry ^^
Regarding the Diode BAT 43, it's polarised (not like a resistor), so i need to put the anode on RST or D0 ?

mr-bridger, 2018-09-18:
That looks wrong way to me. The black and blue PCB tracks are same, just moved slightly the pads for the TP4056

tonarii, 2018-09-18:
Regarding your picture that's what i think too. But i will wait for a confirmation to try to desolder this. But that will be very difficult ! Any idea how to do it correctly ?

fowlerk, 2018-09-18:
@mr-bridger is corrrect, the cathode (ring) should be at the top towards the USB connector with the trace leading to D0.  To desolder, I’d recommend using some solder braid, possibly with some liquid flux, to remove the solder.  After you’ve removed the solder, try heating the diode lead with the iron while gently pulling or prying on the diode with tweezers.  You may have to alternate between the leads while doing so.

tonarii, 2018-09-18:
Ok guys, so i have my confirmation that's the problem :(
Now that will be quite a thing to do that !
Again, thanks.

cherryphilip74, 2018-09-18:
Yes, ring should be at the top towards the USB.

tonarii, 2018-09-18:
You told me the oposite ^_^ 
Nevermind i will try to fix this without destroy everything lol Thanks for your great pcb ! :)

sarico88, 2018-09-18:
Tonarii I did the same but I did soldered the resistor on the side of it and still work so I don’t have to bin a pcb and everything that is soldered in it. Still new in electronics and I still need to master the desoldering technique

fowlerk, 2018-09-18:
Don’t be intimidated by desoldering...it’s really not any more difficult than soldering, just requires patience and a little care with the right tools.  I’m certainly no expert at soldering and have had to tackle this to correct some of my mistakes in the past a few times.  Take it slow, don’t leave the heat on the pad too long at a time, don’t move your iron around much on the pad or you could end up lifting it.

[141 more replies]

closed 2020-08-09 by stale[bot]