universam1/iSpindel #156: excel calibration

onoEsp1, 2018-03-04:
Hi!

im strugling with my excel. I have enabled macros, but stil get error 400 when trying to pres the "formel update" Its just the macro that doesent work. The angel and Plato/SG entered makes a nice curve.
Im wondering can I use the y= line that is in the graph window, and just change x2 to (tilt)*(tilt)?
Or do the macro change more in the formel?

Thanks from

Espen

universam1, 2018-03-05:
Why don't you use the online tool?

onoEsp1, 2018-03-05:
I did, and it seems to work ;) 
But with the online tool I had to change something in the formula. And I was unsure if I did i right. And thought the excel sheet would give me the correct format without having to change something.
Im not so in to Polynominal, but in all examples I have seen the formel looks shorter than mine.
And when I enter my formel in to setup in Portal, it cuts out the last bit of it when I exit setup and enter setup again.
This is my 2 y degree formel with my edits, does it seem right?

0.000071874*tilt^2+-0.005042129*tilt+1.090405889

Best from

Espen

riisefrutti, 2018-03-17:
Would try: 0.000071874*(tilt)*(tilt)+-0.005042129*(tilt)+1.090405889

closed 2018-03-26 by universam1