universam1/iSpindel #174: Still no com port - second kit not working

SimonHanson, 2018-04-09:
This is the second kit i have had put together professionally for me.
First ones there was no com port showing in device manager?
Second one it showed a com3  for a very brief amount of time and it them never re-appeared 
Any advice on helping me to get it sorted

universam1, 2018-04-09:
check your usb cable, is it only a charging one?
Try another computer, on Windows. Driver installed?
Did you charge the battery fully?

SimonHanson, 2018-04-09:
I have USB cable attached to both ports
Witch position should the switch be in
      one with blue light on red flashed very low
      one with blue and red on
Tried i win 10 and win 7 computer
Driver is installed
How long does it take to charge the battery?

JensW2000, 2018-04-09:
Blue AND Red together means that you've soldered somewhere a short circuit or either the Wemos or the GY-521 is broken.

SimonHanson, 2018-04-09:
Any more advice?

JensW2000, 2018-04-09:
Not sure what kind of advice you are looking for Simon.
Take a good magnifying glass and search all solder joints for short circuits. The 3 DS18B20 pins are often a candidate for this.
If can't find any short circurits, then solder down the GY-521. If the battery can then be charged and the firmware can be flashed, the GY-521 is defective. Otherwise the Wemos is broken (or maybe both - Wemos and GY-521).

closed 2018-07-12 by universam1