universam1/iSpindel #177: Wifi configuration Issues

naturalgeek, 2018-04-20:
Hai!

FW 5.8.5
2.2.1(cfd48f3)

wifi named "MAKE ARPANET  1337   AGAIN!" (mind the whitespaces) without any password:

*WM: Sent wifi save page
*WM: Connecting to new AP
*WM: Connecting wifi with new parameters...
*WM: previous settings invalidated
*WM: After waiting...
*WM: 0.00
*WM: seconds
*WM: Connection result:
*WM: WL_IDLE_STATUS
*WM: Failed to connect.

Same Problem for an SSID named like "____________________________" but with an passphrase

naturalgeek, 2018-04-21:
Second Issue seems to be related the length of the SSID.
If the SSID is 32 chars long, on one side the "wifi finder" adds a whitespace and some random character at the end (which changes for every scan). However even if the SSID is entered correctly, the connection fails. As soon as i remove two of the underscore characters in the SSID configured on the router (string lenght now 30) and of course changing on the iSpindel too. it works fine.

universam1, 2018-04-26:
Please try with this test firmware and let me know if it changes the behavior. I extended the buffer for input length 
https://github.com/universam1/iSpindel/files/1929150/firmware.zip

commit 417a49ef (2018-05-13): allow longer SSID and Polynom fields
fix #177 #172
commit 13216b38 (2018-05-13): allow longer SSID and Polynom fields
fix #177 #172
closed 2018-05-13 by universam1