universam1/iSpindel #180: Temp probe not working

GasGasMan74, 2018-06-02:
Good evening,

I try desperately to make the temp probe working but it is always showing me -127°C and I don't know why, I soldered correctly all the components but I did not remove the diodes.

Could you help me ?

Sincerely

Romain

3DMech, 2018-06-15:
Hello Romain,
hopefully the Issue was fixed by replacing the temp-Probe.
Best,
Tjorben
(3D-Mechatronics)

mr-bridger, 2018-07-12:
i saw this on an ispindel i made recently, constantly showed -127. after resetting a few times, and trying to clear settings (for some reason it still wouldn't remove the SSID / password etc) it finally showed the correct temperature. so the ds18b20 obviously isnt faulty. maybe a bug?

universam1, 2018-07-12:
This is clearly a hardware issue. This value expresses there is no communication to the temp sensor.

closed 2018-07-12 by universam1