universam1/iSpindel #181: Mobile web server

Nickolaubr, 2018-06-24:
When connected through iphone, the web server does not allow to swap modes. what happened to me was I was testing my craftbeer pi and when I tried to come back to ubidots mode it stucks.

Only worked when I used my Mac.

Nickolaubr, 2018-06-25:
> You need to delete the WiFi connection with iSpndel on your iPhone in
> order to convince iOS to follow the captive portal.
> It’s not a firmware issue but an iOS one.

The portal opens, but when swaping from CrafbeerPI to Ubidots, the field to
insert the token is frozen with Server address (from craftbeerpi).

I've even tried to input the token, but it didnt work. Only worked when
connected through MAC. It then, when changed the selection to UBIDOTS,
changed automatically to TOKEN.

universam1, 2018-10-18:
No activity here going to close

closed 2018-10-18 by universam1