universam1/iSpindel #183: Temperature not transmitting to ubidots

mr-bridger, 2018-07-18:
Not sure whats going on here, in the console the unit is reading temperture fine. but when in logging mode into ubidots i just get the same constant temperature. looks like the last value seen in the console. 

see ubidots temperature page, 3 different temps, where it changes was where i reopened console locally to check temp was working, and then press reset to go back to logging. also 2 pics of the console showing temp changing ok. 

Any ideas? all other reading logging / changing fine in ubidots. 

[image]

[image]

[image]

universam1, 2018-08-08:
Post a sample of log output from the iSpindel while transmitting to Ubidots. That will give insight if the variables are transmitted and accepted.

quentinchap, 2018-08-09:
For now I have some trouble with my installation I can't get any log by USB.
I just try and I have the same result with a HTTP request. I have the first measure and after it's always the same temperature. All other variable change. 

It's like a cache on temperature or something like this. If I'm connected by the web interface of the Ispindel all is okay...

quentinchap, 2018-08-14:
I have some logs and there are a différence between recorded temperature and data sended

```
FW 5.9.0
2.2.1(cfd48f3)
Worker run!
Boot-Mode: 5
mounting FS... mounted!
reading config file
opened config file

parsed json
applying offsets
parsed config:
{"Name":"iSpindel000","Token":"xxxxxx","Sleep":10,"Server":"192.168.1.11","API":0,"Port":60871,"URL":"/api","DB":"iSpindel","Username":"","Password":"","Job":"ispindel","Instance":"000","Vfact":191.8,"TS":0,"SSID":"Bbox-90980D16","PSK":"C2544A6661953D299AD91343C55ED2","POLY":"-0.00031*tilt^2+0.557*tilt-14.054","aX":-3370,"aY":-841,"aZ":5376}
woken from deepsleep, normal mode
Spl 0: 66.59
Spl 1: 66.63
Spl 2: 66.60
Spl 3: 66.57
Spl 4: 66.57
Spl 5: 66.58
Spl 6: 66.58

a: -520	32767	-14188	absTilt: 66.58	T: 29.68	V: 5.34	owT: 28.25	Gravity: 21.66
After waiting 0ms, result 3
IP: 192.168.1.86

calling Ubidots
{"tilt":66.58323,"temperature":28.25,"battery":5.338894,"gravity":21.65853,"interval":10,"RSSI":-41}Sender: Ubidots posting
POST /api/v1.6/devices/iSpindel000?token=xxxxxxxx HTTP/1.1
Host: things.ubidots.com
User-Agent: ESP8266
Connection: close
Content-Type: application/json
Content-Length: 100

POST /api/v1.6/devices/iSpindel000?token=xxxxxx HTTP/1.1
Host: things.ubidots.com
User-Agent: ESP8266
Connection: close
Content-Type: application/json
Content-Length: 100

HTTP/1.1 200 OK
Server: nginx
Date: Tue, 14 Aug 2018 18:57:05 GMT
Content-Type: application/json
Transfer-Encoding: chunked
Connection: close
Vary: Accept-Encoding
Vary: Cookie
Allow: GET, POST, HEAD, OPTIONS

d1
{"temperature": [{"status_code": 201}], "battery": [{"status_code": 201}], "tilt": [{"status_code": 201}], "interval": [{"status_code": 201}], "gravity": [{"status_code": 201}], "rssi": [{"status_code": 201}]}
0

Final-sleep: 10s; RT:2994
```

universam1, 2018-08-16:
fixed in 5.9.1

closed 2018-08-16 by universam1

mr-bridger, 2018-08-24:
Retrieved my ispindel from beer 3 days ago, updater to 5.9.1 and put in fresh batch today. Temp issue still there. No change since turning on.

universam1, 2018-08-24:
Ok, divided the path for the sensors in FW v6.0.1, cannot happen any more. Please try

quentinchap, 2018-08-24:
It's seem's to be ok on my side i'll compare measure with an other devices during some hours but I think it's ok. Thank you :)