universam1/iSpindel #184: Temperature is not sending to TCP Server

svenp, 2018-07-27:
Hi,
the TCP Server shows Temperature in Charts but it is the tilt value not Temperature . In Spindle Info the Temperature is shown correct
Affter a look in Database i see tilt values in Database temperature field.

So latest iSpindle Firmware send wrong value to TCP server, or Dataformat between TCP Server and Spindel does not match.
Best Regards Sven

svenp, 2018-08-03:
Hi,
look like a timing Problem with DS18B20
When i look at Info there is correct Temperature, maby DS18B20 has more delay time when creating website.
I changed code here delay time from 100 to 200 
`  if (block)
      {
        CONSOLELN(F("OW Retry"));
        initDS18B20();
        delay(OWinterval + 200);  //100 -> 200
        t = getTemperature(false);
      }`
now i get correct Temperature in TCP Server. 
Best Regards
Sven

svenp, 2018-08-03:
Ok i changed back from 200 to 100 and make some changes in Globals.h

`#define OWinterval (950 / (1 << (12 - RESOLUTION))) //800ms is to short for some Sensor changed to 950ms`

universam1, 2018-08-16:
thats apparently an issue with the server, please open a case there

closed 2018-08-16 by universam1