universam1/iSpindel #185: deep sleep even with a BAT43

craboune, 2018-08-02:
Hi everybody,

I encounter some issues with my ispindel even if I have replaced the 390 ohms resistance by a diode BAT43.
I can communicate with my ispindel and also load a firmware without difficulties.
I use Ubidots server to send my data. I bought component on the website 3D-mechatronics.de

When I finished to configure my ispindel on my wifi lan, the blue led stops blinking and that all. I have yet set a period of 15s but nothing happened. I have only one data point on ubidots ...

could you help me please ?

crab a lost french brewer ;-)))

thank a lot

here is my log (PuTTY)

 FW 5.8.6
2.2.1(cfd48f3)
Worker run!
Boot-Mode: 5
mounting FS... mounted!
reading config file
opened config file

parsed json
applying offsets
parsed config:
{"Name":"Red","Token":"A1E-0xIdXsxiew7Jzeaxxxxxxxxxxxxxxxxxxx","Sleep":15,"Server":"                                                                                                                                                             Monitor.beer","API":0,"Port":80,"URL":"/ispindel/?craboune","DB":"ispindel","Job                                                                                                                                                             ":"ispindel","Instance":"000","Vfact":191.8,"TS":0,"SSID":"Appart","PSK":"xxxxxxxxxxxxxxxxx","POLY":"2*tilt","aX":-3387,"aY":-257,"aZ":1209}
                                                               woken from deepsl                                                                                                                                                             eep, normal mode

WARNING: low Battery
Spl 0: 48.12
Spl 1: 50.84
Spl 2: 53.38
Spl 3: 58.56
Spl 4: 62.73
Spl 5: 65.44
Spl 6: 68.29

a: -1756        -15470  5992    absTilt: 58.22  T: 32.32        V: 2.94 owT: 31.                                                                                                                                                             37      Gravity: 116.44
After waiting 3001ms, result 3
IP: 192.168.0.30

calling Ubidots
{"tilt":58.22008,"temperature":31.375,"battery":2.940563,"gravity":116.4402,"int                                                                                                                                                             erval":15,"RSSI":-67}Sender: Ubidots posting
POST /api/v1.6/devices/Red?token=A1E-0xIdXsxiew7Jzeaxxxxxxxxxxxxxxxxxxxxxxx HTTP/1.1
Host: things.ubidots.com
User-Agent: ESP8266
Connection: close
Content-Type: application/json
Content-Length: 101

POST /api/v1.6/devices/Red?token=A1E-0xIdXsxiew7Jzeaxxxxxxxxxxxxxxxxxxxxxxxx HTTP/1.1
Host: things.ubidots.com
User-Agent: ESP8266
Connection: close
Content-Type: application/json
Content-Length: 101

HTTP/1.1 200 OK
Server: nginx
Date: Thu, 02 Aug 2018 21:36:42 GMT
Content-Type: application/json
Transfer-Encoding: chunked
Connection: close
Vary: Accept-Encoding
Vary: Cookie
Allow: GET, POST, HEAD, OPTIONS

d1
{"temperature": [{"status_code": 201}], "battery": [{"status_code": 201}], "tilt                                                                                                                                                             ": [{"status_code": 201}], "interval": [{"status_code": 201}], "gravity": [{"sta                                                                                                                                                             tus_code": 201}], "rssi": [{"status_code": 201}]}
0

WARNING: low Battery

Final-sleep: 3600s; RT:5861

universam1, 2018-08-04:
Look at the two last lines, you are falling into emergency mode because of voltage issue. Fix that and all will work.

closed 2018-08-04 by universam1

craboune, 2018-08-05:
hi, 
thanks for your reply. I saw the warning " low battery" but the battery was pretty full or something like that but not discharged. It is a Samsung 18650 - 3,6V - 3500mAh

the voltage was about 2.9V during the test (seen on ubidots dashboard)

do you think the issue still provides from the battery ?

thank a lot

universam1, 2018-08-05:
Yes, you need to correct the voltage factor

craboune, 2018-08-05:
Ok, thank you very much because it seems working well now ! 
I completely recharged the battery and the Ispindel is now sending datas. That's great !

may I ask you a last question please ? 

the voltage factor is the same as "battery conversion factor" I suppose that I found in the configuration setup of Ispindel ? So how can I modify / correct the voltage factor ? do I need to measure something (battery voltage, ...)

many thanks !