universam1/iSpindel #192: iSpindel freezes after some hours in use

jmelhus, 2018-08-09:
Hi,

I'm using the latest version of iSpindel (5.8.6), together with the recommended components (BAT diode). After a while the blue led on the wemos is steady (no blinking), and it does not report any more data. I've attached my log. It's working fine, until it can't wake up anymore, just reporting gibberish data... If I press the wemos reset button it is working fine again...

Any ideas which may point me in the right direction to solve this issue?

PS! "***" is filled by me to hide sensitive data...

LOG:
Final-sleep: 15s; RT:6224
rll��|�d�|�l�c|����{�b�b��oo�lgg���cp��l{d{dp�o��d��cg�|d��c��nn�$��l`�ggd`o{���gc�l{��gc�l��l�܎pǄd`��g�
FW 5.8.6
2.2.1(cfd48f3)
Worker run!
Boot-Mode: 5
mounting FS... mounted!
reading config file
opened config file

parsed json
offsets not available
parsed config:
{"Name":"iSpindel000","Token":"","Sleep":15,"Server":"***","API":10,"Port":1883,"URL":"","DB":"ispindel","Username":"***","Password":"***","Job":"ispindel","Instance":"000","Vfact":191.8,"TS":0,"SSID":"***","PSK":"***","POLY":"-0.00031*tilt^2+0.557*tilt-14.054","aX":0,"aY":0,"aZ":0}
woken from deepsleep, normal mode
Spl 0: 7.81
Spl 1: 7.83
Spl 2: 7.81
Spl 3: 7.90
Spl 4: 7.85
Spl 5: 7.84
Spl 6: 7.78

a: -1264	1804	16106	absTilt: 7.83	T: 24.01	V: 5.06	owT: 23.81	Gravity: -9.71
After waiting 900ms, result 3
IP: ***

calling MQTT
Attempting MQTT connection
Connected to MQTT
MQTT publish: ispindel/iSpindel000/tilt/7.825686
MQTT publish: ispindel/iSpindel000/temperature/23.8125
MQTT publish: ispindel/iSpindel000/battery/5.057351
MQTT publish: ispindel/iSpindel000/gravity/-9.714078
MQTT publish: ispindel/iSpindel000/interval/15
MQTT publish: ispindel/iSpindel000/RSSI/-80
Closing MQTT connection

Final-sleep: 15s; RT:6516
;ld��|�d�|�d�c|����s�c�c��gg�lgg���cp��lsl{dp�o��l��co�|l��b��oo�l��l`�ggl`ns���o#�ds��'c�l��l���p��d`��g�
FW 5.8.6
2.2.1(cfd48f3)
Worker run!
Boot-Mode: 5
mounting FS... mounted!
reading config file
opened config file

parsed json
offsets not available
parsed config:
{"Name":"iSpindel000","Token":"","Sleep":15,"Server":"***","API":10,"Port":1883,"URL":"","DB":"ispindel","Username":"***","Password":"***","Job":"ispindel","Instance":"000","Vfact":191.8,"TS":0,"SSID":"***","PSK":"***","POLY":"-0.00031*tilt^2+0.557*tilt-14.054","aX":0,"aY":0,"aZ":0}
woken from deepsleep, normal mode
Spl 0: 7.80
Spl 1: 7.82
Spl 2: 7.77
Spl 3: 7.83
Spl 4: 7.75
Spl 5: 7.80
Spl 6: 7.75

a: -1246	1802	16084	absTilt: 7.79	T: 24.04	V: 5.05	owT: 23.75	Gravity: -9.73
After waiting 900ms, result 3
IP: ***

calling MQTT
Attempting MQTT connection
Connected to MQTT
MQTT publish: ispindel/iSpindel000/tilt/7.793439
MQTT publish: ispindel/iSpindel000/temperature/23.75
MQTT publish: ispindel/iSpindel000/battery/5.052137
MQTT publish: ispindel/iSpindel000/gravity/-9.731883
MQTT publish: ispindel/iSpindel000/interval/15
MQTT publish: ispindel/iSpindel000/RSSI/-80
Closing MQTT connection

Final-sleep: 15s; RT:6518
{dd��|�l�|�l�c|����{�c�c��gg�lgg���bp��dslsl��gd�{��oggcl#d`{l�d�cl�sc$cl�rslcl�s���c�c��{�l`��gl�s��oooclcl`sd�d�cl�sclcl�ssdcl�s���c�c��s�l`�....gibberish log continues "forever"...

nppc, 2018-08-16:
You can switch to 74880 baud rate in your serial monitor. Then you will be able to read this "gibberish" output. This is the debug data from your ESP12. This will give a clue for the root of the problem.

jmelhus, 2018-08-20:
Thanks, yes, it says: 
`Fatal exception (0): epc1=0x40100003, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000`

I believe the module itself is broken, I will try to exchange the wemos board today...

universam1, 2018-08-20:
A weak power source on the 3.3V is often the reason for instability of the ESP. You could solder an large cap like 470uF to see if there is a change. Since it is apparently a hardware issue going to close for now

closed 2018-08-20 by universam1