universam1/iSpindel #193: Error sending to Ubidots with v5.9.0, temp_units

kokangit, 2018-08-10:
Hi, first time here.
Updated from 5.8.6 to 5.9.0 and now there are no new data updated in my Ubidots account. Seems to be a problem with "temp_units"
Here is what is printed on console when sending to Ubidots:

calling Ubidots
{"tilt":65.3474,"temperature":29.75,"temp_units":"C","battery":4.607384,"gravity":1.076747,"interval":60,"RSSI":-56}Sender: Ubidots posting
POST /api/v1.6/devices/mguBlue_iSpindel?token=xxxxx HTTP/1.1
Host: things.ubidots.com
User-Agent: ESP8266
Connection: close
Content-Type: application/json
Content-Length: 116

POST /api/v1.6/devices/mguBlue_iSpindel?token=xxxxx HTTP/1.1
Host: things.ubidots.com
User-Agent: ESP8266
Connection: close
Content-Type: application/json
Content-Length: 116

HTTP/1.1 200 OK
Server: nginx
Date: Fri, 10 Aug 2018 14:09:49 GMT
Content-Type: application/json
Transfer-Encoding: chunked
Connection: close
Vary: Cookie
Allow: GET, POST, HEAD, OPTIONS

12d
{"temperature": [{"status_code": 204}], "battery": [{"status_code": 204}], "tilt": [{"status_code": 204}], "interval": [{"status_code": 204}], "gravity": [{"status_code": 204}], "temp_units": [{"status_code": 400, "errors": {"value": ["A valid number is required."]}}], "rssi": [{"status_code": 204}]}
0

Final-sleep: 60s; RT:2991

Scartsjer, 2018-08-13:
same problem here , did a new instal with 5.9.0 and no data was send to ubdots. The device was created with sensors, but no data was send. Flasd back to 5.8.6 and it worked again. exept for "temp_units" but that one was new in 5.9.0?

quentinchap, 2018-08-14:
I have the same issue. Log seems to indicate that problem came form a conversion or somthing like this 
`FW 5.9.0
2.2.1(cfd48f3)
Worker run!
Boot-Mode: 5
mounting FS... mounted!
reading config file
opened config file

parsed json
applying offsets
parsed config:
{"Name":"iSpindel000","Token":"A1E-zMbDFLo2jWRHiT6idLt5lJAIAruXCq","Sleep":10,"Server":"192.168.1.11","API":0,"Port":60871,"URL":"/api","DB":"iSpindel","Username":"","Password":"","Job":"ispindel","Instance":"000","Vfact":191.8,"TS":0,"SSID":"OnePlus3","PSK":"12568974","POLY":"-0.00031*tilt^2+0.557*tilt-14.054","aX":-3370,"aY":-841,"aZ":5376}
woken from deepsleep, normal mode
Spl 0: 85.33
Spl 1: 85.31
Spl 2: 85.32
Spl 3: 85.32
Spl 4: 85.34
Spl 5: 85.34
Spl 6: 85.35

a: -2702	32767	-466	absTilt: 85.33	T: 28.95	V: 5.34	owT: 28.31	Gravity: 31.22
After waiting 1500ms, result 3
IP: 192.168.43.71

calling Ubidots
{"tilt":85.3289,"temperature":28.3125,"temp_units":"C","battery":5.338894,"gravity":31.21708,"interval":10,"RSSI":-21}Sender: Ubidots posting
POST /api/v1.6/devices/iSpindel000?token=A1E-zMbDFLo2jWRHiT6idLt5lJAIAruXCq HTTP/1.1
Host: things.ubidots.com
User-Agent: ESP8266
Connection: close
Content-Type: application/json
Content-Length: 118

POST /api/v1.6/devices/iSpindel000?token=A1E-zMbDFLo2jWRHiT6idLt5lJAIAruXCq HTTP/1.1
Host: things.ubidots.com
User-Agent: ESP8266
Connection: close
Content-Type: application/json
Content-Length: 118

HTTP/1.1 200 OK
Server: nginx
Date: Tue, 14 Aug 2018 11:33:40 GMT
Content-Type: application/json
Transfer-Encoding: chunked
Connection: close
Vary: Accept-Encoding
Vary: Cookie
Allow: GET, POST, HEAD, OPTIONS

12d
{"temperature": [{"status_code": 204}], "battery": [{"status_code": 204}], "tilt": [{"status_code": 204}], "interval": [{"status_code": 204}], "gravity": [{"status_code": 204}], "temp_units": [{"status_code": 400, "errors": {"value": ["A valid number is required."]}}], "rssi": [{"status_code": 204}]}
0

Final-sleep: 10s; RT:3933
`

universam1, 2018-08-14:
@thorrak did you test this?

universam1, 2018-08-16:
fixed in 5.9.1

closed 2018-08-16 by universam1

sonicated, 2018-08-29:
Whilst I can see this has been fixed 5.9.0 is still displaying as "Latest version" and the latter releases are labelled "Pre-release" - could these labels be updated?

I downloaded the latest release rather than a pre-release to avoid bugs but ended up spending hours trying to find out what was wrong!