universam1/iSpindel #197: Manual for configuring influxdb for ispindel

mboy-rnd, 2018-08-21:
Hi, please make a small instruction how to set up influxdb for spindel.

mboy-rnd, 2018-08-21:
I have created a database with the name ispindel in influxdb, but it does not appear the new values. Through tshark it turned out to find the error code:

`Hypertext Transfer Protocol
    HTTP/1.1 400 Bad Request\r\n
        [Expert Info (Chat/Sequence): HTTP/1.1 400 Bad Request\r\n]
            [Message: HTTP/1.1 400 Bad Request\r\n]
            [Severity level: Chat]
            [Group: Sequence]
        Request Version: HTTP/1.1
        Status Code: 400
        Response Phrase: Bad Request
    Content-Type: application/json\r\n
    Request-Id: e833f9a3-a4de-11e8-8045-000000000000\r\n
    X-Influxdb-Build: OSS\r\n
    X-Influxdb-Error: unable to parse 'measurements,source=iSpindel004 tilt=6.143343,temperature=27.25,temp_units=C,battery=4.217935,gravity=0.880556,interval=30,RSSI=-59': invalid boolean\r\n
    X-Influxdb-Version: 1.6.1\r\n
    X-Request-Id: e833f9a3-a4de-11e8-8045-000000000000\r\n
    Date: Tue, 21 Aug 2018 01:09:52 GMT\r\n
    Content-Length: 179\r\n
        [Content length: 179]
    Connection: close\r\n
    \r\n
    [HTTP response 1/1]
    [Time since request: 0.000334721 seconds]
    [Request in frame: 6]
JavaScript Object Notation: application/json
    Object
        Member Key: "error"
            String value: unable to parse 'measurements,source=iSpindel004 tilt=6.143343,temperature=27.25,temp_units=C,battery=4.217935,gravity=0.880556,interval=30,RSSI=-59': invalid boolean`

mboy-rnd, 2018-08-21:
I think I found the problem, if not to screen "C" in temp_units, then error 400 unable to parse

[root@influxdb mboiko]# curl -i -XPOST 'http://localhost:8086/write?db=ispindel&u=admin&p=Pig7ymda0trw' -d 'measurements,source=iSpindel004 tilt=6.174891,temperature=27.375,**temp_units=C**,battery=4.223149,gravity=0.880841,interval=30,RSSI=-65'
HTTP/1.1 400 Bad Request
Content-Type: application/json
Request-Id: 9c486b49-a4e5-11e8-80cf-000000000000
X-Influxdb-Build: OSS
X-Influxdb-Error: unable to parse 'measurements,source=iSpindel004 tilt=6.174891,temperature=27.375,temp_units=C,battery=4.223149,gravity=0.880841,interval=30,RSSI=-65': invalid boolean
X-Influxdb-Version: 1.6.1
X-Request-Id: 9c486b49-a4e5-11e8-80cf-000000000000
Date: Tue, 21 Aug 2018 01:57:51 GMT
Content-Length: 180

{"error":"unable to parse 'measurements,source=iSpindel004 tilt=6.174891,temperature=27.375,temp_units=C,battery=4.223149,gravity=0.880841,interval=30,RSSI=-65': invalid boolean"}

############################################################################

[root@influxdb mboiko]# curl -i -XPOST 'http://localhost:8086/write?db=ispindel&u=admin&p=Pig7ymda0trw' -d 'measurements,source=iSpindel004 tilt=6.174891,temperature=27.375,**temp_units="C"**,battery=4.223149,gravity=0.880841,interval=30,RSSI=-65'
HTTP/1.1 204 No Content
Content-Type: application/json
Request-Id: cf4b15e5-a4e5-11e8-80d2-000000000000
X-Influxdb-Build: OSS
X-Influxdb-Version: 1.6.1
X-Request-Id: cf4b15e5-a4e5-11e8-80d2-000000000000
Date: Tue, 21 Aug 2018 01:59:16 GMT

universam1, 2018-08-21:
you might contact @iceaway who contributed this extension - keeping in mind that iSpindel project is not the place for details about other technologies.

closed 2018-08-22 by mboy-rnd