universam1/iSpindel #200: Battery factor is not saved correctly

FernandoGarcia, 2018-08-30:
Hi!

It's not a network problem as in #198 

Here the console output:

```
Double Reset detected
mounting FS... mounted!
reading config file
opened config file

parsed json
applying offsets
parsed config:
{"Name":"xpto","Token":"gfhghfghgfhfghgfhgfhfghgf","Sleep":60,"Server":"ferduino.com","API":10,"Port":1883,"URL":"","DB":"","Username":"Secret","Password":"Secret","Job":"","Instance":"","Vfact":191.8,"TS":0,"SSID":"Secret","PSK":"Secret","POLY":"-0.00031*tilt^2+0.557*tilt-14.054","aX":-3323,"aY":981,"aZ":273}
going to Config Mode
*WM: Scan done
*WM: Adding parameter
*WM: name
*WM: Adding parameter
*WM: sleep
*WM: Adding parameter
*WM: vfact
*WM: Adding parameter
*WM: 
*WM: Adding parameter
*WM: 
*WM: Adding parameter
*WM: tempscale
*WM: Adding parameter
*WM: 
*WM: Adding parameter
*WM: 
*WM: Adding parameter
*WM: selAPI
*WM: Adding parameter
*WM: token
*WM: Adding parameter
*WM: server
*WM: Adding parameter
*WM: port
*WM: Adding parameter
*WM: url
*WM: Adding parameter
*WM: db
*WM: Adding parameter
*WM: username
*WM: Adding parameter
*WM: password
*WM: Adding parameter
*WM: job
*WM: Adding parameter
*WM: instance
*WM: Adding parameter
*WM: 
*WM: Adding parameter
*WM: POLYN
started Portal
*WM: SET AP
*WM: 
*WM: Configuring access point... 
*WM: iSpindel
*WM: AP IP address: 
*WM: 192.168.4.1
*WM: HTTP server started
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: iSpindel
*WM: Sent iSpindel info page
*WM: iSpindel
*WM: Sent iSpindel info page
*WM: Handle root
*WM: Secret
*WM: -50
*WM: belkin.32fc
*WM: -88
*WM: NET_2GD42504
*WM: -92
*WM: Sent config page
*WM: WiFi save
*WM: Parameter
*WM: name
*WM: xpto
*WM: Parameter
*WM: sleep
*WM: 60
*WM: Parameter
*WM: vfact
*WM: 150
*WM: Parameter
*WM: 
*WM: 
*WM: Parameter
*WM: 
*WM: 
*WM: Parameter
*WM: tempscale
*WM: 0
*WM: Parameter
*WM: 
*WM: 
*WM: Parameter
*WM: 
*WM: 
*WM: Parameter
*WM: selAPI
*WM: 10
*WM: Parameter
*WM: token
*WM: gfhghfghgfhfghgfhgfhfghgf
*WM: Parameter
*WM: server
*WM: ferduino.com
*WM: Parameter
*WM: port
*WM: 1883
*WM: Parameter
*WM: url
*WM: 
*WM: Parameter
*WM: db
*WM: 
*WM: Parameter
*WM: username
*WM: Secret
*WM: Parameter
*WM: password
*WM: Secret
*WM: Parameter
*WM: job
*WM: 
*WM: Parameter
*WM: instance
*WM: 
*WM: Parameter
*WM: 
*WM: 
*WM: Parameter
*WM: POLYN
*WM: -0.00031*tilt^2+0.557*tilt-14.054
*WM: Sent wifi save page
*WM: Connecting to new AP
*WM: Connecting wifi with new parameters...
*WM: previous settings invalidated
*WM: After waiting...
*WM: 3.10
*WM: seconds
*WM: Connection result: 
*WM: WL_CONNECTED
saving config...{"Name":"xpto","Token":"gfhghfghgfhfghgfhgfhfghgf","Sleep":60,"Server":"ferduino.com","API":10,"Port":1883,"URL":"","DB":"","Username":"Secret","Password":"Secret","Job":"","Instance":"","Vfact":191.8,"TS":0,"SSID":"Secret","PSK":"Secret","POLY":"-0.00031*tilt^2+0.557*tilt-14.054","aX":-3323,"aY":981,"aZ":273}saved successfully
Spl 0: 68.16
Spl 1: 68.15
Spl 2: 68.14
Spl 3: 68.18
Spl 4: 68.16
Spl 5: 68.19
Spl 6: 68.16

a: -8136	-15484	258	absTilt: 68.16	T: 28.96	V: 5.23	owT: 28.19	Gravity: 22.47
After waiting 0ms, result 3
IP: 192.168.1.188

calling MQTT
Attempting MQTT connection
Connected to MQTT
MQTT publish: ispindel/xpto/tilt/68.16226
MQTT publish: ispindel/xpto/temperature/28.1875
MQTT publish: ispindel/xpto/temp_units/C
MQTT publish: ispindel/xpto/battery/5.229405
MQTT publish: ispindel/xpto/gravity/22.47209
MQTT publish: ispindel/xpto/interval/1
MQTT publish: ispindel/xpto/RSSI/-63
Closing MQTT connection

Final-sleep: 1s; RT:103237
```
As you can see here the value sent was 150:

> *WM: vfact
> *WM: 150

And the value saved was 191.8 as you can see here:

> saving config...{"Name":"xpto","Token":"gfhghfghgfhfghgfhgfhfghgf","Sleep":60,"Server":"ferduino.com","API":10,"Port":1883,"URL":"","DB":"","Username":"Secret","Password":"Secret","Job":"","Instance":"","Vfact":191.8,"TS":0,"SSID":"Secret","PSK":"Secret","POLY":"-0.00031*tilt^2+0.557*tilt-14.054","aX":-3323,"aY":981,"aZ":273}saved successfully

I haven't any network problem because the MQTT broker is receiving all messages.

Firmware version: 5.9.1

Best regards.

FernandoGarcia, 2018-08-31:
To battery factor be updated the new value should be **>ADCDIVISOR * 0.8** and **< ADCDIVISOR * 1.2** otherwise the value will be changed to default it's  **ADCDIVISOR**.

FernandoGarcia, 2018-08-31:
Now I have it working with **my_vfact** = 246.5 and **ADCDIVISOR** = 251.95

Battery factor > ADCDIVISOR * 0.8
246.5             > 201.56

Battery factor < ADCDIVISOR * 1.2
246.5             < 302.34

I didn't see any information about it here.

[image]

Just for info I'm using a 100K resistor between A0 and Battery + because I haven't 230K resistor at home and due this information: https://arduinodiy.wordpress.com/2016/12/25/monitoring-lipo-battery-voltage-with-wemos-d1-minibattery-shield-and-thingspeak/

closed 2018-08-31 by FernandoGarcia

universam1, 2018-08-31:
Exactly that’s the reason why you need to add a 220k resistor because you are overloading the adc input at 5V now. Just use 2x100k in series if you don’t have them on stock. And the test you found is for a reason.

FernandoGarcia, 2018-08-31:
Hi!

Thanks for your reply.

In fact I didn't the calculation just followed the schematic.

Now I see that max voltage (5.5 V) while charging was not considered on link above. So for a 100 K resistor and 5.5 V charge input the ADC will get about 1.31 V.

While with a fully charged battery (4.2 V) the 100K  resistor is fine because the ADC will get about 1 V.

I think you should add the info bellow on FAQ because if someone makes a wrong calculation will get crazy trying understand why the value doesn't change.

```
displayed voltage / measured voltage * current factor (191.8) = new factor
The new factor should be between current factor more or less 20% (191.8*0.8 - 191.8*1.2).
```

Best regards.

FernandoGarcia, 2018-08-31:
Now I have fixed it I did 100K + 100K +33K. I didn't it before to keep the build more simple and due the convenient info I found above.

The battery factor now is 188.6