universam1/iSpindel #213: iSpindel just send data once

hugodin, 2018-09-13:
Hi. My iSpindel its just sending data to ubidots once it is reseted. After that, it blinks periodically (i putted 20s interval, and its blinking correctly), but no data is sended to ubidots, unless i reset it again. Then, data is sended one time again, and nothing more happen. It just keeps blinking, but not sending data.
According to the log of router, the ispindel, apparently, connects normally.
I assume that its not the problem with the resistor, right?
What should it be?

universam1, 2018-09-13:
It is easy to determine if you capture the log output. 
Set the interval to 5s and trace what happens the following runs. Will give you an error, anything else is wild guessing

hugodin, 2018-09-13:
[image]
This is the log

hugodin, 2018-09-13:
I tried to put another router, and it works fine. It means the problem is the router. Do you have an idea what its blocking the connection on my original router, because I need to use this router.

tonarii, 2018-09-16:
Hi, i have the same issue, i tried with another router but this is still not working.
Maybe it's not the same issue, but i think it is.
Can you help me with that ? Thanks.
[image]

nppc, 2018-09-19:
I had somethig similar. Mine iSpindel was successfully sending the data only after I exit from settings. But when I power off/on, then it refuses to connect.
I made a custom firmware to resolve this issue,
You can try, maybe it will resolve your problem...
[firmware_WIFIconnectionFix.zip](https://github.com/universam1/iSpindel/files/2398041/firmware_WIFIconnectionFix.zip)

hugodin, 2018-09-19:
My iSpindel is working correctly with another router since 6 days ago. But it is a temporary solution, because I will need to find a way to able iSpindel connects with my original router.

tonarii, 2018-09-19:
That indeed would be great, still no idea why it's not working i suppose ?
I noticed that the ip adress is like 192.168.31.112 can that be a problem ? 
Because on my router the adresses goes to 192.168.1.1 to 192.168.1.10 or something like that.

hugodin, 2018-09-19:
Yes. That can be your problem. You might reserve an ip address by your iSpindel MAC address. Try it

tonarii, 2018-09-19:
I tried but that's not working :(
Or i did it wrong, i will ty again^^

hugodin, 2018-09-19:
It depends on your router. Try seeing a tutorial specified for your model on how to reserve IPs

tonarii, 2018-09-19:
How can i check the ip adress that the ispindel use ?

flavioballabio, 2019-02-06:
Hi I have exactly the same problem with ispindel, when I switched on after few powered off days my ispindel is not able to send data due to failed to connect, if I reset it and save the configuration without any changes it works again.

nppc, 2019-02-06:
@flavioballabio Have you tried mine firmware from this answer? https://github.com/universam1/iSpindel/issues/213#issuecomment-422883959
Did it helped?

flavioballabio, 2019-02-06:
@nppc, not yet, I will try it now. Thanks

flavioballabio, 2019-02-06:
@nppc, Hi Pavel I'm testing it with your firmware and seems to works fine, could those changes included into the next release ?

flavioballabio, 2019-02-06:
@nppc, now my ispindel is able to send data but on brewfather only temp data

nppc, 2019-02-06:
that was very old binary. I will compile a fresh one.
So far only for two of us happening this issue with connection failing.
I'm using ESP07 on mine custom board.
Maybe there is some hardware issue...
So, I'm not sure will devs change their firmware if it is working for
99.99% of people.

nppc, 2019-02-06:
Try this one: [firmware604_wifiFix.zip](https://github.com/universam1/iSpindel/files/2837626/firmware604_wifiFix.zip)

flavioballabio, 2019-04-09:
> I am setting New virtual network, with WiFi and wpa encryption, not wpa2.
> Work great. Я настроил на своём роутер виртуальную сеть с ключем и впа
> шифрованием, вместо основновной с wpa2, сейчас проблем нет. Калибрую 2
> iSpindel 1 в основной сети 2 в вертуальной. С первого показания отсутствуют
> бывает по 2 часа, со второго стабильно.

Hi Pavel,

I tried your firmeare 604 and it works fine for me, please send me any new
fresh version.

Many thanks

phastens, 2019-05-07:
Hi everyone,
I have the exact same thing in my side but for TCP. But whatever I use to receive data, I got them once.
I also test with your firmware @nppc but i still have the issue. By the way, what are the modification in code ? Maybe I can take a look with the right direction.
[image]

@universam1 is that possible to reopen this issue because we only have work around for some user.

universam1, 2019-05-09:
You got definitely a hardware issue. 
You could try to short cut the resistor from RST to pin 16 to rule out this possibility. If so reduce its size. Otherwise the esp is probably broken

phastens, 2019-05-09:
I test with a new ESP and new board but I have the exact same error. With or without the resistor for reset. I use a Wemos D1 for this. But it's weird to have multiple failing devices.
I don't know were to look for this issue...

nppc, 2019-05-09:
> I also test with your firmware @nppc but i still have the issue. By the way, what are the modification in code ? Maybe I can take a look with the right direction.

I've changed they way ESP connects to WiFi after reset.
https://github.com/nppc/iSpindel/commit/dcc04c8605b386b7e0e4edc6d17550cfb02a7c2b
I have no idea why for few boards connection is an issue (as well as for mine board). But interestingly, mine ESP07 correctly detects the reason of reset, so, i have a chance to initialize the Wifi at the first boot.
This resolved mine problem.

phastens, 2019-05-09:
Ok i have change the serial output baudrate to read whats esp send. I get that:
[image]

If I refer to the esp documentation it is a reboot from pin and by using flash right ? This is the normal way ?
So, i don't understand why he is no restarting the program if everything is ok from esp side. It seems like he doesn't load the program at all.

phastens, 2019-05-09:
Ok obviously there A LOT of discussion about this issue.
I think I will order some new esp :(
(Related to https://github.com/universam1/iSpindel/issues/59)

trafim-uakraine, 2019-06-16:
I ordered myself 6 ESP mini, and at all such a problem. not knowing what to do.

jklawiter, 2019-06-23:
I've got the same issue here. If I'll go into config and than save it spindle will send my data to ubidots (or anywhere) and than i'll get failed to connect.

```
*WM: 10.20
*WM: seconds
*WM: Connection result: 
*WM: WL_CONNECTED
saving config...{"Name":"iSpindel000","Token":"A1E-VqOdEYIbTkvFI03YjUcjrJK8dE8v44","Sleep":900,"Server":"","API":0,"Port":80,"URL":"","DB":"ispindel","Username":"","Password":"","Job":"ispindel","Instance":"000","Vfact":195.47,"TS":0,"OWpin":12,"SSID":"xxxxxx","PSK":"xxxxxx","POLY":"-0.00031*tilt^2+0.557*tilt-14.054","aX":0,"aY":0,"aZ":0}saved successfully
Samples:42 min:33.86 max:33.96 time:728
x: 5078 y: 6976 z: 12340
Tilt: 33.90
Tacc: 33.99
Volt: 4.14
Temp: 27.37
Gravity: 4.47
IP: 192.168.1.9

calling Ubidots
{"tilt":33.89917,"temperature":27.375,"battery":4.138742,"gravity":4.471598,"interval":1,"RSSI":-47}
                                                                                                    Sender: Ubidots posting
POST /api/v1.6/devices/iSpindel000?token=A1E-VqOdEYIbTkvFI03YjUcjrJK8dE8v44 HTTP/1.1
Host: things.ubidots.com
User-Agent: ESP8266
Connection: close
Content-Type: application/json
Content-Length: 100

HTTP/1.1 200 OK
Server: nginx
Date: Sun, 23 Jun 2019 21:07:41 GMT
Content-Type: application/json
Transfer-Encoding: chunked
Connection: close
Vary: Accept-Encoding
Vary: Cookie
Allow: GET, POST, HEAD, OPTIONS

d1
{"temperature": [{"status_code": 201}], "battery": [{"status_code": 201}], "tilt": [{"status_code": 201}], "interval": [{"status_code": 201}], "gravity": [{"status_code": 201}], "rssi": [{"status_code": 201}]}
0

Final-sleep: 1s; RT: 111047
{dd��<�d�|�d�c|ǃ��s�c�c��oo�lgn���c8��$sd{$p�o��l��cn�|l��c��gn�$��l`�nol`o{���gc�${��gc�d�d�p����`�g�
                                                                                                      FW 6.0.5
3.0.0-dev(c0f7b44)
Worker run!
mounting FS... mounted!
reading config file
offsets not available
parsed config:
{"Name":"iSpindel000","Token":"A1E-VqOdEYIbTkvFI03YjUcjrJK8dE8v44","Sleep":900,"Server":"","API":0,"Port":80,"URL":"","DB":"ispindel","Username":"","Password":"","Job":"ispindel","Instance":"000","Vfact":195.47,"TS":0,"OWpin":12,"SSID":"xxxxxx","PSK":"xxxxxx","POLY":"-0.00031*tilt^2+0.557*tilt-14.054","aX":0,"aY":0,"aZ":0}
Boot-Mode: Deep-Sleep Wake

woken from deepsleep, normal mode
Samples:42 min:33.83 max:33.92 time:755
x: 5074 y: 6954 z: 12342
Tilt: 33.88
Tacc: 33.85
Volt: 4.14
Temp: 27.44
Gravity: 4.46
After waiting 5104ms, result 6
Rescue Wifi credentials
failed to connect

Final-sleep: 900s; RT: 6401
```
same time on other AP all is working fine. I did start mobile AP from my phone, works great. 
So it is something between iSpindle and my router (tplink TL-WR340G). 
Is there any list of rouiters that are working correct? I'm going to change my router anyway, but i will not change it twice so… how can I be sure that all will work? Is it any special router/AP ability ispindle like to have? 
I did try it also at work, i've got other model od tplink there … no problem, all is working just fine. So i'm sure that it is all about AP… but how to not buy wrong one? Any help?

nppc, 2019-06-23:
Good catch. Mine router, where I had issues is very old. What about your?
Maybe we can suggest devs to add a wiki page with known models that
working/not working?

jklawiter, 2019-06-23:
it depend on what does it mean "very old" but yes… last firmware available is dated 2011. But can I be sure that all "modern" routers will work?

Coctus, 2019-08-07:
I'm having the same kind of problem but the led is not blinking at all after the first send. I have 330 ohm resistor between D0 and RST. What could cause that? The last nonsense rows after the "Final-sleep: 20s; RT: 4688" row is coming after 20 seconds has past the send but after that nothing happens.

`
{dl▒▒|▒l▒|
▒
d▒c|▒▒▒▒s▒c▒c▒▒gg▒lgn▒▒▒cp▒▒l{d;lp▒o▒
▒
l▒▒
c
g▒|d▒▒c▒▒gg▒l▒▒d`▒o
l`gs▒▒▒o
c
▒ds▒▒oc
▒d▒d▒x▒▒▒`▒o▒
             FW 6.0.6
3.0.0-dev(c0f7b44)
Worker run!
mounting FS... mounted!
reading config file
offsets not available
parsed config:
{"Name":"CryptSpindel","Token":"XXXX","Sleep":20,"Server":"log.brewfather.net","API":0,"Port":80,"URL":"/ispindel?id=XXXX","DB":"ispindel","Username":"","Password":"","Job":"ispindel","Instance":"000","Vfact":191.8,"TS":0,"OWpin":12,"SSID":"XXX","PSK":"XXX","POLY":"-0.00031*tilt^2+0.557*tilt-14.054","aX":0,"aY":0,"aZ":0}
Boot-Mode: Deep-Sleep Wake

woken from deepsleep, normal mode
Samples:42 min:86.75 max:87.49 time:756
x: -394 y: 15814 z: -588
Tilt: 87.37
Tacc: 29.28
Volt: 4.09
Temp: 27.31
Gravity: 32.25
After waiting 2002ms, result 3
IP: 192.168.0.12

calling Ubidots
{"tilt":87.3716,"temperature":27.3125,"battery":4.092805,"gravity":32.24551,"interval":20,"RSSI":-57}
                                                                                                     Sender: Ubidots posting
POST /api/v1.6/devices/CryptSpindel?token=BBFF-LH8QemaEwPkShkFWBGF1h6DtFXHpoL HTTP/1.1
Host: things.ubidots.com
User-Agent: ESP8266
Connection: close
Content-Type: application/json
Content-Length: 101

HTTP/1.1 200 OK
Server: nginx
Date: Wed, 07 Aug 2019 19:56:23 GMT
Content-Type: application/json
Transfer-Encoding: chunked
Connection: close
Allow: GET, POST, HEAD, OPTIONS
Vary: Origin, Cookie

d1
{"tilt": [{"status_code": 201}], "temperature": [{"status_code": 201}], "battery": [{"status_code": 201}], "gravity": [{"status_code": 201}], "interval": [{"status_code": 201}], "rssi": [{"status_code": 201}]}
0

Final-sleep: 20s; RT: 4688
rld▒▒<▒d▒|▒
l▒
c|▒▒▒▒#▒
c▒▒nn▒$oo▒▒▒
cp▒▒dslsl
`

JonBrews, 2019-11-20:
I Had the same issue regarding the uploading once. 
Having reviewed the soldering/placement of resistors/diode i identified that the Bat43 diode was round the wrong way. Check that the diode is positioned as per the diagrams and re-try uploading.

trafim-uakraine, 2019-11-20:
Check the esp module, it should not be labeled DOIT. AM

mobilars, 2020-08-27:
I have the same problem with 6.3.1. I am able to connect once after config, but not again. The patched firmware above works, but seems to be an old version? Any chance you could make a new with the newest version?

ColdPassionBrewing, 2021-06-16:
> I had somethig similar. Mine iSpindel was successfully sending the data only after I exit from settings. But when I power off/on, then it refuses to connect.
> I made a custom firmware to resolve this [...]

this fixed my issue thanks!

matthewmatt, 2021-06-28:
> this fixed my issue thanks!

It worked for me also!

I was having a similar problem. The problem started suddenly and seems to be affecting multiple previously working iSpindles.

I tried flashing the firmware fix that you sent, and it works, thats great thank you!

The fix uses firmware V6.00. I am just wondering if this update can be incorporated into the latest version firmware?

Thanks,

Matt

universam1, 2021-06-29:
implemented in https://github.com/universam1/iSpindel/releases/tag/7.1.2

matthewmatt, 2021-06-30:
That is great, it seems to be working well so far, thank you so much!