universam1/iSpindel #215: InfluxDB HTTP error 400

SunSun1979, 2018-09-16:
Please change

```
        msg += kv.value.as<String>();
to

        if ( kv.value.is<char*>() ) {
            msg += '"' + kv.value.as<String>() + '"';
        }else{
            msg += kv.value.as<String>();
        }

```
temp_unit=C - wrong Line protocol to InfluxDB
temp_unit="C" - must

stenub, 2018-09-26:
Push! I stumbled over the same error, it costed me several hours to find out. I am using the latest influxdb!

universam1, 2018-09-27:
Feel free to create a MR

stenub, 2018-09-27:
ok, i cloned your repo and changed the code as SunSun1979 suggested. i compiled the code with pio and flashed the firmware file to my iSpindel. Now its working fine with influxdb http. 
How do i create a "MR"? How do i push the changes back to your repo without messing anything up? I did not create a branch but comitted to master.

FernandoGarcia, 2018-09-27:
Hi!

You should make a fork from this repo after make the changes and commit to a new branch (not master) then submit the pull request (PR).

Look at:

https://help.github.com/articles/fork-a-repo/

https://help.github.com/articles/creating-a-pull-request/

Best regards.