universam1/iSpindel #216: Possible Deep sleep issue ?

tonarii, 2018-09-18:
Hi, i'm sorry to post a new issue that might not be a new one, but i don't know if i have a deep sleep issue or if this is from something else.
I used last Cherry Philip PCB (frome here on issue #154) and replaced the 470 resitor by the Bat 43 diode Schottky.
Here is a report when i try to send data to my database.
I would like to know if this can be a soldering problem or something like that.
Thanks.
Regards.
[image]

JohnnyGunn, 2018-09-20:
Hi,

I'm using the same PCB and I'm getting the same issue on the most recent firmware. I can sync to Ubidots, but I'm only getting 1 set of readings, the first ones. No updates to any of the readings after that.

JohnnyGunn, 2018-09-21:
My issue has been resolved. I updated the firmware via the portal, and reduced the interval time to 5 seconds for testing and all is good now.

tonarii, 2018-09-21:
Hi, ok good to know and good for you :)
Me i'm waiting for some news parts to build a brand new device and see if it's ok or not.
I will try the update from the portal.

tonarii, 2018-09-27:
To close, i proceed to a clean and fresh build and now it's working properly :)

closed 2018-09-27 by tonarii

MattCordell, 2018-10-03:
Was about to post my logs for help (having the same issue, same board). However, I used the 330ohm instead of Diode... I'll try update firmware via the portal tonight (Hopefully that works).

tonarii, 2018-10-03:
To me the issue came from a soldering problem, i hope you'll fix your ispindel :)

MattCordell, 2018-10-03:
OK Thanks. Fortunately I purchased 5 of all the parts I needed. Except the Diode... Looks like I'll need to track one of those down. I don't think I'm going to be able to pull the one I've built apart to access the resistors (already done it once, not fun). Sigh.

tonarii, 2018-10-03:
I agree not fun at all ! ^^

simduchaine, 2019-02-27:
I seem to have the same problem. @tonarii what was your soldering problem?