universam1/iSpindel #217: offset calibration result not maintained after power off

lweydert, 2018-09-20:
Hi iSpindel developpers and contributors,

Thanks for the whole project and for its great documentation.

When surfing through the iSpindels configuration websites I came accross the "Offset Calibration" procedure from "Maintenance" screen.
My current assumption is that this "offset calibration" intends to align the accelerometer orientation with the tube direction (for example when during soldering, the user slightly misaligns the accelerometer with the wemos, which happened also in my case).
I preformed the "offset calibration" and in fact the procedure itself went well.
After calibration the new tilt value was close to 90° when lying on a horizontal surface and the new tilt values also seemed okay when tilting the iSpindel .

The problem I am facing now is that after powering off and on the iSpindel, 
the offset calibration result seems to get lost (so not properly saved after/during the "offset calibration" procedure).
To make my current iSpindel work,
I would have to redo this "offset calibration" procedure after every power on.

I flashed the latest firmware 6.0.0 and 
I base my findings based on the tilt values shown in the "iSpindel Info" website.

Hence here my questions:
1) Is this behavior according to the iSpindels intended behavior ? (Should the offset calibration procedure be done at every use after power on)
2) If yes, what is the reason for forcing the user to do it ? I assume that if everything is well fixed and no mechanical changes occured on the iSpindel, this should not be necessary ?

Thanks for your help,
Cheers

universam1, 2018-09-20:
Check the console logs especially about message for loading the offsets. 

Actually after the calibration the iSpindel reboots and you monitor good calibration that is calculated from stored config. Hence your observation is not technically totally possible but the log output may give you the insight. 
Another option is that your sensor is somewhere broken.

lweydert, 2018-09-24:
Dear universam1,

Many thanks for you super fast support !

I checked the console log and indeed offset values were correctly saved.
Also the console log indicated that sensor tests during the offset calibration went fine.
Despite  the following console log:
"reading config file
applying offsets
parsed config:
{"Name":"iSpindelBoufferBouwen","Token":"A1E-qHLPoqFUGeZA84kgxBJPt7yuyxaxCp","Sleep":300,"Server":"","API":0,"Port":80,"URL":"","DB":"ispindel","Username":"","Password":"","Job":"ispindel","Instance":"000","Vfact":191.8,"TS":0,"OWpin":12,"SSID":"0024A5BCBC1E","PSK":"3y4da87g6msbn","POLY":"-0.00031*tilt^2+0.557*tilt-14.054","aX":563,"aY":2040,"aZ":1227}"
I still experienced that the tilt value did not seem to consider the offset calibration results after manually powering off/on.

I also used the X,Y,Z values from console log in order to check whether sensor readings make sense appropriate and they do.

For above reasons, I decided to downgrade firmware version to 5.8.6. and
now my iSpindel works as expected (keeps the calibration result also after manual power off/on).

Thanks for your support.

closed 2018-09-24 by lweydert

the51craw, 2019-05-18:
> Check the console logs especially about message for loading the offsets.
> Actually after the calibration the iSpindel reboots and you monitor good calibration that is calculated from stored config. [...]

Hi universam1, and thanks for this great firmware!

I really hate to contradict you now, on my first contact with you, but 'not technically totally possible' seems to be possible after all. I have found that my iSpindel looses calibration data (but not settings) after power cycle using f/w 6.0.0 or later.  However, calibration data is saved as expected after power cycle using f/w 5.8.6.

There is a difference between reboot and power cycle, and the problem probably lies there. I have analyzed the source code between versions 5.8.6 and 6.0.0, but there are a lot of differences. Unfortunately the two versions in between are not supplied as binaries, and I do not have the platform.io compiler, so I cannot narrow it down further.

I believe that this may be what happens:
After calibration the platform saves the calibration data and reboots.
During reboot, the calibration data fails to be read back for some reason (problem during write or read).
Since the platform has been under power all the time, the calibration therefore remains in RAM.
When power cycling the platform, RAM data is lost, and the calibration data still cannot be read.

However, there was one different that caught my eye. Near the bottom of the source code (line 1023) the following line was removed from version 6.0.0:

delay(100); // workaround for https://github.com/esp8266/Arduino/issues/2750

I checked the issue, and it is still open, so I believe that line should not have been removed.

I would gladly supply log information (and also read the log myself) but I have not been able to figure out how to obtain them from the platform. (I have worked a lot with embedded platforms, so I feel a bit silly for not finding out...)

Hope this helps.

stefschin, 2019-05-20:
Hi, I have same problem with one of mine Ispindels with 6.0.5 firmware version. As I have others without problems I assume that firmware isn't the problem, as say Samuel, I assume it's a hardware problem with sensor... I need more tests to confirm that

universam1, 2019-05-20:
From my experience if the voltage of the battery is under 4V I could encounter issues to update the flash storage properly. 
Turns out that the ESP8266 requires a higher voltage for this write operation than for normal intervals.

Especially this chip is very famous to be picky with power supply or creating various issues if under voltage.
Might be good to check how that actually looks like on these devices. And _always_ take the effort to attach the USB - serial console and monitor whats actually happening. The output should give you perfect insight if the data could be saved and read back later

stefschin, 2019-05-21:
My console monitor shows after calibration shows:

```
Testing device connections...
MPU6050 connection successful
averaging 1000 readings each time
expanding:
	XAccel			YAccel				ZAccel
[0,0] --> [22675,22408]	[0,0] --> [25171,24816]	[0,0] --> [5656,5912]
[-1000,0] --> [13502,22408]	[-1000,0] --> [16975,24816]	[0,0] --> [5654,5912]
[-2000,0] --> [4133,22408]	[-2000,0] --> [8599,24816]	[0,0] --> [5647,5912]
[-3000,0] --> [-5257,22408]	[-3000,0] --> [225,24816]	[0,0] --> [5649,5912]
[-3000,0] --> [-5474,22408]	[-4000,0] --> [-8169,24816]	[0,0] --> [5641,5912]
	XAccel			YAccel				ZAccel
[-3000,0] --> [-5474,22034]	[-4000,0] --> [-8169,24416]	[0,0] --> [5641,5644]
[-3000,0] --> [-5474,22657]	[-4000,0] --> [-8169,25141]	[0,1000] --> [5641,14207]
[-3000,0] --> [-5474,22661]	[-4000,0] --> [-8169,25142]	[0,2000] --> [5641,22945]

closing in:
	XAccel			YAccel				ZAccel
[-3000,-1500] --> [-5474,8894]	[-4000,-2000] --> [-8169,8747]	[1000,2000] --> [14581,22945]
[-3000,-2250] --> [-5474,1694]	[-4000,-3000] --> [-8169,190]	[1000,1500] --> [14581,18674]
[-2625,-2250] --> [-1893,1694]	[-3500,-3000] --> [-4082,190]	[1000,1250] --> [14581,16613]
[-2437,-2250] --> [-238,1694]	[-3250,-3000] --> [-2121,190]	[1125,1250] --> [15491,16613]
[-2437,-2343] --> [-238,666]	[-3125,-3000] --> [-1060,190]	[1187,1250] --> [15992,16613]
	XAccel			YAccel				ZAccel
[-2437,-2390] --> [-238,247]	[-3062,-3000] --> [-516,190]	[1218,1250] --> [16278,16613]
[-2437,-2413] --> [-238,3]	[-3031,-3000] --> [-257,190]	[1218,1234] --> [16278,16422]
[-2425,-2413] --> [-95,3]	[-3015,-3000] --> [-122,190]	[1226,1234] --> [16359,16422]
[-2419,-2413] --> [-43,3]	[-3007,-3000] --> [-55,190]	[1230,1234] --> [16383,16422]
[-2419,-2416] --> [-43,2]	[-3003,-3000] --> [-20,190]	[1230,1232] --> [16383,16401]
averaging 10000 readings each time
	XAccel			YAccel				ZAccel
[-2417,-2416] --> [-17,2]	[-3001,-3000] --> [-7,190]	[1231,1232] --> [16377,16401]
[-2417,-2416] --> [-26,2]	[-3001,-3000] --> [-10,190]	[1231,1232] --> [16364,16401]
-------------- done --------------
a/g:	2	-8	16362
saving config...{"Name":"iSpindel002","Token":"97utjst0v334l8he32pgu89xxx","Sleep":30,"Server":"brew-spy.com","API":3,"Port":80,"URL":"/api/ispindel","DB":"ispindel","Username":"","Password":"","Job":"ispindel","Instance":"000","Vfact":198.1,"TS":0,"OWpin":12,"SSID":"WiFi-Arnet-rxxx","PSK":"xxx","POLY":"-0.00031*tilt^2+0.557*tilt-14.054","aX":-2417,"aY":-3001,"aZ":1231}saved successfully
⸮
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v951aeffa
~ld

FW 6.0.6
3.0.0-dev(c0f7b44)
Worker run!
mounting FS... mounted!
reading config file
applying offsets
parsed config:
{"Name":"iSpindel002","Token":"97utjst0v334l8he32pgu89xxx","Sleep":30,"Server":"brew-spy.com","API":3,"Port":80,"URL":"/api/ispindel","DB":"ispindel","Username":"","Password":"","Job":"ispindel","Instance":"000","Vfact":198.1,"TS":0,"OWpin":12,"SSID":"WiFi-Arnet-rxxx","PSK":"xxx","POLY":"-0.00031*tilt^2+0.557*tilt-14.054","aX":-2417,"aY":-3001,"aZ":1231}
Boot-Mode: Software/System restart

Double Reset detected

going to Config Mode
reboot RFCAL
rd
```

stefschin, 2019-05-21:
And after power ON:

```
s$
FW 6.0.6
3.0.0-dev(c0f7b44)
Worker run!
mounting FS... mounted!
reading config file
applying offsets
parsed config:
{"Name":"iSpindel002","Token":"97utjst0v334l8he32pgu89xxx","Sleep":30,"Server":"brew-spy.com","API":3,"Port":80,"URL":"/api/ispindel","DB":"ispindel","Username":"","Password":"","Job":"ispindel","Instance":"000","Vfact":198.1,"TS":0,"OWpin":12,"SSID":"WiFi-Arnet-rxxx","PSK":"xxx","POLY":"-0.00031*tilt^2+0.557*tilt-14.054","aX":-2417,"aY":-3001,"aZ":1231}
Boot-Mode: Deep-Sleep Wake

woken from deepsleep, normal mode
Samples:42 min:67.74 max:67.85 time:756
x: 7468 y: 17424 z: -4110
Tilt: 67.79
Tacc: 21.79
Volt: 4.01
Temp: 20.62
Gravity: 22.28
IP: 192.168.1.5

calling HTTP
{"name":"iSpindel002","ID":13067955,"token":"97utjst0v334l8he32pgu89xxx","angle":67.79447,"temperature":20.625,"temp_units":"C","battery":4.008077,"gravity":22.28273,"interval":30,"RSSI":-55}HTTPAPI: posting
code: 200
{
  "status" : 200,
  "ts" : "2019-05-20 23:01:10",
  "data" : { }
}

Final-sleep: 30s; RT: 1885
```
Can you traslate me where is the problem that Offset configuration goes back from 90º to 67º aprox?

ErikdBr, 2019-05-21:
I had the exact same problem, offset tilt went back from 89.999 something to 67 after a power off after  calibration and on the next day a power on.
I started the calibration again and laid the Ispindel on the exact same location where I measured the offset before. Outcome was about the same tilt of 89,something. Saved the configuration and power off and on again, after that everything seems fine. Voltage was above 4V and version of the firmware is 6.05

the51craw, 2019-05-21:
So, ErikdBr, if I understand you right, this is a problem that occurs only sometimes, not always nor never?
In that case maybe timing is an issue. Maybe the save operation not always is completed at the re-boot that occurs after calibration.

the51craw, 2019-05-21:
universam1, what you write makes absolute sense. So, there is nothing to do in the firmware unless some re-read check is done with an extra try to write. I do not think that makes much of a difference, it is better so make the recommendation to always fully charge the battery before calibration. Thanks for your info!

ErikdBr, 2019-05-21:
@the51craw It occured after the very first calibration. Next time I will pay more attention to this particular eledged problem, I still have to remove the input diode on the wemos and tweak the exact postion of the Ispindel in the tube. So I have to do the calibration over again anyway.

ErikdBr, 2019-05-22:
This is defenitely a problem. Yesterday I turned my Ispindel on and laid it flat on a 100% spirit level surface.
I expect to see a tilt of about 90 but I got 68.

stefschin, 2019-05-22:
Me too, every time I turn my Ispindel on I get 66-68 degrees. With full battery charge, maybe my battery is not working well, I will try with other hardware with same battery.

ErikdBr, 2019-05-22:
@stefschin did you remove the input diode on the wemos? If you didn't, did you by any chance hook up the wemos thru usb while a battery was attached?

stefschin, 2019-05-31:
Yes @ErikdBr , input diode is removed. I change Wemos D1 and problem still there. I changed the rest of hardware and just keep battery, and now is working great. Maybe it's a hardware problem that doesn't affect when it's version 5.8.6... I will continue to track the issue

CBBC2020, 2020-04-18:
Can anyone help me with my problem whereby my ISpindel is failing to connect to Brewfather. I attach Arduino Serial monitor output. TIA.
[image]