universam1/iSpindel #219: battery conversion factor

MattCordell, 2018-10-04:
I see this on the configuration screen, but can't find any details on it.
When should I change the value? How?

FernandoGarcia, 2018-10-04:
https://github.com/universam1/iSpindel/blob/master/docs/FAQ-en.md

MattCordell, 2018-10-09:
Thanks, I saw that page. But it doesn't explain what the battery conversion is/does.
Perhaps if I don't know what it means, I just leave as default? (Though I'd like to know)

vytux-com, 2018-10-09:
It corrects the voltage reading/display.

FernandoGarcia, 2018-10-09:
The voltage displayed can be different according to real resistor value. So this factor is used to make the voltage displayed equivalent to voltage measured with a multimeter.

universam1, 2018-10-18:
Thanks guys for commenting

closed 2018-10-18 by universam1

MattCordell, 2018-10-26:
Thanks. Is it something necessary to do? Are there consequences of not adjusting the value?
And from where should I measure the voltage? 
(Apologies in advance for commenting on a closed item)

universam1, 2018-10-26:
Measure the battery and do that little math according to FAQ. It’s not that hard!