universam1/iSpindel #220: Look into supporting brewblox

charliwest, 2018-10-08:
Brewblox is the replacement for brewpi, and will come with a rabbitmq event bus to push readings to and api to allow the ispindel to actually affect the brewblox (how the temp prob normally would).

I know the project isn finished yet, but thought having on the drawing board to add at some point would be good.

https://community.brewpi.com/t/development-update/3317/12 a little info is here on the brewpi forum

NFBrewingTech, 2018-10-08:
I think a limitation is the reporting frequency, from what i have seen. I added support for it to my BrewPi setup and send readings once an hour. To use it as a sensor input the readings would need to be much more frequent for the controls to work.

charliwest, 2018-10-08:
Oh yeah that’s a good point. Would still be good to get the readings into brewblox for graphing :)

NFBrewingTech, 2018-10-08:
Definitely. After I built mine i spent a week working on adding the SG to my BrewPi graph. Same method may still apply - unless someone with programming skill makes it easier anyway.   (Short version, mine feeds data to BrewPi as CSV and in the logging cycle BrewPi pulls this into its DB and chart)

charliwest, 2018-10-08:
Yeah I know your brewpi work well, I was the  guy trying to port it to the non-legacy brewpi software...kind of got it working. But with brewblox on the way I gave up with a few bugs that I had as it would all get thrown out once they release this new software.

bdelbosc, 2019-04-13:
Hi, 
FYI you can graph iSpindel's metrics in BrewBlox using the brewblox-ispindel service: 
https://github.com/bdelbosc/brewblox-ispindel

Cheers

timanrebel, 2019-11-11:
@bdelbosc thank you so much for creating the plugin! Already using it in my new BrewBlox setup

closed 2020-06-23 by stale[bot]