universam1/iSpindel #221: 6.0.0 and 6.0.1 failing on wemos d1 mini v3.0.0

charliwest, 2018-10-08:
Hello all, 

Just trying to set my ispindels up again after a move (from amsterdam to nuremberg :) )

Unfortunately with 6.0.0 or 6.0.1 I cannot get onto the wifi of the units. I have 1 ispindel fully made and was working, and one plain wemos d1 mini board. The same thing happens with both. I click the reset button (some places say once, some say 3-4 times, so I tried all) sometimes I can see the wifi, but cannot connect to it, then the wifi disappears :(

Reflash to 5.9.1 and I can get on the wifi for configuration again

I am not sure how to debug further than this at the moment, but happy to try some step if you let me know how I could help see more details about whats happening here.

Cheers

universam1, 2018-10-09:
first approach is always to open the console, and possible post the output here, then I can help.

charliwest, 2018-10-09:
Not sure about the random characters, if thats normal or not... but this is what I got on boot many times

```
{l l⸮⸮| ⸮l⸮|⸮l⸮#|⸮⸮⸮⸮{⸮c⸮c⸮⸮go⸮$gn⸮⸮⸮#x⸮⸮l;l;lx⸮o⸮⸮l⸮⸮cg⸮|d⸮⸮c⸮⸮g'⸮ d⸮⸮$`⸮gnd`gs⸮ۓob⸮lsۓn#⸮$⸮d`⸮⸮g⸮
FW 6.0.0
2.2.1(cfd48f3)
Worker run!
mounting FS... mounted!
scanning for OW device on pin: 5
No devices found!
scanning for OW device on pin: 12
No devices found!
scanning for OW device on pin: 116
No devices found!
scanning for OW device on pin: 121
No devices found!
ERROR: cannot find a OneWire Temperature Sensor!
Acc Test Connection ERROR!
Boot-Mode: Deep-Sleep Wake
...........
ERROR no Wifi credentials

going to Config Mode
Acc Test Connection ERROR!

Exception (28):
epc1=0x4020e06f epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: sys 
sp: 3fffece0 end: 3fffffb0 offset: 01a0

>>>stack>>>
3fffee80:  00340a23 c160dd2f 3fff2c34 401006dc  
3fffee90:  00000002 00000004 3fff2c34 4020118d  
3fffeea0:  00340a23 3ffef550 3fff2c34 40209b00  
3fffeeb0:  3ffe8e5f 3fffeef0 00000000 00000000  
3fffeec0:  3ffe8e64 3fffeee8 00000000 00000000  
3fffeed0:  00000000 00000000 00000000 402093b8  
3fffeee0:  00340a23 3ffef550 3ffefcd8 4020e97e  
3fffeef0:  00000000 00000000 3ffefb50 40209254  
3fffef00:  00340a23 3ffef550 3fff1eb4 40209b8c  
3fffef10:  00000000 001006f4 4022af0b 4021cd26  
3fffef20:  40240ed4 3fff1eb4 3ffef550 4021d05e  
3fffef30:  60000600 3fff1eb4 00000000 40240ee1  
3fffef40:  40240f26 3fffdab0 00000000 3fffdcb0  
3fffef50:  3ffef568 3fffdab0 00000000 00000004  
3fffef60:  40000f49 40000f49 3fffdab0 40000f49  
3fffef70:  40000e19 40001878 00000002 00000000  
3fffef80:  3fffff10 aa55aa55 000000f9 4010441c  
3fffef90:  40104422 00000002 00000000 be32280e  
3fffefa0:  4010000d 5eab61ea 89a60ab9 55f8ef17  
3fffefb0:  40100764 3fffef4c 40100705 3ffffb58  
3fffefc0:  3fffffd0 00000000 00000000 feefeffe  
3fffefd0:  feefeffe feefeffe feefeffe feefeffe  
3fffefe0:  feefeffe feefeffe feefeffe feefeffe  
3fffeff0:  feefeffe feefeffe feefeffe feefeffe  
3ffff000:  feefeffe feefeffe feefeffe feefeffe  
3ffff010:  feefeffe feefeffe feefeffe feefeffe  
3ffff020:  feefeffe feefeffe feefeffe feefeffe  
3ffff030:  feefeffe feefeffe feefeffe feefeffe  
3ffff040:  feefeffe feefeffe feefeffe feefeffe  
3ffff050:  feefeffe feefeffe feefeffe feefeffe  
3ffff060:  feefeffe feefeffe feefeffe feefeffe  
3ffff070:  feefeffe feefeffe feefeffe feefeffe  
3ffff080:  feefeffe feefeffe feefeffe feefeffe  
3ffff090:  feefeffe feefeffe feefeffe feefeffe  
3ffff0a0:  feefeffe feefeffe feefeffe feefeffe  
3ffff0b0:  feefeffe feefeffe feefeffe feefeffe  
3ffff0c0:  feefeffe feefeffe feefeffe feefeffe  
3ffff0d0:  feefeffe feefeffe feefeffe feefeffe  
[... 245 lines cut ...]
v00000000
~ld

FW 6.0.0
2.2.1(cfd48f3)
Worker run!
mounting FS... mounted!
scanning for OW device on pin: 5
No devices found!
scanning for OW device on pin: 12
No devices found!
scanning for OW device on pin: 116
No devices found!
scanning for OW device on pin: 121
No devices found!
ERROR: cannot find a OneWire Temperature Sensor!
Acc Test Connection ERROR!
Boot-Mode: Exception
...........
ERROR no Wifi credentials

going to Config Mode
reboot RFCAL
{l d⸮⸮| ⸮l⸮<⸮d⸮c<ǃ⸮⸮;⸮#l⸮c⸮⸮'o⸮$gn⸮⸮⸮cp⸮⸮l;l{l⸮;⸮⸮g⸮c⸮'c8 ⸮
```

universam1, 2018-10-09:
So that gives you a clear answer without any doubt. You got issues with your temperature sensor, check everything involved as the resistor, the soldering, the sensor itself

charliwest, 2018-10-09:
This is an empty board, was testing them before building. 
I have now built them up, can test again now everything is connected. Just odd that with nothing connected 5.9.1 was working fine to access confit mode, 6.x.x didnt

closed 2018-10-10 by universam1

charliwest, 2018-10-10:
So by closing this are you saying that the WiFi will not work without a temp sensor?

universam1, 2018-10-10:
That has been mentioned so often, little search could have helped

Scales82, 2018-10-18:
I too had almost the exact same problem on a new build. It was working 100% but after resetting the firmware to factory it introduced this problem. I think the factory reset corrupted the config.
After flashing to 5.8.6 and reconfiguring my wifi, then doing a firmware update via the interface, it fixed it and picked up the sensor again.
So I dont think its a straight up hardware issue. even though it reports no temp sensor found

universam1, 2018-10-18:
After firmware update one needs to save the config another time to allow the firmware to store the discovered sensor. Has that been done?

Scales82, 2018-10-18:
Not yet, I want to keep these ones clean, I made them to give to my family "bare-skinned" as it were.

mr-bridger, 2018-10-18:
Is it possible to remove WiFi AP after it's been entered? I have similarly wanting to remove the WiFi ap and password so the unit is "bare" too, bit as soon as it's entered it seems impossible to remove it. Reset doesn't do it, and even deleting the AP and password then saving config still shows it next time you connect to ispindel.

universam1, 2018-10-18:
Read my comment! You can save bogus credentials if you like but you must save config otherwise the sensor won’t work!

Scales82, 2018-10-18:
Ouch. I did read your comment. VERY closely.
Im sorry if I upset you. It's not explained in the WIKI afaik.