universam1/iSpindel #222: 6.0.x quits after 2-4 days / 5.9.1 shows never changing temp readings

Gkire, 2018-10-08:
Hi there,

All of my ispindels 6.0.x (downloaded 6.0.1, was reported as 6.0.0 in the information menu) started with a full charge on 1 october for a test run on duration. The blue outlined ones dropped out due to reasons unknown relatively quickly (within 2 days) On october 4th all black ones stopped with battery charge remaining around 1800 hours. Brewfather confirmed the server was up and running, not a error on that end. 

Fix as suggested by other users was reverting back to 5.9.1. 6/8 ispindels are now still up and running, so definetely an improvement, however, the temperature reading is not ever chancing. This was not the case in 6.0.x which reported different reading. Please advise on how to debug from here.

First 2 pictures 6.0.x, temperature was fluctuating and reporting real stats.
Quits between 2 and 4 days after commencing.

[image]
[image]

Next 2 pictures 5.9.1 (see temperaturenot changing and reporting discrepancy)
Still works when reset on 5 october (3 days encounting).
[image]
[image]

See that the ispindels are so close a discrepancy should not be possible.
[image]

universam1, 2018-10-09:
first approach is always to open the console, and possible post the output here, then I can help.

Gkire, 2018-10-13:
Please see below my output. This is the ispindel #2 that reports temperatures of 30+ degrees Celsius on 5.9.1. It clearly isn't that temp here indoors. It has worked now for 8 days. After this I will reflash #2 to 6.0.0 and report back with those readings.

Spl 0: 86.18
Spl 1: 86.12
Spl 2: 86.13
Spl 3: 86.16
Spl 4: 86.15
Spl 5: 86.09
Spl 6: 86.10

a: -420 16344   1036    absTilt: 86.13  T: 32.38        V: 3.73 owT: 34.25      Gravity: 31.62
After waiting 0ms, result 3
IP: 192.168.2.20

calling HTTP
{"name":"#2_rood_wit","ID":1358142,"angle":86.13267,"temperature":34.25,"temp_units":"C","battery":3.733055,"gravity":31.62206,"interval":1,"RSSI":-77}HTTPAPI: posting
code: 200
{ "result": "success" }

Final-sleep: 1s; RT:83145
sll▒▒|▒l▒|
▒
l▒
#|▒▒▒▒s▒c▒
c▒▒nn▒lgg▒▒▒cp▒▒l{lrlp▒g▒▒
l
▒▒
c
g▒|d▒
▒#▒▒oo▒l▒▒d`▒nnl`'s▒▒▒o
c▒lr▒▒nc
▒$▒
d`▒'▒
     FW 5.9.1
2.2.1(cfd48f3)
Worker run!
Boot-Mode: 5
mounting FS... mounted!
reading config file
opened config file

parsed json
applying offsets
parsed config:
{"Name":"#2_rood_wit","Token":"","Sleep":900,"Server":"log.brewfather.net","API":3,"Port":80,"URL":"/ispindel?id=","DB":"ispindel","Username":"","Password":"","Job":"ispindel","Instance":"000","Vfact":191.8,"TS":0,"SSID":"Nummer16a","PSK":"Redacted!","POLY":"-0.00031*tilt^2+0.557*tilt-14.054","aX":406,"aY":499,"aZ":675}
                                                                                                                                   woken from deepsleep, normal mode
Spl 0: 86.11
Spl 1: 86.08
Spl 2: 86.10
Spl 3: 86.09
Spl 4: 86.08
Spl 5: 86.13
Spl 6: 86.14

a: -400 16348   1030    absTilt: 86.10  T: 32.29        V: 3.73 owT: 34.25      Gravity: 31.61
After waiting 3502ms, result 3
IP: 192.168.2.20

calling HTTP
{"name":"#2_rood_wit","ID":1358142,"angle":86.10352,"temperature":34.25,"temp_units":"C","battery":3.733055,"gravity":31.60738,"interval":900,"RSSI":-71}HTTPAPI: posting
code: 200
{ "result": "ignored" }

Final-sleep: 900s; RT:5734
rll▒▒|▒l▒|
▒d▒
▒c▒c▒▒gg▒doo▒▒▒
#p▒▒l{d{$p▒o▒
▒d
▒▒
c
o▒|d▒▒c▒▒no▒l▒▒l`▒gd`gr▒▒▒n
b
▒l{▒▒g
c▒d▒
l`▒g▒
     FW 5.9.1
2.2.1(cfd48f3)
Worker run!
Boot-Mode: 5
mounting FS... mounted!
reading config file
opened config file

parsed json
applying offsets
parsed config:
{"Name":"#2_rood_wit","Token":"","Sleep":900,"Server":"log.brewfather.net","API":3,"Port":80,"URL":"/ispindel?id=","DB":"ispindel","Username":"","Password":"","Job":"ispindel","Instance":"000","Vfact":191.8,"TS":0,"SSID":"Nummer16a","PSK":"Redacted!","POLY":"-0.00031*tilt^2+0.557*tilt-14.054","aX":406,"aY":499,"aZ":675}
                                                                                                                                   woken from deepsleep, normal mode
Spl 0: 86.15
Spl 1: 86.18
Spl 2: 86.23
Spl 3: 86.19
Spl 4: 86.20
Spl 5: 86.19
Spl 6: 86.22

a: -294 16456   1050    absTilt: 86.19  T: 32.02        V: 3.80 owT: 34.25      Gravity: 31.65
After waiting 2601ms, result 3
IP: 192.168.2.20

calling HTTP
{"name":"#2_rood_wit","ID":1358142,"angle":86.19244,"temperature":34.25,"temp_units":"C","battery":3.800834,"gravity":31.65216,"interval":900,"RSSI":-76}HTTPAPI: posting
code: 200
{ "result": "success" }

Final-sleep: 900s; RT:5373
;l$▒▒|▒$▒|
▒d▒c|▒▒▒▒c▒b▒▒gg▒dno▒▒▒
cp▒▒dsd{d8▒o▒
▒d▒▒
bo▒|l▒▒c▒▒gn▒$▒▒l`▒g
d`o;▒▒▒nc
▒${▒▒o
c▒l▒
l`▒o▒
     FW 5.9.1
2.2.1(cfd48f3)
Worker run!
Boot-Mode: 5
mounting FS... mounted!
reading config file
opened config file

parsed json
applying offsets
parsed config:
{"Name":"#2_rood_wit","Token":"","Sleep":900,"Server":"log.brewfather.net","API":3,"Port":80,"URL":"/ispindel?id=","DB":"ispindel","Username":"","Password":"","Job":"ispindel","Instance":"000","Vfact":191.8,"TS":0,"SSID":"Nummer16a","PSK":"Redacted!","POLY":"-0.00031*tilt^2+0.557*tilt-14.054","aX":406,"aY":499,"aZ":675}
                                                                                                                                   woken from deepsleep, normal mode
Spl 0: 86.15
Spl 1: 86.25
Spl 2: 86.19
Spl 3: 86.20
Spl 4: 86.19
Spl 5: 86.20
Spl 6: 86.16

a: -276 16476   1072    absTilt: 86.19  T: 32.10        V: 3.88 owT: 34.25      Gravity: 31.65
After waiting 2601ms, result 3
IP: 192.168.2.20

calling HTTP
{"name":"#2_rood_wit","ID":1358142,"angle":86.19226,"temperature":34.25,"temp_units":"C","battery":3.884254,"gravity":31.65207,"interval":900,"RSSI":-73}HTTPAPI: posting
code: 200
{ "result": "success" }

Final-sleep: 900s; RT:5243

Gkire, 2018-10-13:
Device #3, on version 5.9.1 constantly reports 21.56 as temperature, no changes ever. This firmware continued to provide readings for at least 8 days. Interval was set to 3 seconds for collecting output readings. Normal operation is set at 900seconds

*WM: Connection result:
*WM: WL_CONNECTED
saving config...{"Name":"#3_Zwart_Geel","Token":"","Sleep":5,"Server":"log.brewfather.net","API":3,"Port":80,"URL":"/ispindel?id=RedactedemXZAkzyyu","DB":"ispindel","Username":"","Password":"","Job":"ispindel","Instance":"000","Vfact":191.8,"TS":0,"SSID":"Nummer16a","PSK":"Redacted","POLY":"-0.00031*tilt^2+0.557*tilt-14.054","aX":-1952,"aY":-3707,"aZ":1174}saved successfully
Spl 0: 83.78
Spl 1: 83.84
Spl 2: 83.81
Spl 3: 83.87
Spl 4: 83.72
Spl 5: 83.84
Spl 6: 83.75

a: 1832 16122   152     absTilt: 83.81  T: 27.09        V: 3.39 owT: 21.56      Gravity: 30.45
After waiting 0ms, result 3
IP: 192.168.2.25

calling HTTP
{"name":"#3_Zwart_Geel","ID":6108437,"angle":83.80832,"temperature":21.5625,"temp_units":"C","battery":3.394161,"gravity":30.44984,"interval":1,"RSSI":-65}HTTPAPI: posting
code: 200
{ "result": "ignored" }

Final-sleep: 1s; RT:71650
{dd▒▒|▒d▒|
▒d▒
c|▒▒▒r▒c▒c▒▒'n▒lgg▒▒▒cx▒▒drlsdx▒'▒
▒d▒▒
co▒|d▒
▒c▒▒oo▒l▒▒d`▒o
l o;▒▒▒n
#
▒l{ےgc
▒d▒
$`▒g▒
     FW 5.9.1
2.2.1(cfd48f3)
Worker run!
Boot-Mode: 5
mounting FS... mounted!
reading config file
opened config file

parsed json
applying offsets
parsed config:
{"Name":"#3_Zwart_Geel","Token":"","Sleep":5,"Server":"log.brewfather.net","API":3,"Port":80,"URL":"/ispindel?id=RedactedemXZAkzyyu","DB":"ispindel","Username":"","Password":"","Job":"ispindel","Instance":"000","Vfact":191.8,"TS":0,"SSID":"Nummer16a","PSK":"Redacted","POLY":"-0.00031*tilt^2+0.557*tilt-14.054","aX":-1952,"aY":-3707,"aZ":1174}
                                                                                                                                        woken from deepsleep, normal mode
Spl 0: 83.85
Spl 1: 83.79
Spl 2: 83.76
Spl 3: 83.79
Spl 4: 83.80
Spl 5: 83.77
Spl 6: 83.78

a: 1824 16144   162     absTilt: 83.79  T: 27.01        V: 3.39 owT: 21.56      Gravity: 30.44
After waiting 2501ms, result 3
IP: 192.168.2.25

calling HTTP
{"name":"#3_Zwart_Geel","ID":6108437,"angle":83.78506,"temperature":21.5625,"temp_units":"C","battery":3.394161,"gravity":30.4381,"interval":5,"RSSI":-67}HTTPAPI: posting
code: 200
{ "result": "ignored" }

Final-sleep: 5s; RT:4703
sll▒▒|▒l▒|▒
l▒c|▒▒▒{▒b▒c▒▒gg▒doo▒▒▒bp▒▒l{$rlp▒g▒▒
l▒▒
co▒|l▒▒c▒▒'n▒l▒▒d`▒nol`n{▒ےgc
▒ds▒▒o
[... 393 lines cut ...]
opened config file

parsed json
applying offsets
parsed config:
{"Name":"#3_Zwart_Geel","Token":"","Sleep":5,"Server":"log.brewfather.net","API":3,"Port":80,"URL":"/ispindel?id=RedactedemXZAkzyyu","DB":"ispindel","Username":"","Password":"","Job":"ispindel","Instance":"000","Vfact":191.8,"TS":0,"SSID":"Nummer16a","PSK":"Redacted","POLY":"-0.00031*tilt^2+0.557*tilt-14.054","aX":-1952,"aY":-3707,"aZ":1174}
                                                                                                                                        woken from deepsleep, normal mode
Spl 0: 83.70
Spl 1: 83.75
Spl 2: 83.82
Spl 3: 83.84
Spl 4: 83.82
Spl 5: 83.82
Spl 6: 83.82

a: 1806 16130   178     absTilt: 83.82  T: 26.93        V: 3.39 owT: 21.56      Gravity: 30.46
After waiting 3602ms, result 3
IP: 192.168.2.25

calling HTTP
{"name":"#3_Zwart_Geel","ID":6108437,"angle":83.81992,"temperature":21.5625,"temp_units":"C","battery":3.394161,"gravity":30.4557,"interval":5,"RSSI":-66}HTTPAPI: posting
code: 200
{ "result": "ignored" }

Final-sleep: 5s; RT:5896

Gkire, 2018-10-13:
Device #3, on version 6.0.0 (strange, I downloaded 6.0.1 https://github.com/universam1/iSpindel/releases/download/6.0.1/firmware.bin),  has changing temps. 

As noted before, 6.0.0 seemed to last only a few days before quitting (it did not hit the 8minute mark). I will reset the interval to 900 seconds now and report back when I do not receive new readings any more. 

calling HTTP
{"name":"#3_Zwart_Geel","ID":6108437,"angle":86.13243,"temperature":25.9375,"tem                                         p_units":"C","battery":3.717414,"gravity":31.62194,"interval":5,"RSSI":-70}HTTPA                                         PI: posting
code: 200
{ "result": "success" }

Final-sleep: 5s; RT: 8535
{dd▒▒|▒d▒|
▒d▒
c|▒▒▒r▒c▒c▒▒'n▒lgg▒▒▒cx▒▒drlsdx▒'▒
▒
d▒▒
bo▒|d▒
▒c▒▒oo▒l▒▒d`▒o
l`o;▒▒▒n#
▒l{▒▒gc
▒d▒
$`▒o▒
     FW 6.0.0
2.2.1(cfd48f3)
Worker run!
mounting FS... mounted!
reading config file
applying offsets
parsed config:
{"Name":"#3_Zwart_Geel","Token":"","Sleep":5,"Server":"log.brewfather.net","API"                                         :3,"Port":80,"URL":"/ispindel?id=RedactedemXZAkzyyu","DB":"ispindel","Username":"","                                         Password":"","Job":"ispindel","Instance":"000","Vfact":191.8,"TS":0,"SSID":"Numm                                         er16a","PSK":"Redacted","POLY":"-0.00031*tilt^2+0.557*tilt-14.054","aX":-195                                         2,"aY":-3707,"aZ":1174}
scanning for OW device on pin: 5
No devices found!
scanning for OW device on pin: 12
Found device with ROM = 28 50 C0 77 91 16 2 45
  Chip = DS18B20
  Data = 1 A3 1 55 5 7F 7E 81 66 AF  CRC=AF
  Temperature = 26.19 Celsius,
Boot-Mode: Deep-Sleep Wake

woken from deepsleep, normal mode
Samples:42 min:85.97 max:86.06 time:755
x: 1142 y: 16184 z: 36
Tilt: 86.02
Tacc: 28.23
Volt: 3.73
Temp: 26.31
Gravity: 31.56
After waiting 1300ms, result 3
IP: 192.168.2.25

calling HTTP
{"name":"#3_Zwart_Geel","ID":6108437,"angle":86.01904,"temperature":26.3125,"tem                                         p_units":"C","battery":3.733055,"gravity":31.56483,"interval":5,"RSSI":-68}HTTPA                                         PI: posting
code: 200
{ "result": "ignored" }

Final-sleep: 5s; RT: 4246
{$d▒▒|▒$▒|
▒d▒
c|ǃ▒s▒c▒
c▒▒no▒lgg▒▒▒
cx▒▒dsl{dx▒n▒
▒d
▒▒
c
g▒|d▒
▒c▒▒oo▒l▒▒$`▒g
l`or▒▒▒n
b
▒l{▒▒'c
▒$▒
l`▒o▒
[... 280 lines cut ...]
No devices found!
scanning for OW device on pin: 12
Found device with ROM = 28 50 C0 77 91 16 2 45
  Chip = DS18B20
  Data = 1 C3 1 55 5 7F 7E 81 66 4E  CRC=4E
  Temperature = 28.19 Celsius,
Boot-Mode: Deep-Sleep Wake

woken from deepsleep, normal mode
Samples:42 min:62.73 max:63.66 time:755
x: -12048 y: -11976 z: 310
Tilt: 63.42
Tacc: 28.60
Volt: 3.46
Temp: 28.31
Gravity: 20.03
After waiting 1300ms, result 3
IP: 192.168.2.25

calling HTTP
{"name":"#3_Zwart_Geel","ID":6108437,"angle":63.42239,"temperature":28.3125,"temp_units":"C","battery":3.46194,"gravity":20.02533,"interval":5,"RSSI":-68}HTTPAPI: posting
code: 200
{ "result": "ignored" }

Final-sleep: 5s; RT: 4313

universam1, 2018-10-13:
If you have that low intervals the device will heat up itself a lot. The same is while charging. But that has been discussed already a lot of times. Check the FAQ

closed 2018-10-13 by universam1

Gkire, 2018-10-13:
Please reopen. As you can see on the earlier pictures the issue also occurs with 15 min intervals.

universam1, 2018-10-13:
If so, how is that related to the software? Rules out to the hardware then.

Gkire, 2018-10-13:
6.0.0 has right fluctuating temps (=ok)
6.0.0 stops reporting after 4 days max (=not okay)

5.9.1 has never changing temps (=not ok, not hardware related because hardware works on 6.0.0)
5.9.1 continues reporting after 4 days (=ok)

universam1, 2018-10-14:
5.9 is buggy that’s why I deleted these releases. 

On the 6.x version you can see scanning actions for the DS, means you have hardware issues as it can’t read it.

Gkire, 2018-10-14:
Thank you for your reply. I am not sure I understand though. 6.x results in a working ispindel for me (pin 12 reports a good temp) only the ispindel reports for <4 days.  I understand you are trying to say reverting back is not an option. Would you like me to run a new trial with my 7 spindels on 6.x for duration to see if the problem for me on 6.x persists?

universam1, 2018-10-14:
Exactly. See you have the scanning for DS in your log, means you either did not save your settings out the DS cannot be found reliable

Gkire, 2018-10-14:
Ok. I will update my spindels to 6.x and report back next week if they are still updating. If they discontinue before that time  will notify you and ask where to go from there.

Gkire, 2018-10-18:
Sorry to be back, but all of the ispindels have failed in the last few days on 6.0.1. Not a single one out of 7 reports back to me anymore: please see the statistics below:

#1: **14 oct 15:36:** flashed to 6.0.1: **Last reading 17 okt 21:43** (lasted 3 days on 6.x) last battery state: **3.01** 
#2: **14 oct 15:38** flashed to 6.0.1: **Last reading 15 okt 21:54** (lasted 1 day on 6.x) last battery state: **3.32**
#3: **14 oct 15:56** OTA flashed to 6.0.1: **Last reading 14 okt 17:27** (lasted 0 days on 6.x) last battery state: **2.83**
#4: 1**4 oct 15:42** flashed to 6.0.1: **Last reading 14 okt 21:41**(lasted 0 day on 6.x) last battery state: **3.23**
#5: **14 oct 15:46** flashed to 6.0.1: **Last reading 14 okt 20:20** (lasted 0 day on 6.x)last battery state: **3.27**
#6: **14 oct 15:56** OTA flashed to 6.0.1: **Last reading 15 okt 14:22** (lasted 1 day on 6.x) last battery state: **2.55**
#7  **14 oct 16:03** OTA flashed to 6.0.1: **Last reading 14 okt 16:03**(lasted 0 days on 6.x) last battery state:  **2.88**

Please note how on 5.9.1 (flatlines) the temperature was not accurate, but the ispindels were able to last more than 3 days. 

**I would like the stability of 5.9.1 (prolonged output: multipledays) with the sensor accuracy of 6.0.1.** 

[image]

universam1, 2018-10-18:
Something  is terribly wrong with your setups or your iSpindels, i have never seen battery emptied within 3 days! Normally they last 3 months for me.

universam1, 2018-10-19:
Anyone else aside of Brewfather users having this problem?

universam1, 2018-10-19:
@gkire I like to see the battery graphs

tgangso, 2018-10-19:
I can confirm it is not a Brewfather issue, the data is never recieved on my end.

Gkire, 2018-10-23:
On it! They are fully charged and logging now. Will report back in a few days.

@tgangso, can you make a graph available in the chart for battery indication?

Gkire, 2018-10-23:
Update: number 5 sent  readings this night, and quit again this morning:
[image]

 Putty gave me back the following:
reading config file
applying offsets
parsed config:
{"Name":"#5_Zwart_Wit","Token":"","Sleep":900,"Server":"log.brewfather.net","API":3,"Port":80,"URL":"/ispindel?id=","DB":"ispindel","Username":"","Password":"","Job":"ispindel","Instance":"000","Vfact":191.8,"TS":0,"OWpin":12,"SSID":"Nummer16a","PSK":"","POLY":"-0.00031*tilt^2+0.557*tilt-14.054","aX":-4974,"aY":2820,"aZ":1413}
Boot-Mode: Deep-Sleep Wake

woken from deepsleep, normal mode
Samples:42 min:69.33 max:71.91 time:758
x: 4332 y: -17406 z: 4354
Tilt: 70.94
Tacc: 17.69
Volt: 4.10
Temp: 17.75
Gravity: 23.90
After waiting 6003ms, result 6
Rescue Wifi credentials
failed to connect

Final-sleep: 900s; RT: 7405

Seems that it forgot its password over night. I will reset and then report back.

universam1, 2018-10-24:
you might try the new 6.0.2 using the new SDK

Gkire, 2018-10-27:
Thanks for the 6.0.2. Howeverthe problem seems to persist on occasion:

Current time is 07:00am

[image]

For #3 and #7 I wasn't able to get the putty results, it went straight to config.

Putty on Ispindel #4 reported back on reset::

     FW 6.0.2
3.0.0-dev(c0f7b44)
Worker run!
mounting FS... mounted!
reading config file
applying offsets
parsed config:
{"Name":"#4_Groen_Wit","Token":"","Sleep":900,"Server":"log.brewfather.net","API":3,"Port":80,"URL":"/ispindel?id=","DB":"ispindel","Username":"","Password":"","Job":"ispindel","Instance":"000","Vfact":191.8,"TS":0,"OWpin":12,"SSID":"","PSK":"","POLY":"-0.00031*tilt^2+0.557*tilt-14.054","aX":-1108,"aY":1905,"aZ":1311}
Boot-Mode: Deep-Sleep Wake

woken from deepsleep, normal mode
Samples:42 min:69.89 max:71.18 time:755
x: -4438 y: -16728 z: 4506
Tilt: 70.45
Tacc: 19.42
Volt: 3.85
Temp: 19.12
Gravity: 23.65
After waiting 5986ms, result 6
Rescue Wifi credentials
failed to connect

Final-sleep: 900s; RT: 7278

Gkire, 2018-10-27:
Current time 07:49

The #8 reports:
reading config file
applying offsets
parsed config:
{"Name":"#8_Blauw_Wit","Token":"","Sleep":900,"Server":"log.brewfather.net","API":3,"Port":80,"URL":"/ispindel?id=MnF0emXZAkzyyu","DB":"ispindel","Username":"","Password":"","Job":"ispindel","Instance":"000","Vfact":191.8,"TS":0,"OWpin":12,"SSID":"","PSK":"!","POLY":"-0.00031*tilt^2+0.557*tilt-14.054","aX":406,"aY":499,"aZ":675}
Boot-Mode: Deep-Sleep Wake

woken from deepsleep, normal mode
Samples:42 min:83.85 max:83.94 time:757
x: -16476 y: 1338 z: 1212
Tilt: 83.89
Tacc: 24.84
Volt: 3.86
Temp: 23.87
Gravity: 30.49
After waiting 5102ms, result 6
**Rescue Wifi credentials
failed to connect**

Final-sleep: 900s; RT: 6401

universam1, 2018-10-27:
See the RT: output, anything above 3000ms means you got WiFi issues connecting to your Access Point. 
This makes sense in context of your issue. Monitor the RSSI value you are most probably under the limit, shouldn’t be under -70dBm