universam1/iSpindel #225: WPA2 Not connecting - WPA okay. DD-WRT

Scales82, 2018-10-18:
I'm using a linksys DD-WRT router. none of my ispindels like to connect to my network using wpa2. as soon as i switch to wpa it works fine. Any ideas?

universam1, 2018-10-18:
WPA2 works fine for me, try another access point

Scales82, 2018-10-18:
I am sorry, I only own one....

universam1, 2018-10-24:
you might try the new 6.0.2 using the new SDK

Scales82, 2018-10-29:
I’ll reflash when I am home next and try again (out of country) I’ll let you know.
Does that mean 6.02 has been updated?

brendongough, 2018-12-06:
I've had mixed results with different wifi routers, I restored to using an old Apple Airport Extreme (802.11n version) as my express and Mikrotik didn't reliably work. I found this cross platform on the same device (BrewPiLess). I have not tested this on 6.0.2 yet or the BrewPiLess 3.4 though.

Scales82, 2018-12-17:
I ended up changing my routers firmware back to stock Linksys firmware, connects fine now. Still an outstanding issue though I believe

Scales82, 2018-12-17:
Okay, so an update,
I've gone back to DD-WRT, using the latest beta for my device. (v37961). The problem still exists, but I gave the ispindel ESP a static IP reservation in the DD-WRT DHCP server. Now it connects using WPA2 no problem!
[image]

universam1, 2018-12-17:
Hence it is obviously not an WPA2 issue at all, since this is authentication and not dhcp related.

universam1, 2018-12-19:
There is nothing we can do in this regard since it first of all is not confirmed with other access points, More over that would even be outside of this project using the SDK from Espressif eventually that does the hard lifting. If there is more to add on it feel free, for now I will close this issue since not a bug or so

closed 2018-12-19 by universam1