universam1/iSpindel #228: Strange behaviour

Perimec, 2018-10-22:


Perimec, 2018-10-22:
Hi, 
I'm experience a very strange behaviour of the inspidel. I have installed the lasted available firmware (6.0.1 even if it says 6.0.0 on the information tab). the inspidel is not sending data. It send the data only when pressing the reset button. However, if I reset while connecting the usb cable an using putty, everything work fine. 

I must admit, that because some wrong resistor on the battery reading, the battery is reading the wrong voltage of 4.7 but rather than that all the rest seems fine

thanks in advance for the help

Perimec, 2018-10-22:
and yes confirmed, even removing the usb connection, everything work.

Perimec, 2018-10-22:
PS: I did not removed the diode on the wemos... going to do it asap... but still it should be not the problem for this.

universam1, 2018-10-23:
Set the interval to 10s and let it post several times. Attach these console logs here that should shed some light. 

BTW there is the battery correction factor you need to adjust to reflect your correct voltage, but make sure your battery is charged fully!

Perimec, 2018-10-23:
Thanks :) I will do it later today :)

Perimec, 2018-10-23:
Hi, following the log. 

I have checked again the connection, there was a small problem with the voltage. Fixed it and all the rest seems ok. However still not working propertly. If I plug the usb, it works fine (even after removing the USB), otherwise it doesn't work (send data only when pressing the reset). 

thanks again for the help.

{dd▒▒|▒d▒<
▒
l▒c|▒▒▒{▒b▒
b▒▒og▒dno▒▒▒
cp▒▒$sd{dp▒o▒
▒d
▒▒
c
g▒|d▒
▒c▒▒no▒l▒▒d`▒gn$`'s▒▒▒o
c▒lr▒▒n
#
▒l▒
l`▒n▒
     FW 6.0.0
2.2.1(cfd48f3)
Worker run!
mounting FS... mounted!
reading config file
offsets not available
parsed config:
{"Name":"iSpindel000","Token":"XXXXXXXXXXXXXXX","Sleep":20,"S                                                                                                      erver":"","API":0,"Port":80,"URL":"","DB":"ispindel","Username":"","Password":""                                                                                                      ,"Job":"ispindel","Instance":"000","Vfact":229,"TS":0,"OWpin":12,"SSID":"Telenor                                                                                                      9498vel","PSK":"XXXXXXXXXXXX","POLY":"-0.00031*tilt^2+0.557*tilt-14.054","aX":0                                                                                                      ,"aY":0,"aZ":0}
Boot-Mode: Deep-Sleep Wake

woken from deepsleep, normal mode
Samples:42 min:85.22 max:86.01 time:755
x: 876 y: 16544 z: -966
Tilt: 85.57
Tacc: 25.69
Volt: 4.47
Temp: 21.44
Gravity: 31.34
After waiting 501ms, result 3
IP: 10.0.0.55

calling Ubidots
{"tilt":85.57055,"temperature":21.4375,"battery":4.471616,"gravity":31.33888,"in                                                                                                      terval":20,"RSSI":-63}
                      Sender: Ubidots posting
POST /api/v1.6/devices/iSpindel000?token=XXXXXXXXXXXXXXXXX HTTP                                                                                                      /1.1
Host: things.ubidots.com
User-Agent: ESP8266
Connection: close
Content-Type: application/json
Content-Length: 102

HTTP/1.1 200 OK
Server: nginx
Date: Tue, 23 Oct 2018 19:55:28 GMT
Content-Type: application/json
Transfer-Encoding: chunked
Connection: close
Vary: Accept-Encoding
Vary: Cookie
Allow: GET, POST, HEAD, OPTIONS

d1
{"temperature": [{"status_code": 201}], "battery": [{"status_code": 201}], "tilt                                                                                                      ": [{"status_code": 201}], "interval": [{"status_code": 201}], "gravity": [{"sta                                                                                                      tus_code": 201}], "rssi": [{"status_code": 201}]}
0

Final-sleep: 20s; RT: 2743
sdl▒▒|▒l▒|▒
l▒b|▒▒▒{▒c▒
c▒▒oo▒lgn▒▒▒
cx▒▒dsl{dx▒n▒
[... 377 lines cut ...]
calling Ubidots
{"tilt":85.52521,"temperature":21.4375,"battery":4.471616,"gravity":31.31603,"interval":20,"RSSI":-56}
                                                                                                      Sender: Ubidots posting
POST /api/v1.6/devices/iSpindel000?token=XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX HTTP/1.1
Host: things.ubidots.com
User-Agent: ESP8266
Connection: close
Content-Type: application/json
Content-Length: 102

HTTP/1.1 200 OK
Server: nginx
Date: Tue, 23 Oct 2018 19:58:07 GMT
Content-Type: application/json
Transfer-Encoding: chunked
Connection: close
Vary: Accept-Encoding
Vary: Cookie
Allow: GET, POST, HEAD, OPTIONS

d1
{"temperature": [{"status_code": 201}], "battery": [{"status_code": 201}], "tilt": [{"status_code": 201}], "interval": [{"status_code": 201}], "gravity": [{"status_code": 201}], "rssi": [{"status_code": 201}]}
0

Final-sleep: 20s; RT: 2806

Perimec, 2018-10-26:
Hi, so anything strange that youy can spot?

thanks

Perimec, 2018-10-26:
check the voltage... 4.1V.... charged sometime ago... then I really don't know where it can be the problem.. I shuld check the voltage under load. keep you posted.. but I can use as it is anyway... just need to let it start while connected under USB...

thanks again for this amazing project.

bye

Perimec, 2018-10-26:
bzttery fine also under loading...

rasale12, 2018-12-07:
Hello Perimec
i've the same problem, my two ispindle sends data to craftbeerpi only when I push the reset button. i never use Putty, what i have to do with?

Perimec, 2018-12-10:
just download putty, select the rigth com port and then plug the usb in the wemos. Reastart the womos and wait check the info on the screen

universam1, 2019-02-27:
no activity here, going to close

closed 2019-02-27 by universam1