universam1/iSpindel #229: Poor battery life

petjez, 2018-11-06:
Hi,

i ordered this DIY set: http://www.3d-mechatronics.de/en/ispindel-diy-set-3500mah_649.html. The iSpindel works great, but the battery life is very poor. It takes about 8 hours to charge. I use the latest 6.0.1 firmware which should improve battery life. The battery is 3500mAh. It only runs for 25 times and then the battery is flat. Is have an update interval of 3 hours to safe battery power. But it still only lasts 6 days.

Any thoughts?

Regards,
Patrick

mr-bridger, 2018-11-06:
Are there any LED's on constantly? Did you remove the one off the GY521?

petjez, 2018-11-06:
No LED are on, only when charging. GY521 is removed!

universam1, 2018-11-06:
Any comments why closed out of a sudden? @petjez

petjez, 2018-11-06:
Sorry, by was closed by accident, reopened

universam1, 2018-11-06:
can you paste the battery graph from Ubidots, this is very strange

petjez, 2018-11-09:
A little update
I recharged the battery, i took a bit over 8 hours. I connected to Ubidots and set the interval to 30 seconds. The past days the battery keeps running fine. There are over 2200 measurements made. I will throw the iSpindel back in the fermenter and see what happens.

Patrick

petjez, 2018-11-20:
Thanks for thinking with me. In my case it doesn't matter if i set the interval to 30 seconds or 3 hours. The battery will be drained in 4 days anyway! I try to revert back to firmware 5.8.6 also and throw the iSpindel back in the fermenter on saturday.

Cheers, Patrick

Misenkooo, 2018-11-25:
Did you try 5.8.6 ? I've noticed lot of issues with 6.x release so I reverted back to 5.9.1. For example zero point calibration does not work properly as the results are not read from config and after the device is restarted the tilt is incorrect. I'm on latest 5.9.1 so I wonder if battery is problem here as well. It's funny as version 6 using IRQs so battery life should be 20% better than 5.x, Also 3 degree polynomial formula can't be used because the field size is just to small to accommodate 73 characters formula. I've modified this in firmware together with brewersfriend native support and created pull request to merge it into master branch.

avollkopf, 2018-11-26:
I tried different version but cannot see significant differences between pre-and post 6.0.0. This test is with a spindel that has a 3500 mah battery. I also teid it with another spindel that has a 2600 mah battery. Drainage is higher compared to the larger battery, but also here no significant differences betwenn the versions. wakeup is every 600 seconds in a steel fermenter. RSSI is around 70%. (with 586 and 600 battery was charged and tested immediately after charging. Initial 602 readings were little lower as device got discusonnected for some time (a day or 2) after charging)

[image]

petjez, 2018-11-30:
My iSpindel runs out of battery even when it’s switched off. So i think i made a mistake somewhere, but i don’t know where. Any ideas?

Patrick

avollkopf, 2018-12-05:
my impression is that the 3500 mAh battery showed really good performance in the beginning, but it's perfromance is degrading faster compared to the 2600 mAh battery (at least for me). -> It could be also the battery itself.