universam1/iSpindel #230: What settings do I need to enter to send to Thingspeak IoT?

woody600, 2018-11-13:
I can't figure out what I need to enter with configuration to send data to Thingspeak IoT?
- Service type: ?
- Token: ?
- Server Address:?
- Server Port: ?
- Server URL: ?

Please help me.

brendongough, 2018-12-06:
This might be better answered by the Thingspeak IoT community as your account. I can see they have documentation for this type of thing, it'd be best to consult this or their forum.

https://au.mathworks.com/help/thingspeak/writedata.html

universam1, 2018-12-19:
thanks for support

closed 2018-12-19 by universam1

uribeer, 2019-03-11:
Hi Sam,
Since ubidots is not free anymore I also tried to connect to Thingspeak. I tried to follow the link above which gives all the information how to send data to Thingspeak via HTTP but it is not clear what fields should be filled in the iSpindel menu. For instance there is a field for URL and a field for Server address. Also the address is according to the protocol. So some information is needed in order to configure the iSpindel more than the info which is in the Thingspeak forum. Is it possible to get an example of HTTP connection to Thingspeak?
Thanks.