universam1/iSpindel #233: Zero point calbration does not work in 6.x

Misenkooo, 2018-11-25:
Hi, I've noticed after calibration it is working properly only until I restart the device. After that the offset is not applied properly even though I can see it stored in the config file and also debug message "Applying offset" is displayed.

Please have a look. It works fine in 5.x.

superbee383, 2018-12-27:
Hello,
Has anyone have an answer to this issue. Every time I power the iSpindle on, I have to do a calibration. If I do not do the calibration, the tilt will read between 45° and 59°. When I do the calibration the reading is between 0° and 89°.

Misenkooo, 2018-12-27:
I've just loaded 5.x firmware. For me it works great and I don't see a point why to use 6.x. Battery life is even better for me.

superbee383, 2018-12-28:
Ok, I'll give that a try. Hopefully, they will carry this over to the newer version.
Thanks

superbee383, 2018-12-29:
5.x worked fine. Thanks.

universam1, 2019-03-07:
Fixed with 6.0.5. Please make sure to `reset` under `Maintenance` menu

closed 2019-03-07 by universam1

stefschin, 2019-06-03:
> 
> 
> Fixed with 6.0.5. Please make sure to `reset` under `Maintenance` menu

Hi @universam1 , What do you mean about this reset? software other hardware reset?