universam1/iSpindel #234: Temperature and tilt of iSpindel

jmelhus, 2018-11-29:
The temperature affects the buoyancy of the liquid, and therefore also the tilt of the iSpindel. Does anyone know how much the offset is? Is it linear? 

When beer is kept at the temperature the iSpindel was calibrated at there is no issue. I calibrated my iSpindel at 20 degrees celsius because my oechsel scale was calibrated at that temperature. But when brewing lagers by example at other temperature than 20 degrees celsius, or when using yeast temperature profiles, this will affect the tilt. Are there any way of compensating this, we know the temperature of the liquid, so maybe we can compensate this by adding a formulae to it?

brendongough, 2018-12-06:
This would probably be something good to have on one of the integration sites or something like Fermentrack. You could take tilt measurements in different liquids at different temperatures to give you some idea of the coefficient shift. I do stepped lagers and notice the same, however, I use my iSpindel as more of a guide than exact science.

bralexc, 2018-12-06:
The tilt of the device only depends on the liquid gravity, not which liquid
you're using. Same for hydrometers. That's why we can do the calibration
using water and sugar.

avollkopf, 2018-12-07:
> 
> The tilt of the device only depends on the liquid gravity, not which liquid you're using. Same for hydrometers. That's why we can do the calibration using water and sugar.

But gravity depends on temperature as density is changing with temperature. This is the reason, why a beer botte can burst when temperature goes below 0°C / 32 F

jmelhus, 2019-04-18:
Hi, thank you for your answer and solution! Unfortunately I do not have the skills to modify the excel sheet to contain one more polynominal degree and calculate the coefficient shift. Would you, or someone else please help me with that?

jmelhus, 2019-05-04:
Anyone, please? :)

stefschin, 2019-05-07:
My experience is that density changes with temperature, like does with our typical glass hydrometer.
[image]
As I see on 5th day fermentation was ready with 1013 but when we let temperature to go up, until 23ºC, it goes to 1016, and after day 9 we le let temp to go down (5ºC) and density continue to go up until 1019... Normally it would get down  with higher temp and higher with lower temp... maybe CO2 is affecting too. 
Anyone has similar behaviour of you iSpindel?

jmelhus, 2019-05-08:
[image]
I also see some strange behaviour. When the temperature increases I assume the specific gravity to decrease, and when the temperature decrease I assume the specific gravity to increase, for all my different ispindels. This thery is not consistent over my different ispindels. I guess it depends on how the device is balanced to how distinct this shows. Therefore I would like to calibrate each device with the temperature coefficient. But unfortunately I do not know how to implement this into the formulae.

stefschin, 2019-05-22:
I share nice explanation of this behaviour from [Vito](http://vito.tw/?p=971).

OpenSourceDistilling, 2019-11-06:
@stefschin I read your post. Thanks for that. It helped me make sense of it. 

@universam1 Does a temperature variable exist in the iSpindel calibration formula field that we can make use of? Can we use a formula like the one in the link below, in combination with the calibration formula for the iSpindel to get temperature correct gravity readings? Or would it be better to perform the temperature correction calculation outside of the iSpindel after the iSpindel feeds me the uncorrected gravity reading and current temp value? 
https://www.homebrewtalk.com/forum/threads/temp-correction-formula-for-hydrometer.10684/#post-100698

universam1, 2019-11-10:
@OpenSourceDistilling There is indeed the variable `temp` available that one can add to the formula, which is mentioned in the docs actually.
So yes you are free to create a more advanced calibration formula including the temperature coefficient. However, from my experiments I was not able to derive a one fits all coefficient of temperature, as apparently the temperature has an impact on the Accelerometer that is different btw. every IC and so iSpindel. If you are lucky to derive something more generic I'd be happy to learn

OpenSourceDistilling, 2019-11-12:
@universam1 Thanks for the response. I should have read the docs first, sorry about that. I will go through them with a fine-tooth comb. Looks like everything I need is already here so I will play around with this and let you know if I find anything interesting or helpful. Cheers.

ErikdBr, 2019-12-15:
I have an iSpindel in my fermentation fridge in my shed just to monitor the temperature so I can intervene if the temperature droppes to low because I have bottles of beer in there which I don't want to freeze. The iSpindel Is in a fixed position thus can not move. Gravity reading dropped 9 points as the temperature drops. 
[image]

patadata, 2020-04-24:
> @universam1 Thanks for the response. I should have read the docs first, sorry about that. I will go through them with a fine-tooth comb. Looks like everything I need is already here so I will play around with this and let you know if I find anything interesting or helpful. Cheers.

@OpenSourceDistilling did you ever find a formula for this? I ferment under high pressure(35psi) and high temperature (up to 30 C) and there are big deviations.

OpenSourceDistilling, 2020-04-24:
@patadata Yes, I did research about it. There were two formulas I found, one for hydrometers calibrated at 60F (I believe this to be the more accurate one as it ties into more of the online calculators better) and another formula that can be used with any hydrometer regardless of the temp it's calibrated at (my hydrometer is calibrated at 68F, I think most of them are). I had a spreadsheet but can't seem to find it at the moment. 

The second formula can be found in the link below. The denominator of the equation is static so I would suggest calculating that once outside of the iSpindel and feed in the other "MG" and "TR" parameters. I'll take another look for the excel workbook as it just does everything for you including adding the iSpindel parameters. 
https://straighttothepint.com/hydrometer-temperature-correction/

I hope that all makes sense. Let me know if you need any clarification. Just check the degrees F on the hydrometer you used for iSpindel calibration and feed it into the formula.

OpenSourceDistilling, 2020-04-24:
@patadata I found the references I was looking for, the spreadsheet wasn't as complete as I remember it being. Read all the solutions (there are only 2 or 3) in the post below. It will guide you. Let me know if you need any help along the way. 

https://homebrew.stackexchange.com/questions/4137/temperature-correction-for-specific-gravity

patadata, 2020-04-25:
Thank you very much for fast reply!

I have done some research since I posted the question. I came up with the
same solution as you link to above. My conclusion though is that I think
it's not so much the temperature that mess everything up as it is the
pressure. I am fermenting under pressure (might have mentioned this before)
and under quite high pressure, 2.4 bar (35 psi) and also up to 30 C. I have
today actually made some tests with wort with different SG and I have also
been experimenting with different temperatures, all the way up to about
29,5C. I didn't actually see as much deviations as I saw when the iSpindel
was in the fermenter under pressure.

I will do some more experimenting and see what I come up with.

Thank you again for your answers!

/Patrik

/Patrik

OpenSourceDistilling, 2020-04-25:
Take care @patadata!

thegreatgunbantoad, 2020-04-26:
I'm not sure that you'd be able to just create a long complex formula using temp with the current firmware.  The input for the poly is only 100 characters long that would need extending, I'm not sure how long a string tinyexpr can handle though.  I was considering trying to rob the calcs from the HTML calibration sheet to update the firmware, but i never managed to get Platform IO working properly in VS Code.

OpenSourceDistilling, 2020-04-27:
@thegreatgunbantoad I didn't realize there was a limit on the input. The formula as written below is 189 characters and that would need to be in addition to the iSpindel calibration formula. I think we could consolidate the numerator (currently 91 characters) to about 15 characters because it is static. That still doesn't leave even close to enough room. I suppose the only way to do this right now is to temp correct after collecting data on the reporting side of things (aka Ubidots or other data collection services). 

cg = corrected gravity
mg = measured gravity
tr = temperature at time of reading
tc = calibration temperature of hydrometer

Formula:
cg = mg * ((1.00130346 - 0.000134722124 * tr + 0.00000204052596 * tr^2 - 0.00000000232820948 * tr^3) / (1.00130346 - 0.000134722124 * tc + 0.00000204052596 * tc ^2 - 0.00000000232820948 * tc^3))

thegreatgunbantoad, 2020-04-27:
It should be quite possible to update the firmware to take a longer polynomial.  Also there is some stuff to be done to shorten what you have.  Rip out all the spaces (this can cause a bit of odd behaviour with the current firmare anyway, so do take them out) and convert numbers like 0.00000000232820948 to use E notation, 2.32820948E-9.  Also I think 6 s.f. is fine so 2.32821E-9

patadata, 2020-04-27:
Maybe add another input field, so that ther is one polinominal for Plato
and another for Temperature corrected Plato. Then ther has to be another
parameter as Wall, maybe "Plato"?

closed 2020-07-03 by stale[bot]

ColOfAbRiX, 2021-03-24:
The above correction formula is for Fahrenheit. This is the formula for Celsius rounded to max 10 significant figures:

CG = Corrected Gravity
MG = Measured Gravity
TR = Temperature at time of Reading
TC = Temperature at time of Calibration
```
CG = MG * (33.0033 - 0.0000203053 * TR + 5.887137834E-6 * TR^2 - 1.357811769E-8 * TR^3) / (33.0033 - 0.0000203053 * TC + 5.887137834E-6 * TC^2 - 1.357811769E-8 * TC^3)
```

thegreatgunbantoad, 2021-03-28:
There's another thread on this and it is quite doable in the poly when the char limit extends to 250.  Originally I though I was going to be a pig but it's not so bad.