universam1/iSpindel #236: wifi visibility problem

rasale12, 2018-12-01:
Hello folks, i'm trying my new setup for controlling the fermentation of 4 keg with craftbeerpi and all works fine. I want to use 2 ispindle i've bought for take the input temperature but i didn't find the devices over wi-fi. I pull the switch of the ispindle, the blu led blink 3 times very fast but nothing happens..

Misenkooo, 2018-12-01:
You need to press it twice within a second to switch into config mode - blue diode must be blinking. Otherwise you will never see them because they are not in AP mode. You may connect in into PC and check in debug mode what's going on.

rasale12, 2018-12-07:
Hello, now i config my two ispindle and I connect to my wifi network. I create in craftbeerpi my 2 fermenter and I create 3 sensor (temperature, battery voltage, density) each ones. At the moment I create the sensor I can see the right value but there is no refresh (i set 15s in ap mode). If reset manually the ispindle pressing the switch there is a refresh of the value. what's going wrong?
thank you
Alessio

universam1, 2018-12-10:
Your reset pin wiring is not correct. Also always investigate the log output first

closed 2018-12-10 by universam1