universam1/iSpindel #239: HTTP address truncation IFTTT webhook

Colin66, 2018-12-11:
Hi, I'm having some problems sending data to IFTTT service, which requires data posting to :

http://maker.ifttt.com/trigger/XXXXXXXXX/with/key/XXXXXXXXXXXXXXXXXXXXX

I've selected HTTP as service type but when I save the configuration I see it appear to write the full HTTP address  OK but but when it attempts to post to that address it's shortened to "HTTP://maker.ifttt.com/trigger/iSpindel" and fails : 

WM: WiFi save
*WM: Parameter
*WM: name
*WM: iSpindel000
*WM: Parameter
*WM: sleep
*WM: 25
*WM: Parameter
*WM: vfact
*WM: 191.80
*WM: Parameter
*WM: 
*WM: 
*WM: Parameter
*WM: 
*WM: 
*WM: Parameter
*WM: tempscale
*WM: 0
*WM: Parameter
*WM: 
*WM: 
*WM: Parameter
*WM: 
*WM: 
*WM: Parameter
*WM: selAPI
*WM: 3
*WM: Parameter
*WM: token
*WM: 
*WM: Parameter
*WM: server
*WM: HTTP://maker.ifttt.com/trigger/XXXXXX/with/key/XXXXXXXXXXXXXXX
*WM: Parameter
*WM: port
*WM: 80
*WM: Parameter
*WM: url
*WM: HTTP://maker.ifttt.com/trigger/XXXXXXX/with/key/XXXXXXXXXXXXXXX
*WM: Parameter
*WM: db
*WM: ispindel
*WM: Parameter
*WM: username
*WM: 
*WM: Parameter
*WM: password
*WM: 
*WM: Parameter
*WM: job
*WM: ispindel
*WM: Parameter
*WM: instance
*WM: 000
*WM: Parameter
*WM: 
*WM: 
*WM: Parameter
*WM: POLYN
*WM: -0.00031*tilt^2+0.557*tilt-14.054
*WM: Sent wifi save page
*WM: Connecting to new AP
*WM: Connecting wifi with new parameters...
*WM: previous settings invalidated
*WM: After waiting...
*WM: 3.50
*WM: seconds
*WM: Connection result: 
*WM: WL_CONNECTED
saving config...{"Name":"iSpindel000","Token":"","Sleep":25,"Server":"HTTP://maker.ifttt.com/trigger/iSpindel","API":3,"Port":80,"URL":"HTTP://maker.ifttt.com/trigger/iSpindel","DB":"ispindel","Username":"","Password":"","Job":"ispindel","Instance":"000","Vfact":191.8,"TS":0,"OWpin":12,"SSID":"XXX","PSK":"XXX","POLY":"-0.00031*tilt^2+0.557*tilt-14.054","aX":0,"aY":0,"aZ":0}saved successfully
Samples:30 min:63.28 max:63.36 time:512
x: 10240 y: 12208 z: -2074
Tilt: 63.32
Tacc: 33.17
Volt: 4.77
Temp: 21.37
Gravity: 19.97
IP: 192.168.1.250

calling HTTP
{"name":"iSpindel000","ID":11768341,"angle":63.3151,"temperature":21.375,"temp_units":"C","battery":4.770594,"gravity":19.96978,"interval":1,"RSSI":-80}HTTPAPI: posting
code: -1
[HTTP] POST... failed, error: connection refused

Final-sleep: 1s; RT: 278006
{$

I also seem to be having trouble changing between service types, e.g I select HTTP, save, and then I see it trying (and failing) to contact CraftBeerPi service.

Help much appreciated. Great project by the way.

Colin66, 2018-12-16:
Hi, does anyone know why http post addresses are cut short?
Thanks,
Colin

universam1, 2018-12-19:
Looks like you are misunderstanding the fields usage.
The `server` would be something like `maker.ifttt.com`  and   
`url` field then would be `/trigger/XXXXXX/with/key/XXXXXXXXXXXXXXX`. If this is latter part is more then 40 characters it will be truncated, but first test

Colin66, 2018-12-19:
Hi thanks, yes I was misunderstanding the fields. I just tested and it does still truncate the url field (at 37 characters I think).  Can the field be lengthened to allow this ifttt service to be used ?  Looks like it needs at least an additional 17-20 characters for the url field.  It seems really neat and useful for IoT stuff.
thanks

Colin66, 2018-12-22:
Hi, I played around with the source code and got this working - extended url variable length ( char my_url[TKIDSIZE *2];  etc..) and relabelled 3 variables in JSON string so IFTTT accepts them (it's limited to 3). I've not noticed any negative effects.
With an easy IFTTT rule I can now relay the data straight to a Google spreadsheet for logging & plotting.
Thanks & great project, Merry Christmas!

universam1, 2018-12-27:
I’m considering to rewrite the http generic to use variables that would allow such unknown service formatting

Colin66, 2018-12-27:
If you can implement HTTPS as well that would allow direct access into Google drive documents such as sheets without going via an http intermediate service. This would simplify things. Is HTTPS too resource heavy for the d1 mini  ? 
Cheers 
Colin

universam1, 2018-12-27:
So far it was necessary to store the fingerprint of the endpoint which is super difficult with a distributed service like google. 
Also the handshake for encryption is super power intensive and the benefits are not paying off. Said that there are some compromises wit bearssl possible meantime, could make sense to look into.