universam1/iSpindel #240: Missing reports from iSpindel

terragady, 2018-12-12:
I have set iSpindel to report every 15 minutes, I do use brewpiless in the AP+Station mode so iSpindel is directly reporting to BrewPiless network. 
Every some time I am missing reports, sometimes even 2-3 in a row. Does anyone has this issue?

universam1, 2018-12-14:
check your wifi reception RSSI to be good enough

terragady, 2018-12-14:
RSSI of brewpiless? I can check only that but this is RSSI of BrewPiLess <-> my router and iSpindel is reporting directly to BPL station (to 192.168.4.1) and not my home network as there is no signal in the freezer of my home network. The BPL device is attached to freezer wall so the distance is 10-15cm between iSpindel and BPL, of course there is also a freezer wall in between but it should be good. 
Anyway to check the RSSI of iSpindel<->BPL connection? Now it is in mid fermentation so can not switch to service mode in iSpindel

Misenkooo, 2018-12-14:
I had the same problem, transmission was very unreliable and lost lot of reports despite a repeater next to the fermentor. Also had very poor battery life. Switched to firmware v5.x and problems solved. But I switched to raspberry pi and tcp server so this may helped as well.

terragady, 2018-12-14:
yes I also heard from Vito that on his test setup iSpindel was stopping reporting after some time and he went back to version before v6. I have no problems with battery life. 14 days went from 4.21 to 4.18 with 15min interval time and some playing and calibrating at the beginning.
@Misenkooo you mean you do not use BPL at all? You just collect data from iSpindel on RPi? Is that right? 

For next brew or tests after brew I will switch back to 5.x version and check if that helps.

universam1, 2018-12-18:
If the iSpindel cannot make a Wifi connection within 5s it will break and go to sleep until next iteration. This can happen if you are on the border of wifi limits. Thats why I mention check the RSSI sent metric to see if that is the case.

closed 2019-05-19 by universam1