universam1/iSpindel #241: Single post to Ubidots

mrpalmer84, 2018-12-15:
Hi there,

I've got the iSpindel built and running, looking at the config pages it appears that the accelerometers and temp sensors are working well. The same with ubidots, however, I only ever get one post to Ubidots coming out of config and then nothing more, reading the logs I get one good output then something unintelligible, probably related to baud rate, and then nothing more. Is this something that someone has had before?

Thanks

R

tizoc, 2018-12-16:
I had this issue, looking at the logs my conclusion was that there was a timeout when trying to connect to the router. I changed the source code to increase the timeout (the value is hardcoded and not configurable), and thats how I solved the issue.

What I did was increase this number: https://github.com/universam1/iSpindel/blob/master/pio/src/iSpindel.cpp#L1200

A better solution would be to make this configurable.

mrpalmer84, 2018-12-16:
Thank you for the suggestion.

Unfortunately that doesn't appear to be my issue, I raised the wait to 15000 and with the 100 millisecond delay per iteration that should have produced a noticable delay to the output in the log if I was looping there but it appeared immediately on the 20s that I have configured to debug this.

Thanks anyway

R

tizoc, 2018-12-16:
@mrpalmer84 one thing, note that the variable that holds the wait variable is an `uint_8`, that makes 255 the maximum value you can use there, you may want to try 255 instead (or increase the delay from 100 to something bigger).

If you apply this diff you are going to get a new setting that lets you set the timeout in seconds:

https://gist.github.com/tizoc/81b241498879af15496220e7ec86a1b0

255 seconds is the maximum value, but if thats not enough to solve timeout issues nothing is.

@universam1 let me know if you want me to submit that change as a pull request (not sure if you are interested in adding that setting).

mrpalmer84, 2018-12-17:
Hi there,

I checked again, it doesn't appear to be the Wifi issue (at least as far as I can tell). I dropped some debugging logging in the disconnected while loop and I never hit it when listening with putty.

The last few lines of my log output are:

Final-sleep: 20s; RT: 3661
rld▒▒|▒d▒|
▒d▒b|▒▒▒▒r▒cl▒c▒▒nn▒lgg▒▒▒
cp▒▒lslsl▒s▒g▒c▒gcx

I'm hoping this means something to someone, I suspect that the baud rate is reset so I'm not getting good logging after waking

R

universam1, 2018-12-18:
thats a definite hardware issue. Your device does not boot up after reset. Either your power is too low (test it with voltage >4V) or the ESP is really broken. Also a floating RST pin could make that issue.

universam1, 2018-12-18:
@tizoc This value is actually a 'fuse' to prevent your iSpindle to forever unhappily wait on a Wifi connection when not possible for some reason, and draining your battery in a short time. Anyway you are in an unhappy connection state, but next iteration might be better - better than a dead battery.

closed 2018-12-18 by universam1