universam1/iSpindel #242: can't connect to iSpindel for initial config

p0ttsworth, 2018-12-18:
Hi

I've just built up my iSpindel and powered it up, and I get a wireless point, but it's called ESP_047D0F, not iSpindel, and when I connect to it and try to go to 192.168.4.1 I just get a can't connect page.

I've tried to update the firmware using the linux instructions (I'm using a mac), but stalled out at the point of trying to identify which usb port the iSpindel was on (I have no programming experience)

Has anyone had similar / got any idea how to resolve?

Thanks in advance.

brunoheins, 2018-12-22:
Hi,
What version are you trying? When I use the 5.9.1 version, the ip for connection is 192.168.244.1 .. Check the wi-fi connection to know the right mask, maybe thats it.
I have not used the 6.x.x versions yet, I can't see the flashing led when I reboot the ESP to enter configuration mode. I see the wifi point, but the power drops. I don't know what's wrong, but it doesn't happen with 5.x.x versions.
Good luck

p0ttsworth, 2018-12-23:
I’m not totally sure to be honest  

When I connect to the hotspot the network address comes up as 4.1

 I bought the kit from 3D mechatronics, so assumed it would come pre-flashed, but I assume it isn’t. Is ESP8266MOD what you use to communicate with the Wemos?

brunoheins, 2018-12-24:
Hi!

Sorry for my mess, the ESP8266MOD is a type of WeMos and I'm using the COM port via USB to send the flash to Wemos. I'm using Windows 10 on the Mac via VMWare because I had trouble installing the drivers direct on the Mac ..

Exec this command befor and after Wemos plugged:

ls -l /dev/ttyUSB*

If your drive is ok, you see the port after Wemos plugged on USB, if not, you need to review it driver, look at

https://github.com/universam1/iSpindel/blob/master/docs/Firmware_en.md

p0ttsworth, 2018-12-26:
I finally managed to work out.

I had a couple of evenings of trying installing various things, so there may be a driver that I previously installed that allowed this to work.

If anyone else is struggling to connect via mac, I found working through this article was a massive help.
https://diyprojects.io/esptool-update-of-firmware-nodemcu-esp8266-on-macos/#.XCEk79Gnyf1

The only thing that didn't work straight away from that article was that the first time I tried to run the esptool command to flash the Wemos I got an error saying "Pyserial is not installed for /usr/bin/python"
I found the installer on the pyserial website, unpacked it, then ran "sudo python setup.py install" from terminal which installed PySerial, and then the esptool command worked to flash the Wemos.

Now roll on the next brew day to give it a try!!