universam1/iSpindel #243: 6.x.x not come into config mode

brunoheins, 2018-12-24:
Hi,
I’m trying to update to 6.x version but I can’t start the config mode. The flashing led is very short when I reboot the ESP to enter configuration mode. I see the wifi point, but the power drops when I try to connect. I don't know what's wrong, but it doesn't happen with 5.x.x versions only for all 6.x version.
Any idea?

universam1, 2018-12-24:
Where is the log file?

brunoheins, 2018-12-24:
Log for 5.9.4 version - working config mode

PORT OPEN 115200

Communication with MCU..Got answer! Communication with MCU established.
AutoDetect firmware...

Can't autodetect firmware, because proper answer not received (may be unknown firmware). 
Please, reset module or continue.
FW 5.9.1
3.0.0-dev(c0f7b44)
Worker run!
Boot-Mode: 5
mounting FS... mounted!
	

ERROR config corrupted
...........
ERROR no Wifi credentials

going to Config Mode
I2C ERROR: 0 con:0
ERROR: OW DISCONNECTED
I2C ERROR: 0 con:0
ERROR: OW DISCONNECTED
*WM: Scan done
*WM: DUP AP: #NET-CLARO-WIFI
*WM: DUP AP: #NET-CLARO-WIFI
*WM: Adding parameter
*WM: name
*WM: Adding parameter
*WM: sleep
*WM: Adding parameter
*WM: vfact
*WM: Adding parameter
*WM: 
*WM: Adding parameter
*WM: 
*WM: Adding parameter
*WM: tempscale
*WM: Adding parameter
*WM: 
*WM: Adding parameter
*WM: 
*WM: Adding parameter
*WM: selAPI
*WM: Adding parameter
*WM: token
*WM: Adding parameter
*WM: server
*WM: Adding parameter
*WM: port
*WM: Adding parameter
*WM: url
*WM: Adding parameter
*WM: db
*WM: Adding parameter
*WM: username
*WM: Adding parameter
*WM: password
*WM: Adding parameter
*WM: job
*WM: Adding parameter
*WM: instance
*WM: Adding parameter
*WM: 
*WM: Adding parameter
*WM: POLYN
started Portal
*WM: SET AP
*WM: 
*WM: Configuring access point... 
*WM: iSpindel
*WM: Custom AP IP/GW/Subnet
*WM: AP IP address: 
*WM: 192.168.244.1
*WM: HTTP server started
I2C ERROR: 0 con:0
ERROR: OW DISCONNECTED
I2C ERROR: 0 con:0
ERROR: OW DISCONNECTED
I2C ERROR: 0 con:0
ERROR: OW DISCONNECTED
I2C ERROR: 0 con:0
ERROR: OW DISCONNECTED
I2C ERROR: 0 con:0
ERROR: OW DISCONNECTED

PORT CLOSED

brunoheins, 2018-12-24:
log for 6.0.4 - not working config mode

PORT OPEN 115200

Communication with MCU..Got answer! Communication with MCU established.
AutoDetect firmware...

Can't autodetect firmware, because proper answer not received (may be unknown firmware). 
Please, reset module or continue.

FW 6.0.4
2.2.1(cfd48f3)
Worker run!
mounting FS... mounted!
scanning for OW device on pin: 5
No devices found!
scanning for OW device on pin: 12
No devices found!
scanning for OW device on pin: 116
No devices found!
scanning for OW device on pin: 121
No devices found!
ERROR: cannot find a OneWire Temperature Sensor!
Acc Test Connection ERROR!
Boot-Mode: External System
power-cycle or reset detected, config mode

Double Reset detected
...........
ERROR no Wifi credentials

going to Config Mode
reboot RFCAL

FW 6.0.4
2.2.1(cfd48f3)
Worker run!
mounting FS... mounted!
scanning for OW device on pin: 5
No devices found!
scanning for OW device on pin: 12
No devices found!
scanning for OW device on pin: 116
No devices found!
scanning for OW device on pin: 121
No devices found!
ERROR: cannot find a OneWire Temperature Sensor!
Acc Test Connection ERROR!
Boot-Mode: Deep-Sleep Wake

Double Reset detected
...........
ERROR no Wifi credentials

going to Config Mode
Acc Test Connection ERROR!

Exception (28):
epc1=0x4020e5c3 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: sys 
sp: 3fffed00 end: 3fffffb0 offset: 01a0

>>>stack>>>
3fffeea0:  00339913 c160dd2f 3fff1ffc 401006dc  
3fffeeb0:  00000002 00000004 3fff1ffc 4020118d  
3fffeec0:  00339913 3ffef6a0 3fff1ffc 40209bc0  
3fffeed0:  3ffe8e0f 3fffef10 00000000 00000000  
3fffeee0:  3ffe8e14 3fffef08 00000000 00000000  
3fffeef0:  00000000 00000000 00000000 40209478  
[... 273 lines cut ...]
chksum 0x2d
csum 0x2d
vbb28d4a3
~ld

FW 6.0.4
2.2.1(cfd48f3)
Worker run!
mounting FS... mounted!
scanning for OW device on pin: 5
No devices found!
scanning for OW device on pin: 12
No devices found!
scanning for OW device on pin: 116
No devices found!
scanning for OW device on pin: 121
No devices found!
ERROR: cannot find a OneWire Temperature Sensor!
Acc Test Connection ERROR!
Boot-Mode: Exception
...........
ERROR no Wifi credentials

going to Config Mode
reboot RFCAL

vytux-com, 2018-12-24:
It's been noted before that this firmware will only work if everything is properly built and you seem to be missing the temp sensor so of cause it's not going to work.

BTW both versions are giving you the same error

universam1, 2018-12-25:
Thanks for the comment, that’s indeed true and the case here. Hence we can close the issue

closed 2018-12-25 by universam1

brunoheins, 2018-12-29:
You're right, I did not read the notes. I'm sorry for this. Happy 2019 for everyone. Tks!!