universam1/iSpindel #253: InfluxDB HTTP POST fails with 'connection refused'

mharbach, 2019-02-01:
Using firmware v6.0.4, iSpindel fails to send values to InfluxDB. I recreated this successfully several times. Using curl on a laptop on the same wifi continues to work without problems. The issue does not occur with firmware v6.0.2 under identical configuration. Here's the Serial log:

```
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v00000000
@cp:0
ld

FW 6.0.4
2.2.1(cfd48f3)
Worker run!
mounting FS... mounted!
reading config file
applying offsets
parsed config:
{"Name":"iSpindelMH","Token":"<redacted>","Sleep":30,"Server":"<redacted>","API":8,"Port":18086,"URL":"","DB":"ispindel","Username":"ispindel","Password":"<redacted>","Job":"ispindel","Instance":"000","Vfact":191.8,"TS":0,"OWpin":12,"SSID":"Harbachs-Mobile","PSK":"<redacted>","POLY":"0.003776433*tilt^2+0.028145299*tilt+1.524019679","aX":-689,"aY":1212,"aZ":1109}
Boot-Mode: Software/System restart

Double Reset detected

going to Config Mode
reboot RFCAL
{d l⸮⸮| ⸮d⸮|⸮l⸮c|⸮⸮⸮⸮r⸮b⸮c⸮⸮gg⸮lgg⸮⸮⸮cx⸮⸮lslslp⸮o⸮⸮l⸮⸮co⸮|l⸮⸮c⸮⸮oo⸮ l⸮⸮l`⸮nnd`ns⸮⸮⸮nc⸮ls⸮⸮gc⸮d⸮ccc⸮⸮⸮⸮l`⸮⸮n⸮
FW 6.0.4
2.2.1(cfd48f3)
Worker run!
mounting FS... mounted!
reading config file
applying offsets
parsed config:
{"Name":"iSpindelMH","Token":"<redacted>","Sleep":30,"Server":"<redacted>","API":8,"Port":18086,"URL":"","DB":"ispindel","Username":"ispindel","Password":"<redacted>","Job":"ispindel","Instance":"000","Vfact":191.8,"TS":0,"OWpin":12,"SSID":"Harbachs-Mobile","PSK":"<redacted>","POLY":"0.003776433*tilt^2+0.028145299*tilt+1.524019679","aX":-689,"aY":1212,"aZ":1109}
Boot-Mode: Deep-Sleep Wake

woken from deepsleep, normal mode
Samples:38 min:67.39 max:67.51 time:667
x: -13620 y: 7562 z: 600
Tilt: 67.45
Tacc: 33.44
Volt: 4.13
Temp: 26.56
Gravity: 20.60
After waiting 0ms, result 0
Rescue Wifi credentials
failed to connect

Final-sleep: 30s; RT: 1488
{d l⸮⸮| ⸮d⸮|⸮l⸮c|⸮⸮⸮⸮s⸮c⸮c⸮⸮gg⸮lgg⸮⸮⸮cx⸮⸮l{d{dx⸮g⸮⸮l⸮⸮cg⸮|l⸮⸮c⸮⸮gg⸮ l⸮⸮l`⸮ood`o{⸮⸮⸮oc⸮l{⸮⸮oc⸮l⸮ccc⸮⸮⸮⸮d`⸮⸮o⸮
FW 6.0.4
2.2.1(cfd48f3)
Worker run!
mounting FS... mounted!
reading config file
applying offsets
parsed config:
{"Name":"iSpindelMH","Token":"<redacted>","Sleep":30,"Server":"<redacted>","API":8,"Port":18086,"URL":"","DB":"ispindel","Username":"ispindel","Password":"<redacted>","Job":"ispindel","Instance":"000","Vfact":191.8,"TS":0,"OWpin":12,"SSID":"Harbachs-Mobile","PSK":"<redacted>","POLY":"0.003776433*tilt^2+0.028145299*tilt+1.524019679","aX":-689,"aY":1212,"aZ":1109}
Boot-Mode: Deep-Sleep Wake

woken from deepsleep, normal mode
Samples:42 min:67.32 max:67.37 time:757
x: -13646 y: 7648 z: 616
Tilt: 67.34
Tacc: 32.05
Volt: 4.13
Temp: 26.62
Gravity: 20.55
After waiting 4812ms, result 3
IP: 192.168.0.161

calling InfluxDB
Sending to db: ispindel w/ credentials: ispindel:<redacted>
INFLUXDB: posting to db: /write?db=ispindel
POST data: measurements,source=iSpindelMH tilt=67.34377,temperature=26.625,temp_units="C",battery=4.134515,gravity=20.54625,interval=30,RSSI=-46
code: -1
[HTTP] POST... failed, error: connection refused

Final-sleep: 30s; RT: 11324
```

m0cs, 2019-02-19:
@mharbach i worked on a fix. I'll do a pr soon. If you want apply the fix

https://github.com/universam1/iSpindel/blob/master/pio/lib/Sender/Sender.cpp#L204

`http.begin(server, port, uri);`
to 
`http.begin(server + ":" + port + uri);`

closed 2019-02-27 by universam1

f4b0, 2020-01-22:
Hi, i have the actual software 6.2.0 installed and want to use it with influxdb but i get the same error as described above. The database is up and running and also the table is available. 
What could be the problem? My db is working and i am using it for a lot of other stuff.

Here is my log:
FW 6.2.0
2.2.1(cfd48f3)
Worker run!
mounting FS... mounted!
reading config file
parsed config:
{"Name":"iSpindel000","Token":"","Sleep":30,"Server":"192.168.178.21","API":8,"Port":8086,"Channel":0,"URI":"","DB":"ispindel","Username":"myUser","Password":"myPw","Job":"ispindel","Instance":"000","Vfact":191.8,"TS":0,"OWpin":12,"POLY":"-0.00031*tilt^2+0.557*tilt-14.054","SSID":"mySSID","PSK":"myWifiPw","Offset":[0,0,0,0,0,0]}
offsets not available
Boot-Mode: Deep-Sleep Wake

woken from deepsleep, normal mode
Samples:42 min:64.55 max:64.59 time:756
x: -10150 y: -13342 z: -194
Tilt: 64.57
Tacc: 20.26
Volt: 4.25
Temp: 21.00
Gravity: 20.62
IP: 192.168.178.47

calling InfluxDB
Sending to db: ispindel w/ credentials: myUser:myPw
INFLUXDB: posting to db: /write?db=ispindel
POST data: measurements,source=iSpindel000 tilt=64.56648,temperature=21,temp_units="C",battery=4.254432,gravity=20.61719,interval=30,RSSI=-70
code: -1
[HTTP] POST... failed, error: connection refused

Final-sleep: 30s; RT: 6192

Thanks in advanced!

m0cs, 2020-01-24:
> @mharbach i worked on a fix. I'll do a pr soon. If you want apply the fix
> 
> https://github.com/universam1/iSpindel/blob/master/pio/lib/Sender/Sender.cpp#L204
> 
> `http.begin(server, port, uri);`
> to
> `http.begin(server + ":" + port + uri);`

Hi! Did you try update the code with this solution?

f4b0, 2020-01-24:
Hi, thanks for your quick response!

Yes i already tried this and its still not working... 
Its working great with curl so i guess its a problem of the code but i am not 100% sure.
Later on i can pass the curl code here, maybe you can take a look at if there is a difference to the code.

The curious thing ist that MQTT is also not working but i still using it for other things . One question to MQTT.
My MQTT broker at home does not have user and pw configured but in ispindel interface i can pass
user and pw. 
Are these fields required or is it optional?

m0cs, 2020-01-24:
Ok @f4b0 , i'll update mine and test if this error shows, ok? i have mine on 6.0.3 with this code, i never updated anymore, because, its working LOL! But i'll update so i can help you and check.
I'm not sure what its MQTT for you? iSpindel doesn require any password (if influxdb doesn't have) Are you asking about a public ip from outside your home?

Greetings!

f4b0, 2020-01-25:
Hi mOcs, thanks for your support. I found both issues today. 

**InfluxDb Problem:**
It was my Influxdb. The database used the whole memory of my Raspberry and the query took to long. I had to remove an old database and now its working....

**MQTT Problem:**
I am running an old version of Mosquito  (MQTT broker) which is based on an old MQTT protocol version 3.1.
The PubSubClient.h library is basically configured for MQTT version 3.1.1. 
I had to change the PubSubClient.h as following:

```
#define MQTT_VERSION_3_1      3
#define MQTT_VERSION_3_1_1    4

// MQTT_VERSION : Pick the version
//#define MQTT_VERSION MQTT_VERSION_3_1
#ifndef MQTT_VERSION
#define MQTT_VERSION MQTT_VERSION_3_1
#endif

```

Everything is working now and i can go brewing :-)

Thanks!

PR #258 (2019-02-27): Influxdb http post parameter fix.
Fix #253 

Influxdb http post connection refused.