universam1/iSpindel #255: MQTT improvements: LWT (last will) and remote config?

fodi666, 2019-02-09:
Can you add a possibility to include a LWT in the MQTT topic? E.g. if the device goes offline from the broker for a given amount of time (let's say 5 times the set interval), then the broker would consider the device to be offline and would publish a message that is set previously (for example "offline" or empty values)
Could you also add the possibility to change at the interval via MQTT by publishing to a given topic? This way we could go with frequent measurements in the beginning and then gradually increase the interval to keep up with battery voltage when the beer is not likely to change too much anymore.

FernandoGarcia, 2019-07-19:
Hi!
You can find infos about this feature request here: https://github.com/universam1/iSpindel/issues/210
Best regards.

universam1, 2020-01-18:
happy to accept a PR

closed 2020-06-23 by stale[bot]